arjenhiemstra/ithowifi #198: Offline status in MQTT lwt topic

broekd, 2023-08-03:
**Describe the bug**
I get randomly on the MQTT topice lwt an offline status. With my homeassistant integration this leads to the device going offline. 

Below an overview off the received lwt messages
Received 04:28:16
QoS: 0
Payload: online
Received 04:40:21
QoS: 0
Payload: offline
Received 04:40:21
QoS: 0
Payload: online
Received 04:51:15
QoS: 0
Payload: offline
Received 04:51:15
QoS: 0
Payload: online
Received 05:25:21
QoS: 0
Payload: offline
Received 05:25:21
QoS: 0
Payload: online
Received 05:28:21
QoS: 0
Payload: offline
Received 05:28:21
QoS: 0
Payload: online
Received 10:58:02
QoS: 0
Payload: online

**To Reproduce**
Have not been able to find a way to reproduce this

**Expected behaviour**
99,9% uptime

**Device information**
THO Wifi Add-on
door Arjen Hiemstra
Firmware: 2.5.6
Hardware: NON-CVE 1
- Type/model number Itho (on sticker outside the box)
- CC1101 RF module enabled yes
- In case of wifi issues, wifi access point make and model:

**Debug logging**
Please provide debug logging from the debug page if possible:

broekd, 2023-08-03:
Hmm.☹️
All my devices (50+) are connected via zigbee2mqtt. None of them are having these intermediate connecting issues. Also wifi is rock solid.
Can we somehow trace on ITHO add-on side?

arjenhiemstra, 2023-08-03:
What is in the syslog of the add-on?

broekd, 2023-08-03:
2023-08-02 00:00:25 I: I2C init: QueryStatus
     16887 I: I2C init: QueryStatusFormat - items:0
     16708 I: I2C init: QueryDevicetype - fw:4 hw:43
     16501 I: Hostname: WTW
     15868 I: mDNS: started
     15469 I: Webserver: started
     15199 I: MQTT: connected, System config: 1
     14542 I: Setup: init of CC1101 RF module successful
     12543 I: WiFi: connection successful
     12340 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6764 E: Unable to set wifi disconnect
      4214 I: I2C sniffer capable hardware: yes
      4054 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3816 I: System boot, last reset reason: SW_RESET
     16230 I: I2C init: QueryStatus
     15869 I: I2C init: QueryStatusFormat - items:33
     15524 I: I2C init: QueryDevicetype - fw:4 hw:43
     15290 I: Hostname: WTW
     15079 I: mDNS: started
     14876 I: Webserver: started
     14688 I: MQTT: connected, System config: 1
     14396 I: Setup: init of CC1101 RF module successful
     12821 I: WiFi: connection successful
     12307 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6718 E: Unable to set wifi disconnect
      3972 I: I2C sniffer capable hardware: yes
      3721 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3460 I: System boot, last reset reason: SW_RESET
2023-08-01 03:30:08 I: I2C init: QueryStatus
2023-08-01 03:30:08 I: I2C init: QueryStatusFormat - items:33
2023-08-01 03:30:08 I: I2C init: QueryDevicetype - fw:4 hw:43
2023-08-01 03:30:07 I: Hostname: WTW
2023-08-01 03:30:07 I: mDNS: started
2023-08-01 03:30:07 I: Webserver: started
2023-08-01 03:30:06 I: MQTT: connected, System config: 1
2023-08-01 03:30:06 I: Setup: init of CC1101 RF module successful
     12509 I: WiFi: connection successful
     12321 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6741 E: Unable to set wifi disconnect
      4259 I: I2C sniffer capable hardware: yes
      4072 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3533 I: System boot, last reset reason: OWDT_RESET
     15961 I: I2C init: QueryDevicetype - fw:4 hw:43
     15587 I: Hostname: WTW
     15342 I: mDNS: started
     15139 I: Webserver: started
     14948 I: MQTT: connected, System config: 1
     14420 I: Setup: init of CC1101 RF module successful
     12811 I: WiFi: connection successful
     12629 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6800 E: Unable to set wifi disconnect
      4002 I: I2C sniffer capable hardware: yes
      3741 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3475 I: System boot, last reset reason: SW_RESET
     16577 I: I2C init: QueryStatus
     16268 I: I2C init: QueryStatusFormat - items:33
     15988 I: I2C init: QueryDevicetype - fw:4 hw:43
     15724 I: Hostname: WTW
     15262 I: mDNS: started
     15005 I: Webserver: started
     14753 I: MQTT: connected, System config: 1
     14515 I: Setup: init of CC1101 RF module successful
     12468 I: WiFi: connection successful
     12281 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6718 E: Unable to set wifi disconnect
      4183 I: I2C sniffer capable hardware: yes
      4037 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3815 I: System boot, last reset reason: SW_RESET
2023-07-30 13:05:40 I: I2C init: QueryStatus
2023-07-30 13:05:40 I: I2C init: QueryStatusFormat - items:33
2023-07-30 13:05:40 I: Hostname: WTW
2023-07-30 13:05:39 I: mDNS: started
2023-07-30 13:05:39 I: Webserver: started
2023-07-30 13:05:39 I: MQTT: connected, System config: 1
2023-07-30 13:05:38 I: I2C init: QueryDevicetype - fw:4 hw:43
2023-07-30 13:05:38 I: Setup: init of CC1101 RF module successful
     13400 I: WiFi: connection successful
     13053 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6965 E: Unable to set wifi disconnect
      4276 I: I2C sniffer capable hardware: yes
      4089 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3857 I: System boot, last reset reason: SW_RESET
2023-07-29 00:17:04 I: I2C init: QueryStatus
2023-07-29 00:17:03 I: I2C init: QueryStatusFormat - items:33
2023-07-29 00:17:03 I: I2C init: QueryDevicetype - fw:4 hw:43
2023-07-29 00:17:01 W: Warning: I2C timeout, trying I2C bus reset...
2023-07-29 00:16:58 E: Error: I2C bus could not be cleared!

arjenhiemstra, 2023-08-03:
I would expect wifi reconnect of mqtt connect error in the log but none seem to be there. 
Anything in the log on your mqtt server?

btw, I also see this:
2023-07-29 00:17:01 W: Warning: I2C timeout, trying I2C bus reset...
2023-07-29 00:16:58 E: Error: I2C bus could not be cleared!

Do you have i2c debug switched on?

arjenhiemstra, 2023-08-03:
And more verbose logging from the add-on is available if you use a remote syslog server (see menu syslog)

broekd, 2023-08-03:
Will let you know as soon as it happens again

closed 2023-10-09 by stale[bot]