arjenhiemstra/ithowifi #206: Remotes cannot connect to Add-On

dvandonkelaar, 2023-09-30:
**Describe the bug**
Both my remotes, a RFT Remote W ([536-0124](https://github.com/arjenhiemstra/ithowifi/raw/master/remotes/remote1.png)) and a RFT CO2 (as mentioned in the docs) have unrecognized commands and won't connect as RF device in the Add-On. The RFT CO2 blinks a red LED to indicate pairing failed.
Maybe the soldering is wrong, but commands are logged indicating the CC1101 works fine and the issue it seems similar to #76.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Debug'
2. Click on 'RF Debug Level 3'
3. Log with added button presses

**Expected behaviour**
The expected behaviour was a remote getting connected to the Add-On

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
- Firmware version: 2.5.6
- Hardware revision: 2
- Type/model number Itho: unfortunately, no sticker on the box. Itho settings according to the Add-On:
```
Itho device type: CVE-Silent
Itho fw version: 21
```
- CC1101 RF module enabled: yes
  * soldered myself since I had one lying around
- In case of wifi issues, wifi access point make and model:
  * no wifi issues

**Debug logging**
RFT Remote W (RFT bediening mode 1,2,3,T) (536-0124)
```
_W --- --,--,-- --,--,-- 50,49,E1 31DA 29:00,C8,40,02,03,EF,EF,7F,FF,7F,FF,7F,FB,7F,FF,F8,08,EF,0D,7D,00,00,0A,EF,EF,7F,FF,7F,FF (cmd:unknown) -> mode 1
_W --- --,--,-- --,--,-- 50,49,E1 31DA 29:00,C8,40,02,03,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,01,00,00,00,00,00,00,00,00,00,00 (cmd:unknown) -> mode 2
_W --- --,--,-- --,--,-- 50,49,E1 31DA 29:00,C8,40,02,03,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,03,00,00,00,00,00,00,00,00,00,00 (cmd:unknown) -> mode 3
_W --- --,--,-- --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B2,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,00,00,00,00,00,00,00,00,00,00 (cmd:unknown) -> mode T1
_W --- --,--,-- --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B2,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,2B,00,00,14,EF,EF,7F,FF,7F,FF (cmd:unknown) -> mode T2
_W --- --,--,-- --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B2,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,00,00,00,00,00,00,00,00,00,00 (cmd:unknown) -> mode T3
```

RFT CO2 Sensor
```
_W --- 50,49,E1 97,DF,2B --,--,-- 31DA 29:00,C8,40,02,2F,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,00,00,00,00,00,00,00,00,00,00,00,00,00 (cmd:unknown) -> mode A
_W --- 50,49,E1 97,DF,2B --,--,-- 31DA 29:00,C8,40,03,3E,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,01,C8,00,00,00,EF,EF,7F,FF,7F,FF (cmd:unknown) -> mode 1
_W --- 50,49,E1 97,DF,2B --,--,-- 31DA 29:00,C8,40,03,66,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,02,2A,00,00,00,EF,EF,7F,FF,7F,FF (cmd:unknown) -> mode 2
_W --- 50,49,E1 97,DF,2B --,--,-- 31DA 29:00,C8,40,02,B4,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,00,00,00,00,00,00,00,00,00,00,00,00,00 (cmd:unknown) -> mode 3
_W --- 50,49,E1 97,DF,2B --,--,-- 31DA 29:00,C8,40,01,B3,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,00,00,00,00,00,00,00,00,00,00,00,00,00 (cmd:unknown) -> mode 1 + timer
_W --- 50,49,E1 97,DF,2B --,--,-- 31DA 29:00,C8,40,01,B5,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,C8,00,00,14,EF,EF,7F,FF,7F,FF (cmd:unknown) -> mode 2 + timer
_W --- 50,49,E1 97,DF,2B --,--,-- 31DA 29:00,C8,40,01,BA,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,AA,00,00,1E,EF,EF,7F,FF,7F,FF (cmd:unknown) -> mode 3 + timer
```

General logging
```
30-9-2023 16:16:09: _W 173 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,AA,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 16:16:02: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B5,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,00,00,00,00,00,00,00,00,00,00,00,00,00 (cmd:unknown)
30-9-2023 16:16:02: _W 172 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,C8,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 16:12:51: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 16:12:48: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B5,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,C8,00,00,04,EF,EF,7F,FF,7F,FF (cmd:unknown)
30-9-2023 16:06:15: _W 171 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,C8,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 16:02:49: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 16:02:45: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B8,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,00,00,00,00,00,00,00,00,00,00,00,00,00 (cmd:unknown)
30-9-2023 15:56:12: _W 170 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,C8,00,20,20,20,20,20,20,20,20,28,20,20,20,00 (cmd:unknown)
30-9-2023 15:52:43: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 15:52:42: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B2,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,C8,00,00,18,EF,EF,7F,FF,7F,FF (cmd:unknown)
30-9-2023 15:46:10: _W 169 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,C8,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 15:46:03: _W 168 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,7E,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 15:45:57: _W --- --,--,-- --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B2,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,00,00,00,00,00,00,00,00,00,00 (cmd:unknown) -> 536-0124 timer 3rd press
30-9-2023 15:45:57: _W --- --,--,-- --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B2,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,2B,00,00,14,EF,EF,7F,FF,7F,FF (cmd:unknown) -> 536-0124 timer 2nd press
30-9-2023 15:45:57: _W --- --,--,-- --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B2,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,0D,00,00,00,00,00,00,00,00,00,00 (cmd:unknown) -> 536-0124 timer 1st press
30-9-2023 15:42:41: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 15:42:40: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B2,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,18,2A,00,00,00,EF,EF,7F,FF,7F,FF (cmd:unknown)
30-9-2023 15:36:35: _W 167 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,2B,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 15:32:40: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 15:32:40: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B7,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,18,2A,00,00,00,EF,EF,7F,FF,7F,FF (cmd:unknown)
30-9-2023 15:26:32: _W 166 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,2B,00,20,20,20,20,20,20,20,20,20,20,20,00,00 (cmd:unknown)
30-9-2023 15:22:37: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 15:22:34: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B4,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,18,2B,00,00,00,EF,EF,7F,FF,7F,FF (cmd:unknown)
30-9-2023 15:16:29: _W 165 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,2A,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 15:12:35: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 15:12:31: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,BB,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,00,00,00,00,00,00,00,00,00,00,00,00,00 (cmd:unknown)
30-9-2023 15:06:26: _W 164 50,49,E1 --,--,-- 50,49,E1 31D9 17:08,06,2B,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 15:02:28: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 15:02:28: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,BE,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,18,2B,00,00,00,EF,EF,7F,FF,77,FF (cmd:unknown)
30-9-2023 14:56:24: _W 163 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,2A,00,20,20,20,20,20,21,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 14:52:26: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 14:52:25: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,B7,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,18,2B,00,00,00,EF,EF,7F,FF,7F,FF (cmd:unknown)
30-9-2023 14:46:20: _W 162 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,2A,00,20,20,20,20,20,20,20,20,20,20,20,00,00 (cmd:unknown)
30-9-2023 14:42:25: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 14:42:23: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,BC,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,00,00,00,00,00,00,00,00,00,00,00,00,00 (cmd:unknown)
30-9-2023 14:36:17: _W 161 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,2A,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
30-9-2023 14:34:53: _W --- 50,49,E1 --,--,-- 50,49,E1 12C8 3:00,C8,40 (cmd:unknown)
30-9-2023 14:34:53: _W --- 50,49,E1 --,--,-- 50,49,E1 31DA 29:00,C8,40,01,A8,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,18,2E,00,00,00,EF,EF,7F,FF,7F,FF (cmd:unknown)
30-9-2023 14:26:15: _W 160 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,2A,00,20,20,20,20,20,20,20,20,20,20,00,20,00 (cmd:unknown)
```

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version 117.0.5938.132 

**Smartphone (please complete the following information):**
 - Not tested

arjenhiemstra, 2023-09-30:
31DA and 31D9 are not remote commands but fan status messages, probably originating from your fan.
I do not know what a 12C8 message is but it originates from the same RF ID as the 31DA and 31D9 so I think they are also not related to your remote.
No other messages in the RF log?

dvandonkelaar, 2023-09-30:
Strange, these are the exact messages I see when I have debug level 3 enabled and press a button on the remote.
Maybe these aren't the messages from the remote. If so, the remote isn't visible at all.

dvandonkelaar, 2023-10-14:
After some more investigation I came to the following conclusion. The logs below are from the Debug (level 3) log.
Could it be that the CC1101 isnt't soldered correctly?

Join message RFT Remote W:
```14-10-2023 13:09:58: _W --- --,--,-- --,--,-- 50,49,E1 1FC9 12:00,31,D9,50,49,E1,08,31,DA,50,49,E1 (cmd:unknown)```

Join message RFT CO2:
```14-10-2023 13:10:26: _I --- 50,49,E1 97,DF,2B --,--,-- 1FC9 12:00,31,D9,50,49,E1,00,31,DA,50,49,E1 (cmd:unknown)```

The general logging for this testing time-window:
```
14-10-2023 13:10:36: _W 8 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,C6,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
14-10-2023 13:10:29: RP --- 50,49,E1 97,DF,2B --,--,-- 31DA 29:00,EF,00,7F,FF,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,08,EF,03,5E,00,00,00,EF,EF,7F,FF,7F,FF (cmd:unknown)
14-10-2023 13:10:29: _W --- 50,49,E1 --,--,-- 50,49,E1 10E0 38:00,00,01,00,1B,28,15,01,FE,FF,FF,FF,FF,FF,0B,0B,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00 (cmd:unknown)
14-10-2023 13:10:26: _I --- 50,49,E1 97,DF,2B --,--,-- 1FC9 12:00,31,D9,50,49,E1,00,31,DA,50,49,E1 (cmd:unknown) <<-- join RFT CO2
14-10-2023 13:10:26: _W 7 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,5E,00,20,30,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
14-10-2023 13:10:25: RP --- 50,49,E1 97,DF,2B --,--,-- 2411 22:00,00,21,20,00,00,00,00,02,00,00,00,01,00,00,00,00,00,00,00,00,00 (cmd:unknown)
14-10-2023 13:10:20: RP --- 50,49,E1 97,DF,2B --,--,-- 31DA 29:00,EF,00,7F,FD,EF,EF,7F,FF,7F,FF,7F,FF,7F,FF,F8,00,EF,18,BF,00,00,00,EF,EF,7F,FF,7F,FF (cmd:unknown)
14-10-2023 13:10:11: _W 6 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,C8,00,20,20,20,20,20,20,20,20,20,20,20,00,00 (cmd:unknown)
14-10-2023 13:10:06: _W 5 50,49,E1 --,--,-- 50,49,E1 31D9 17:00,06,A0,00,20,20,20,20,20,20,20,20,20,20,20,20,00 (cmd:unknown)
14-10-2023 13:09:59: _W --- 50,49,E1 --,--,-- 50,49,E1 10E0 38:00,00,01,00,1B,28,15,01,FE,FF,FF,FF,FF,FF,0B,0B,07,E0,43,56,45,2D,52,46,00,00,00,00,00,00,00,00,00,00,00,00,02,00 (cmd:unknown)
14-10-2023 13:09:58: _W --- --,--,-- --,--,-- 50,49,E1 1FC9 12:00,31,D9,50,49,E1,08,31,DA,50,49,E1 (cmd:unknown)  <<-- join RFT Remote W
```

arjenhiemstra, 2023-10-14:
Hmm, the messages you logged aren't actually join commands, they are the reply of you itho to join commands of those devices. 
It seems that close range (ie. inside your itho) commands are being received correctly but anything else is out of range. This could very well be a hardware issue (as your itho is clearly receiving signals). Could you send me an e-mail with your order details. I'll make sure to replace the add-on

closed 2023-12-20 by stale[bot]

dvandonkelaar, 2024-01-19:
It has been a while, but after some short email conversation with Arjen we came to the conclusion that the mounted CC1101 module was wrong. The CC1101 was mounted by me (since I had one lying around), not ordered from the webshop.

The wrong module had this antenna:
[image]

The correct module has this antenna:
[image]