arjenhiemstra/ithowifi #208: Itho I2C connection status: init failed

klaaspieter, 2023-10-19:
**Describe the bug**
I'm seeing `I2C init: QueryDevicetype - failed` 

**Device information**
- Firmware version [e.g. 2.2.1] 
    2.5.6
- Hardware revision (e.g. non-cve 1.1)
    NON-CVE 1
- Type/model number Itho (on sticker outside the box)
    HRU ECO 350

- CC1101 RF module enabled [ yes, no ]
    Yes

**Debug logging**

```
2023-10-19 15:54:41 E: I2C init: QueryDevicetype - failed
2023-10-19 15:54:41 I: Hostname: nrg-itho-5cdc
     15640 I: mDNS: started
     15403 I: Webserver: started
     15116 I: MQTT: connected, System config: 1
     14882 I: Setup: init of CC1101 RF module successful
     13042 I: WiFi: connection successful
     12827 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6900 E: Unable to set wifi disconnect
      4081 I: I2C sniffer capable hardware: yes
      3785 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3487 I: System boot, last reset reason: NO_MEAN
2023-10-19 15:49:27 I: Hostname: nrg-itho-5cdc
2023-10-19 15:49:27 I: mDNS: started
2023-10-19 15:49:27 I: Webserver: started
2023-10-19 15:49:26 I: MQTT: connected, System config: 1
2023-10-19 15:49:26 E: I2C init: QueryDevicetype - failed
2023-10-19 15:49:25 I: Setup: init of CC1101 RF module successful
     12387 I: WiFi: connection successful
     12152 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6463 E: Unable to set wifi disconnect
      3828 I: I2C sniffer capable hardware: yes
      3599 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3383 I: System boot, last reset reason: POWERON_RESET
2023-10-19 15:40:04 I: Hostname: nrg-itho-5cdc
     16301 I: mDNS: started
     16053 I: Webserver: started
     15829 I: MQTT: connected, System config: 1
     15388 E: I2C init: QueryDevicetype - failed
     14710 I: Setup: init of CC1101 RF module successful
     13112 I: WiFi: connection successful
     12902 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6987 E: Unable to set wifi disconnect
      4143 I: I2C sniffer capable hardware: yes
      3829 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3517 I: System boot, last reset reason: POWERON_RESET
2023-10-19 13:49:53 I: Hostname: nrg-itho-5cdc
2023-10-19 13:49:53 I: mDNS: started
2023-10-19 13:49:52 I: Webserver: started
2023-10-19 13:49:52 I: MQTT: connected, System config: 1
2023-10-19 13:49:51 E: I2C init: QueryDevicetype - failed
2023-10-19 13:49:50 I: Setup: init of CC1101 RF module successful
2023-10-19 13:49:48 I: WiFi: connection successful
2023-10-19 13:49:48 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2023-10-19 13:49:42 E: Unable to set wifi disconnect
2023-10-19 13:49:40 I: I2C sniffer capable hardware: yes
2023-10-19 13:49:40 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
2023-10-19 13:49:39 I: System boot, last reset reason: SW_RESET
2023-10-19 13:49:34 I: Reboot requested
2023-10-19 13:49:06 I: Hostname: nrg-itho-5cdc
2023-10-19 13:49:06 I: mDNS: started
2023-10-19 13:49:06 I: Webserver: started
2023-10-19 13:49:05 I: MQTT: connected, System config: 1
     15200 E: I2C init: QueryDevicetype - failed
     14355 I: Setup: init of CC1101 RF module successful
     12839 I: WiFi: connection successful
     12340 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6743 E: Unable to set wifi disconnect
      3977 I: I2C sniffer capable hardware: yes
      3721 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
      3454 I: System boot, last reset reason: POWERON_RESET
2023-10-19 13:46:36 I: Hostname: nrg-itho-5cdc
2023-10-19 13:46:36 I: mDNS: started
2023-10-19 13:46:36 E: I2C init: QueryDevicetype - failed
2023-10-19 14:38:45 I: Webserver: started
2023-10-19 14:38:44 I: MQTT: connected, System config: 1
2023-10-19 14:38:43 I: Setup: init of CC1101 RF module successful
2023-10-19 14:38:41 I: WiFi: connection successful
2023-10-19 14:38:41 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2023-10-19 14:38:35 E: Unable to set wifi disconnect
2023-10-19 14:38:33 I: I2C sniffer capable hardware: yes
2023-10-19 14:38:33 I: HW rev: NON-CVE 1, FW ver.: 2.5.6
2023-10-19 14:38:32 I: System boot, last reset reason: SW_RESET
```

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser: Firefox
 - Version: 118.0.2

klaaspieter, 2023-10-26:
Updated to the new firmware (2.5.7). Still the same issue.

arjenhiemstra, 2023-10-26:
Did it work at sometime?

klaaspieter, 2023-10-26:
Yeah. It worked in the past. I noticed something was wrong because all the MQTT sensors in home assistant were “unavailable”. It's been like that for a while I only had time now to take a more serious look.

arjenhiemstra, 2023-10-30:
Then there are 3 options I guess:
1- the cable is broken
2- the add-on is broken
3- your itho is broken

Please try another cable and if that doesn't help, I can replace the add-on

galadril, 2023-10-30:
I just got my HRU 350 WtW unit installed and have the same connection issue when connecting the  NON-CVE v1 addon:
 _Itho I2C connection status: init failed - please power cycle the itho unit -_

Version 2.5.7

[image]

arjenhiemstra, 2023-10-30:
> I just got my HRU 350 WtW unit installed and have the same connection issue when connecting the NON-CVE v1 addon: _Itho I2C connection status: init failed - please power cycle the itho unit -_
> 
> Version 2.5.7
> 
> ps dont want to hijack this thread, so should i make a new issue or is it maybe related? Btw i've tried multiple cables, and the WtW is freshly installed today
> 
> [image]

No idea to be honest. A query devicetype failing all together means no i2c communication or the command is not accepted by the HRU.
This weekend I tested with a HRU350 and fw. version 2.5.7 and that was working fine. That would mean the fw of the add-on seems not the issue. Either a broken add-on or maybe itho made  changes to their firmware leading to these issues?
You're using short high quality cables?

galadril, 2023-10-30:
Ill try another cable of 25cm tomorrow, just othered one.

I tried 2 cables of 1m today. And then I created my own 15cm cable. Tested all cables with my cable tester btw.

arjenhiemstra, 2023-10-30:
1m is definitively on the long end. I believe Itho delivers a 20cm CAT6 S/FTP cable with their service tool.
But I would expect a good self created cable of 15cm to be fine imho

galadril, 2023-10-31:
Can we downgrade firmware without issues via the UI? then we can test some older ones as well if needed

arjenhiemstra, 2023-10-31:
absolutely, the more dangerous versions in terms of bugs have been removed. All versions you can find on GH currently can be used. Worst thing that can happen is that you need to reconfigure the add-on.

galadril, 2023-10-31:
Ok i've also tested the new cable that came in, same connection issue.
_I2C init: QueryDevicetype - failed_

I've also tried 4 different FW versions, going back to '_nrgitho-noncve-v2.4.3_'

My WtW type is 'HRU Eco 350' as it states on the manual.
Any ideas? I'm not going to open the PCB as i failed / broke my previous one 🦺

arjenhiemstra, 2023-10-31:
> Ok i've also tested the new cable that came in, same connection issue.
>
> _I2C init: QueryDevicetype - failed_
>
>
>
> I've also tried 4 different FW versions, going back to '_nrgitho-noncve-v2.4.3_'
>
>
>
> My [...]

Can send you a different module, that's the last thing I can think of

galadril, 2023-10-31:
Yeah we could try that?

klaaspieter, 2023-11-04:
It looks like in my case it is the cable. I used the itho cable to connect my WiFi access point to my modem and my WiFi no longer worked 😅 . The module works with the cable I freed up even though it's longer than the suggested length.

phwschroth, 2023-11-04:
I use an HRU-300 duozone with the non-cve hardware. There is a special problem with this.

Two CO2 controls and one for the bathroom are connected to the system.

If the system has been running for a number of days, one of the controls will go from automatic to the high position. With the touch button it no longer wants to go to another position and keeps jumping back to the high position. The HRU-300 also revs high.

After removing the non-cve hardware everything seems to work normal again.

I also tried the new software version version 2.6.0, but the Itho status no longer works. This does work with version 2.5.6.