arjenhiemstra/ithowifi #211: co2 sensor

gerben-f1, 2023-10-28:
hello,

I have itho cve eco.
and installed the print with rf module.
now i add the humudity sensor and co2 sensor.
co2 sensor is 230v powered.

but how can i add this to home assistant?
the humudity sensor works and is battery pwered.

arjenhiemstra, 2023-10-28:
does this example help?
https://github.com/arjenhiemstra/ithowifi/wiki/Home-Assistant#co2-sensor

gerben-f1, 2023-10-29:
When i add that i getting configuration fault in home assistant and cant reboot.And the name to change which one need to change? M.v.g Gerben

https://github.com/arjenhiemstra/ithowifi/wiki/Home-Assistant#co2-sensor

arjenhiemstra, 2023-10-29:
Not sure but using the example and translating that to this in my situation works:

 ```YAML
sensor:
  - name: Itho zolder CO2
    state_topic: "itho/remotesinfo"
    unit_of_measurement: "ppm"
    value_template: "{{ value_json.rft-co2.co2 }}"
    device_class: "carbon_dioxide"
    unique_id: "itho_zolder_co2"
    device:
      identifiers: "mv"
```

B-Lourens, 2023-10-30:
or: 
`mqtt:
  sensor:
    - name: "nrg-itho-CO2"
      unique_id: "nrg-itho-co2"
      device_class: "carbon_dioxide"
      state_topic: "itho/ithostatus"
      unit_of_measurement: "ppm"
      value_template: "{{ value_json['Highest CO2 concentration (ppm)'] }}"`

gerben-f1, 2023-10-30:
Can u see what i do wrong?M.v.g Gerben

B-Lourens, 2023-10-30:
Not really, because we have no idea what your configuration.yaml looks like.
But, if you're getting errors in Hass, the most likely issue is that you're placing the sensor examples in the wrong place, under the wrong category.

gerben-f1, 2023-10-30:
It already works.One question.Is it possible to complete stop the ventilationAnd that it only turns on when co2 is to high or the humadityM.v.g Gerben

arjenhiemstra, 2023-10-30:
If you have the CO2 sensor connected to the itho directly, no
If you have connected the CO2 sensor to the add-on, yes

Here's some more info on this topic:
https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan#help-fan-wont-go-lower-than-ie-30

closed 2024-01-06 by stale[bot]