arjenhiemstra/ithowifi #213: Monitor only RF Devices not in remotesinfo

bartvanraalte, 2023-11-06:
**Describe the bug**
I have a RFT CO2 added to the addon with remfunc "Monitor Only". On the "RF Devices setup" page, the "capabilities" column is filled with the correct information: `timestamp:1699130340, lastcmd:11, lastcmdmsg:IthoAuto, co2:481`

This information is however not published to the remotesinfo MQTT topic. Also via the API (`/api.html?get=remotesinfo`) it does not show this remote.

**To Reproduce**
Steps to reproduce the behaviour:
1. Add remote with remfunc "monitor only"
2. Validate information is added in the "capabilities" column on the webui
3. Check the `/api.html?get=remotesinfo` api
4. Remote is not listed

**Expected behaviour**
I would expect to also see remotes that are in "montor only" mode.

**Device information**
- Firmware version:  2.6.0
- Hardware revision: NON-CVE 1
- Type/model number Itho: Itho HRU eco 300 duo-zone
- CC1101 RF module enabled: Yes

arjenhiemstra, 2023-11-06:
Thanks for reporting this! I'll look into it.

arjenhiemstra, 2024-01-17:
I cannot reproduce this in 2.6.0. I have a monitor only device myself and this shows up on MQTT. Could you provide a screenshot of the web interface?

bartvanraalte, 2024-01-23:
Does this screenshot help, or are you looking for another page?
[image]

(Note: device with index 1 is out of reach, but the rest do work fine)

bartvanraalte, 2024-01-23:
Now see all the remotes on the /api.html?get=remotesinfo endpoint, and also on MQTT. The software version is exactly the same as when I reported the issue, and I didn't change the config. Maybe it only started to work after a restart (powercycle), but that would just be a guess.