arjenhiemstra/ithowifi #214: [Feature] Add OpenApi Spec / SwaggerUI to document api

galadril, 2023-11-09:
**Describe the bug**
It would be really cool to add an openapi specification to document all the possible options of your web api.

**Sample**
```
openapi: 3.0.0
info:
  title: Web API
  version: 1.0.0
paths:
  /api.html:
    get:
      parameters:
        - name: vremotecmd
          in: query
          schema:
            type: string
            enum: ["away", "low", "medium", "high", "timer1", "timer2", "timer3", "join", "leave", "auto", "autonight", "cook30", "cook60"]
        - name: vremoteindex
          in: query
          schema:
            type: string
            pattern: "^[0-11]$"
        - name: vremotename
          in: query
          schema:
            type: string
        - name: username
          in: query
          schema:
            type: string
            maxLength: 20
        - name: password
          in: query
          schema:
            type: string
            maxLength: 20
        - name: clearqueue
          in: query
          schema:
            type: string
            enum: ["true"]
        - name: get
          in: query
          schema:
            type: string
            enum: ["ithostatus"]
        - name: speed
          in: query
          schema:
            type: string
            pattern: "^[0-255]$"
        - name: command
          in: query
          schema:
            type: string
            enum: ["low", "medium", "high", "timer1", "timer2", "timer3", "clearqueue"]
        - name: timer
          in: query
          schema:
            type: string
            pattern: "^[0-65535]$"
        - name: currentspeed
          in: query
          schema:
            type: string
            enum: ["ithostatus"]
      responses:
        '200':
          description: Successful response
          content:
            text/plain:
              example: OK
        '400':
          description: Unsuccessful response
          content:
            text/plain:
              example: NOK
      description: Format the request as http://[DNS or IP]/api.html?[param]=[value]
      summary: Simple Web API

# Add more paths and details as needed for specific commands and functionalities

```

**Expected behaviour**
The API page would be an SwaggerUI api describing the API and allowing users to directly test the API from the portal.

**Screenshots**
[image]

[image]

arjenhiemstra, 2023-11-10:
good one! I'm currently working on changing the API response from the OK / NOK that it is now to a reply JSON based with some more info. This could be a change to accompanying that nicely.

galadril, 2024-09-05:
This is a sample on how we could add a test api link on the api documentation page.
And load swaggerUI with the openapi spec file of the api so people have a api developer portal
https://github.com/arjenhiemstra/ithowifi/pull/268

[image]

arjenhiemstra, 2026-03-28:
This has been implemented in **v3.0.0-beta1**:
- OpenAPI 3.0 specification at `/api/openapi.json`
- Interactive Swagger UI at `/swagger` (loaded from CDN)
- Full REST API v2 with proper GET/POST/PUT endpoints at `/api/v2/*`

See the [release notes](https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-3.0.0-beta1) for the complete list of endpoints.

closed 2026-03-29 by galadril