arjenhiemstra/ithowifi #215: Virtual Remotes update remote broken

Julianvantveld, 2023-12-22:
Using the virtual remote tab in the web portal, updating the virtual remote does not work.
Resetting and copying the ID works as expected, the update button however does not trigger an update.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Virtual Remotes tab'
2. Select a virtual remote
3. Change ID, Name and RemType
4. Click the Update button
5. Nothing happens

It seems the WebSocket event is posted on button tap, but no reload/saved data is triggered.
[image]

**Device information**
- Firmware version **2.6.0**
- Hardware revision **NON-CVE 1.1**
- Type/model number **Itho HRU 350**
- CC1101 RF module enabled **YES**

**Desktop (please complete the following information):**
 - OS: OSX
 - Browser Arc, Chrome, Safari
 - Version 1.15.1

arjenhiemstra, 2024-01-03:
Thanks for reporting this! Will be fixed in the next version.

arjenhiemstra, 2024-01-17:
should be fixed as of this version https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.7.0

closed 2024-01-17 by arjenhiemstra