arjenhiemstra/ithowifi #217: Ventilation setpoint never goes below 30%

JoostGadellaa, 2023-12-30:
**Describe the bug**
I might be missing something, but whatever I set the fan speed to, in the Itho status tab it never reports anything lower than 30%.

**To Reproduce**
Steps to reproduce the behaviour. I normaly use MQTT, but this is via the web interface just to be sure.
1. On the main page of the web interface, set the fan to something below 76
2. Click Itho status
3. Look at Ventilation setpoint (%)
4. Notice it is at 30%

**Expected behaviour**
I would expect the setpoint to become the set value 

**Screenshots**
[image]
[image]

**Device information**
- Current firmware version: 2.6.0
- Hardware revision: 2 
- Itho device type: CVE-Silent
- Itho fw version: 27
- CC1101 RF module enabled: no

**Debug logging**
Please provide debug logging from the debug page if possible:

arjenhiemstra, 2023-12-30:
see:
https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan#help-fan-wont-go-lower-than-ie-30

JoostGadellaa, 2023-12-30:
Ah, I could have guessed that this was known. Thanks!

closed 2023-12-30 by JoostGadellaa