arjenhiemstra/ithowifi #218: [Feature] Add fan/ventilation percentage to MQTT itho/ithostatus messages

FrankvdAa, 2023-12-30:
At the moment only the `Fan setpoint (rpm)` and `Fan speed (rpm)` are in the MQTT status messages, but it would also be nice to see the fan percentage (0-100%).

There's already a `Ventilation setpoint (%)`, so maybe adding `Ventilation speed (%)`?

I'm able to calculate percentage through `itho/state`, but that is only sent when speed is changed.

arjenhiemstra, 2023-12-30:
The info available on the MQTT topic is based on available information from within (with 2 exceptions) the itho firmware. Try enabling some of the I2C commands found behind the "System settings" menu of the add-on. I have "Speed status" and "ExhFanSpeed (%)" available which both seem to be % values of the actual fan speed.

FrankvdAa, 2023-12-30:
Sorry, didn't know that. After enabling `Fan system status (31D9)` I get the additional `Speed status`.

Thanks!

closed 2023-12-30 by FrankvdAa