arjenhiemstra/ithowifi #219: Fan not keeping set speed

FrankvdAa, 2024-01-02:
**Describe the bug**
Fan is not keeping the speed as set in the MQTT message.

**To Reproduce**
Add-on is connected to NodeRed through MQTT and CVE is running in 'auto' mode. Humidity and CO2 are low and fan speed is around 30%.

When sending a `{ "vremote": "high" }` or `{ "vremote": "medium" }` message all works as expected. When sending `{ "speed": 50 }` or `{ "speed": 220 }` I do get a confirmation on the `itho/state` topic showing the same speed and I see the full command also in the `itho/lastcmd` topic. However, you can hear the fan change speed only briefly before it returns to its original speed. 

**Expected behaviour**
I would expect it to keep the speed as set in the message. I know sending commands only work when in 'auto'/'medium' mode, but that is the case here. Am I overlooking something?

**Device information**
- Firmware version: 2.6.0
- Hardware revision: 2
- Type/model number Itho: CVE-S firmware version 27
- CC1101 RF module enabled: yes

**Debug logging**
I couldn't find any useful logging, but let me know what you need to investigate.

arjenhiemstra, 2024-01-02:
If your fan returns to another speed than set by a speed command, the command is being overruled by something else. This could be another command, remote device or internal Hum or CO2 sensor.

FrankvdAa, 2024-01-02:
That was what I was thinking at first too, but would it also lower the fan speed?

FrankvdAa, 2024-01-02:
Other remotes are not being used, nor is another command being sent to the add-on, so it can only be humidity or co2 that is overruling it. Any way to be sure?

arjenhiemstra, 2024-01-02:
some input is changing the speed. Most of the times it is a persistent MQTT topic or something else.
What does `http:://[your add-on ip]/api.html?get=lastcmd` show as output?

arjenhiemstra, 2024-01-02:
I see you mentioning CO2, what kind of CO2 sensor are we talking about?

FrankvdAa, 2024-01-02:
The `itho/lastcmd` topic shows just my last command. The CO2 I'm talking about is the build-in one.

arjenhiemstra, 2024-01-02:
Have you read this on the product page?
[image]

See for more information also the wiki on this github:
https://github.com/arjenhiemstra/ithowifi/wiki/CO2-sensors

FrankvdAa, 2024-01-02:
Yes, did see that, but may have misread it, looking at it now ;-)

I read it as if the CO2 would overrule any commands in case CO2 became too high, but I reckon it overwrites them always, right?

arjenhiemstra, 2024-01-02:
yep, every 30 seconds or so. A (virtual) remote command low will overrule the CO2 sensor.

FrankvdAa, 2024-01-02:
Okay clear, but then the speed command still doesn't work, or does it then?

arjenhiemstra, 2024-01-02:
not persistent because the co2 add-on inside your itho itself sends a speed command every time

FrankvdAa, 2024-01-02:
Clear, thanks. I can manage with the `vremote` commands, so no problem.

closed 2024-01-02 by FrankvdAa

arjenhiemstra, 2024-01-02:
If you have suggestions to make the explanation more clear, feel free to suggest an edit!

FrankvdAa, 2024-01-02:
Maybe add a short explanation and change:

```
The I2C to PWM protocol is overruled by the CO2 sensor. Virtual remote commands work as expected.
```

To something like:

```
The I2C to PWM protocol is overruled by the CO2 sensor, meaning it is not possible to persistently set a different fan speed, as it is overruled by the CO2 sensor. Virtual remote commands, such as 'low', 'medium' and 'high' do work as expected.
```