arjenhiemstra/ithowifi #223: Request; turn off unit from HA

dzjr, 2024-01-13:
Hello,

I'm not sure if I'm in the right place here, but I'll try anyway.

I just got a new Itho CVE-S ECO and immediately ordered this add-on. When testing the control from HA, I noticed that the unit cannot be turned off, nor can it be sent to 0%.

However, now I would like to be able to switch the unit off from HA in the event of an environmental alarm.

Is this possible, or is this not technically possible with the Itho fan unit, and is it better to switch off the power supply remotely?

arjenhiemstra, 2024-01-13:
Pls have a look at this wiki:

https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan#help-fan-wont-go-lower-than-ie-30

dzjr, 2024-01-13:
Thank you, I clearly overlooked that

I have made the adjustment and can now switch the ITHO to 0% or off.

Thank you for the cool add-on.

closed 2024-01-17 by arjenhiemstra