arjenhiemstra/ithowifi #225: HA: MQTT - TypeError: unsupported operand type(s)

roelvandermade, 2024-01-14:
Using firmware version: 2.7.0-beta1
Hardware: CVE-S ECO HP
HA: 2024.1.3

I noticed mqtt errors in my HA log and am missing some working functionality in HA now, probably due to this. Unsure if this is a HA configuration or mqtt messages from the addon?

```
2024-01-14 12:24:01.051 ERROR (MainThread) [homeassistant.components.mqtt.models] TypeError: unsupported operand type(s) for /: 'str' and 'float' rendering template for entity 'fan.nrg_itho_db0c_fan', template: '{{ (value / 2.55) | round | int }}'
2024-01-14 12:24:01.177 ERROR (MainThread) [homeassistant.util.logging] Exception in percentage_received when handling msg on 'itho/state': '20'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 43, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 537, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/fan.py", line 379, in percentage_received
    rendered_percentage_payload = self._value_templates[ATTR_PERCENTAGE](
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 258, in async_render_with_possible_json_value
    raise exc
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 246, in async_render_with_possible_json_value
    self._value_template.async_render_with_possible_json_value(
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 746, in async_render_with_possible_json_value
    return _render_with_context(self.template, compiled, **variables).strip()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2305, in _render_with_context
    return template.render(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
TypeError: unsupported operand type(s) for /: 'str' and 'float'

2024-01-14 12:24:01.178 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: offline
2024-01-14 12:24:01.179 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.state }}'
2024-01-14 12:24:01.185 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: offline, template: {{ value_json.state }})
2024-01-14 12:24:01.229 INFO (MainThread) [custom_components.tesla_custom.teslamate] Un-subbing from all MQTT Topics.
2024-01-14 12:24:01.229 INFO (MainThread) [custom_components.tesla_custom.teslamate] Un-subbed from all MQTT Topics.
```

arjenhiemstra, 2024-01-15:
The value on the itho/state topic should be an integer but seems to be a string in this case.
You cannot divide strings and that is why Home Assistant is giving the type error.

I'll change the template to first cast the input to an int and then do the calculation.
ie. template: '{{ (value / 2.55) | round | int }}
will become
template: '{{ (value | int / 2.55) | round | int }}

Thanks for reporting this.

roelvandermade, 2024-01-15:
> The value on the itho/state topic should be an integer but seems to be a string in this case. You cannot divide strings and that is why Home Assistant is giving the type error.
>
> I'll change the [...]

That makes sense indeed, thanks for the follow-up!

arjenhiemstra, 2024-01-17:
Should be fixed with https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.7.0

closed 2024-01-17 by arjenhiemstra