arjenhiemstra/ithowifi #226: [Request] Amber heat pump support

FeikoJoosten, 2024-01-22:
Hi there, couple months back an [Amber heat pump](https://www.ithodaalderop.nl/nl-NL/professional/product/03-00764) was installed in our house. This heat pump has a control module which doesn't have any form of internet access. I did notice the back of the screen has some usb ports, a serial port and an rj45 port available. Which may be usable for exposing access? The PCB itself also seems to have some available headers. I was wondering if there's a possibility to add support for the Amber heat pump?

Here are some pictures of the PCB and the back of the screen from the control module.
[image]
[image]
[image]
[image]

FeikoJoosten, 2024-01-22:
The software does have a setting where you can enable wifi monitoring, but for this you either need an wifi module or hard wire it using ethernet. But I'm not sure what that would then expose as I haven't tried this yet.

arjenhiemstra, 2024-01-26:
This looks like the one here?
https://gathering.tweakers.net/forum/list_messages/2165440/0

the RJ45 connection of my add-on is not ethernet so that is not an option. I see SCL and SDA connectors on the PCB though. That is I2C and also the protocol that the add-on uses. If this is the same protocol as this heatpump cannot be said until I have access to such a device and can analyse the bus.

FeikoJoosten, 2024-01-26:
I agree, does indeed look somewhat similar. 

If I can help with anything please let me know! 
I’m also already happy if I can have read-only access to the heat pump. Mainly want the statistics available in HomeAssistant haha.

FeikoJoosten, 2024-01-26:
According to this post they use the same protocol https://gathering.tweakers.net/forum/list_message/75784064#75784064

arjenhiemstra, 2024-01-26:
I believe they are talking about modbus, which is something different and not related to the I2C interface of other itho devices

FeikoJoosten, 2024-01-26:
Ah I meant it more as in the indoor control unit is the same for both the Amber and the HP-S 95

FeikoJoosten, 2024-05-31:
@arjenhiemstra is there something I can do to help out with this? I'd love to be able to monitor and possibly control my heatpump.

tmbreuker, 2025-05-16:
@FeikoJoosten i have a similar setup and look for possible HA integration. Were you able to set this up?