arjenhiemstra/ithowifi #227: api.html?get=currentspeed not updating

Fanman76, 2024-01-29:
api.html?get=currentspeed is not updating value upon rampup  of the CVE Eco fanbox.
Issue started after updating to current firmware. Commands to set speed seem to work, but the status of the fan(speed) is not updating.
api.html?get=ithostatus is correct. Blue bar on webpage is also not moving on fan speed increase.

Config version: 005
Itho I2C connection status: connected
File system: 28672 bytes used / 131072 bytes total

System memory usage:
CC1101 task memory: 1476 bytes free
MQTT task memory: 3718 bytes free
Web task memory: 2848 bytes free
Config and Log task memory: 3252 bytes free
SysControl task memory: 3488 bytes free
Loop task memory: 1480 bytes free

Itho device type: CVE-Silent
Itho hw version: 40
Itho fw version: 21
Manufacturer: 0

arjenhiemstra, 2024-01-29:
how are you controlling the speed of the fan?
The dan can increase its speed in 2 ways, pwm2i2c speed control and (virtual) remote speed control.
more info: https://github.com/arjenhiemstra/ithowifi/wiki/Controlling-the-speed-of-a-fan

the pwm2i2c speed is represented on get=currentspeed, get=ithostatus is a representation of values directly from the itho firmware itself.

Fanman76, 2024-01-30:
I increase the speed by virtual or RF remote. In the old firmware the blue slider would go up to mach the set speed, and the api command get=currentspeed would give me the actual set speed for the fan. Now both the blue bar and the currenspeed stay on there base value (20).

arjenhiemstra, 2024-01-30:
The code and the value it uses didn't change since at least 2.4.4, and that change was probably the change from sprintf to snprintf.

[image]

I think something else must be the reaseon. As said,  get=currentspeed is the representation of the last pwm2i2c command speed setting and that has always been the case. (And so far as I can test also still the case).
Remote commando's are only represented on get=currentspeed if the are connected to the CC1101 RF module (then remote commands are translated to pwm2i2c commands)

Could you downgrade firmware and reproduce a working config/?

Fanman76, 2024-01-30:
I'll try that tonight when i'm home again.

closed 2024-04-09 by stale[bot]