arjenhiemstra/ithowifi #228: [Request] AutoTemp fw 13 support

BakkerA, 2024-02-05:
Hi there. I think I'm on a quite new version of the autotemp firmware. Is it possible to get support for this one?
The add-on is not showing messages right now due to a high fw number.

jasperslits, 2024-02-06:
Same here with different house now:

Itho device type: AutoTemp
Itho hw version: 51
Itho fw version: 13
Manufacturer: 0

arjenhiemstra, 2024-02-11:
Should be fixed in https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.7.1

jasperslits, 2024-02-11:
I have empty Itho status.

Query format: 80 82 A4 00 01 78 00 00 00 00 00 00 92 92 00 12 92 92 00 12 92 92 00 12 92 92 00 12 92 92 00 12 92 92 00 12 92 92 00 12 92 92 00 12 92 92 00 12 92 92 00 12 92 92 00 12 92 92 00 12 0C 0C 0C 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 10 10 10 10 10 10 10 10 10 10 92 10 10 10 10 10 10 10 10 10 10 10 10 10 10 20 F7

Query status
Result: 80 82 A4 01 01 B8 03 02 00 00 00 00 08 48 08 34 00 00 00 07 8D 07 08 00 00 00 07 91 07 08 00 00 00 07 C8 07 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 64 64 64 64 64 00 00 00 00 00 00 00 64 64 64 64 64 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 10 00 00 00 00 00 00 00 00 1F 00 3F 00 00 03 84 01 B2 00 00 00 99 00 E9 00 0E 00 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 59

arjenhiemstra, 2024-02-11:
stange, I cannot explain it. Just an emtpy page? No errors nothing? 
Are the settings also not available?

jasperslits, 2024-02-11:
It seems to be OK now. I restarted the non-CVE multiple times. The settings page worked from the beginning and syslog did not show errors.

BakkerA, 2024-02-11:
> Should be fixed in https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.7.1

Thanks Arjen! Seems to be working. Spider data is send to HA now.

arjenhiemstra, 2024-02-11:
Great! Thanks for testing!

closed 2024-02-24 by arjenhiemstra