arjenhiemstra/ithowifi #232: Error on updating System Settings

Dixet, 2024-02-16:
**Describe the bug**
On the "System Settings" page the save button will report "error: Activated settings input value is not a valid JSON array!" and exits the screen without saving. 
Even without changing anything on the page this error occurs

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'System Settings'
2. Click on 'Save'
4. See error

**Expected behaviour**
Save the settings without error

**Device information**
- Firmware version 2.7.1
- Hardware revision NON-CVE 1
- Type/model number Itho : WPU 5G
- CC1101 RF module enabled no

**Smartphone (please complete the following information):**
 - Device: Xiaomi 12
 - OS: Android 12
 - Browser Vivialdi

arjenhiemstra, 2024-02-16:
I'm not able to reproduce this. What value is in the input field of setting "Activate settings indexes"?

Dixet, 2024-02-16:
"Activate settings indexes" is blank. 

I get the same message on both modules I have, one on WPU5G and one on Autotemp after updating to 2.7.1

arjenhiemstra, 2024-02-16:
blank is not a valid array. I apparently removed the default setting by accident.  The default config is to have at least "[]" in there (without quote marks).

Please fill in "[]" and then try again.

I'll update it in the next release, thanks for reporting this!

Dixet, 2024-02-16:
Check, that works! Thanks for the quick replies

arjenhiemstra, 2024-02-16:
I'll leave this open for now as guide to other users if they experience the same issue
Fixed in 2.7.2
https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.7.2

closed 2024-05-21 by stale[bot]