arjenhiemstra/ithowifi #233: Cannot switch off AP mode

vharten, 2024-02-19:
**Describe the bug**
After firmware upgrade to 2.7.2, AP mode suddenly went on. I try to switch it off by setting the AP timeout to zero, but it remains in Active state, even after a reboot.

**To Reproduce**
Happens always.

**Expected behaviour**
AP mode switched off after setting AP timeout to zero

**Screenshots**
[image]

**Device information**
- Firmware version 2.7.2
- Hardware revision cve 2
- Type/model number Itho (on sticker outside the box)
- CC1101 RF module enabled no
- In case of wifi issues, wifi access point make and model: Meraki MR33,42

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows
 - Browser [e.g. chrome, safari] Edge
 - Version [e.g. 22] latest

arjenhiemstra, 2024-02-20:
Thanks for spotting this! I found the issue:
https://github.com/arjenhiemstra/ithowifi/blob/7fa822487636c403f4896e97c7f00b88a85b61ea/software/NRG_itho_wifi/main/tasks/task_syscontrol.cpp#L553-L558

The order is wrong here. setupWiFigeneric  call should be after config load...
This was apparently masked by an immediate shutdown of the AP code somewhere else in the code but that is not triggered anymore in 2.7.2 due to some other changes.

I'll fix the issue and release 2.7.2 probably later today.

closed 2024-02-20 by arjenhiemstra

vharten, 2025-04-07:
Hoi Arjen,

 

Ik gebruikt de MQTT data van de WiFi module voor de Itho voor Home Assistant. Wat ik mis is de fan speed, die zit wel in de API. Is het mogelijk dat die toegevoegd wordt. Ik ben niet zo handig met Github, vandaar op deze manier.

 

Alvast bedankt,

Peter