arjenhiemstra/ithowifi #238: [Feature Request] Support for WPU Version 41

RobBie1221, 2024-03-12:
First of all, kudos for the very nice prints and code, I use them on my WPU as well as all my AutoTemp units!

Today, they exchanged my heatpump. The print used to work perfectly, but they bumped me to hw version 83, fw version 41.
It seems from wpu.h that up to fw version 37 is currently supported.

Obviously, I'd like to get support in for fw version 41.
I can add some effort, but need some guidance. It seems I need to extend wpu.h? Could I test this by locally running build_script.py from Python to generate a binary?
How would I be able to determine if they added / changed some settings / status registers? Do I need to enable sniffing?

arjenhiemstra, 2024-03-12:
Thanks for you this issue. I see that there is indeed a new version in the servicetool.

>How would I be able to determine if they added / changed some settings / status registers? Do I need to enable sniffing?

Unfortunately there is no easy way to get these labels out of the tool, for starters because all labels are in Dutch... 
I'll work on it; maybe @tomkooij has an idea how to create this file in an easier way?

RobBie1221, 2024-03-12:
Dutch is not an issue ;-) Do you need an account for the service tool to be able to see the labels?

Theoretically, if I would build a new version for myself to test where I copy the status registers from version 37, could that work?
Then at least I can have monitoring again.

arjenhiemstra, 2024-03-12:
Ok, the status label are easy, not a lot of changes, commit https://github.com/arjenhiemstra/ithowifi/commit/aa65d7bf8259f7d7a1e6d1f7f024ba8f8bdd4f69

settings labels need a bit more work

arjenhiemstra, 2024-03-12:
> Dutch is not an issue ;-) Do you need an account for the service tool to be able to see the labels?
> 
> Theoretically, if I would build a new version for myself to test where I copy the status registers from version 37, could that work? Then at least I can have monitoring again.

if you pull this repo and build it you should have status labels
 the service tool is free and downloadable from the itho website

arjenhiemstra, 2024-03-12:
I theory the settings should be an easy fix as well. There is one change on an existing index (228) and there are 10 additions.
But unfortunately I have some issues with the script to generate the .h header files, I' getting a totally different result than currently is in de firmware...

RobBie1221, 2024-03-12:
Thanks for the fast response!

Took me some time to build, commit https://github.com/arjenhiemstra/ithowifi/commit/aa65d7bf8259f7d7a1e6d1f7f024ba8f8bdd4f69 doesn't actually build with me (gives 2 errors regarding type conversion).

I was able to build the change of this commit on top of 2.8.0. I now see status data again!

arjenhiemstra, 2024-03-12:
> Thanks for the fast response!
> 
> Took me some time to build, commit [aa65d7b](https://github.com/arjenhiemstra/ithowifi/commit/aa65d7bf8259f7d7a1e6d1f7f024ba8f8bdd4f69) doesn't actually build with me (gives 2 errors regarding type conversion).
> 
> I was able to build the change of this commit on top of 2.8.0. I now see status data again!

Hmm, your correct. I accidentally checked in two files extra... trying to do too many things at once...

I'll fix that

here is a reference to the script in use to generate the files
https://github.com/arjenhiemstra/ithowifi/issues/49#issuecomment-971422213

But somehow its not working anymore, need to check this one into version control as well cleary

RobBie1221, 2024-03-12:
Thanks for the hint towards the script. I just manually opened the par file from the tool, still trying to understand how it's all built together.

For now at least my monitoring is running again. I'll see if I can get the script working to also get parameters in for version 41.

arjenhiemstra, 2024-03-12:
Here is some useful info about the par files and how to download them:
https://github.com/pommi/python-itho-wpu

and here is the "latest" version of the script. I'm far from an PHP expert and as said, should have better kept track of things using vc. If you can help out, that would be great!

```PHP
<?php

$lang	 = 'NL';
//$lang	 = 'GB';
//$lang	 = 'other';
$product = 'WPU'; //(cve14, cve1B, demandflow, autotemp, hrueco, hru200, hru250-300, hru350, WPU)
$dbname  = 'wpu.db';
$pwlevel = 500;

//use versionlist instead of DB query
//cve14
//$versions = [1,2,3,4,5,6];
//cve1B
//$versions = [6,7,8,9,10,11,17,18,20,21,22,24,25,26,27];
//demandflow
//$versions = [1,3,4,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21];
//autotemp
//$versions = [1,2,3,4,5,6,7,8,9,10,12,13];
//hrueco
//$versions = [3,4,6,7,8,10,11,12];
//hru200
//$versions = [12,13,14,15,17,18,19,20,21,25,26,27];
//hru250-300, $lang	 = 'GB';
//$product = 'HRU250_300';
//$versions = [1,2,3,4,6,7,8,9,10];
//hru350
//$versions = [1,3,4];
//wpu & set $verlist to TRUE
$versions = [];
$verlist = TRUE;

$db = new SQLite3($dbname);

$dl = [];

if ($verlist == TRUE) {
	$q = 'select DataLabel from VersieBeheer';
	$res = $db->query($q);
	
	// 	All datalabel numbers in an array
	
	while ($row = $res->fetchArray()) {
		$dl[$row['DataLabel']] = [];
	}
}
else {
	
	foreach ($versions as $value) {
		$dl[$value] = [];
	}
	
}

if($lang == 'NL') {
	$col_unit = 'Eenheid_NL';
	$col_title = 'Tekst_NL';
	$col_name = 'Naam';
}
else if($lang == 'GB') {
	$col_unit = 'Eenheid_GB';
	$col_title = 'Tooltip_GB';
	$col_name = 'Naam';
}
else {
[... 221 lines cut ...]

print("\n");

// Display the relevant labels numbers per version
foreach($dl as $labelnr=>$nameArr)
{
	$fieldseq = [];
	foreach($nameArr as $name=>$id) {
		$n = $labelsequence[$name];
		$fieldseq[] = array_keys($n)[0];
	}
	printf("const uint16_t itho_%ssetting%d[] { %s,999};\n",$product,$labelnr,join(",",$fieldseq));
}

print("\n");

print("const char* itho{$product}SettingsLabels[] =  {\n");
// Flash messages
foreach($f as $pos => $flashlabel) {
	
	printf("    \"%s\",\n",$flashlabel);
}
print("};\n");

```

arjenhiemstra, 2024-03-12:
Looking at my own earlier reply in the mentioned issue:
>Hmm, in the DB apparently not all data info is sorted the same way...

It probably has something to do with the data alignment (rows in the DB do not appear to be stored in index order), the script does not correct for that

tomkooij, 2024-03-13:
I have a tool that needs cleaning up that does just this...

Shall I dig into this and add the setting labels for v41, or is this already fixed by @RobBie1221 ?

RobBie1221, 2024-03-13:
I didn't fix anything yet, I'm still in a learning curve ;)

If you can do this that would be perfect.

RobBie1221, 2024-03-13:
If I look in the database, `parameterlijst_v41` is inserted in the database between v4 and v5, which probably means scripts parse v41 before v5, v6 etc. This is I think also what @arjenhiemstra means by correction for ordering?

If this is the case, I wonder how parsing has been done with scripts before, because e.g. v37 is also inserted before v4...

I can make a script which parses from v1 to v41 (first to last), but that would also maybe change `wpu.h` quite a bit?

RobBie1221, 2024-03-13:
I'm debugging`convert-itho-db.py` script, but something goes seriously wrong with this script. From `Parameterlijst_v17` onwards, it is not able to parse the `Index`. Some get a strange number, most get `nan`

[image]

tomkooij, 2024-03-13:
@RobBie1221 I just read the tables from the Access database (parameter file). Using the code in `convert-itho-db.py`.

I have two problems which prevent me from auto-generating much op `wpu.h`:
 - The english descriptions from the .mdb are quite different from `wpu.h`. I cannot recreate them from the parameter file. 
 - The settings are really different between v27 and v31 in `wpu.h`. Many of the settinglabels  are in the list twice. https://github.com/arjenhiemstra/ithowifi/blob/master/software/NRG_itho_wifi/main/devices/wpu.h#L28

The StatusLabels seems to autogen just fine (except from different descriptions, but the order is an exact match)

I'll upload my python notebook to GitHub so you can have a look

RobBie1221, 2024-03-13:
I also noticed StatusLabels are different in sense of descriptions, this probably has to do with different descriptions. Some StatusLabels have among versions the same `Naam` but different `Tekst_NL` and `Tekst_GB`. It seems to depend on order of parsing which label ends up in the header.

I'm trying to wrap my head around this:
https://github.com/pommi/python-itho-wpu/blob/7048014a223431d9024f0c5fca820f5ad146e19c/convert-itho-db.py#L55-L58

That should simply fetch the `Index` (among other keys) but somehow for `Parameterlijst` 17 and up this does not result in indices. Can't figure out what is wrong here.

arjenhiemstra, 2024-03-13:
> If I look in the database, `parameterlijst_v41` is inserted in the database between v4 and v5, which probably means scripts parse v41 before v5, v6 etc. This is I think also what @arjenhiemstra [...]

Also if I order the versions ASC the result is the same, also with old db files unfortunately

arjenhiemstra, 2024-03-13:
> @RobBie1221 I just read the tables from the Access database (parameter file). Using code copied from `convert-itho-db.py`.
> 
> I have two problems which prevent me from auto-generating much of `wpu.h`:
> 
> * The english descriptions from the .mdb are quite different from `wpu.h`. I cannot recreate them from the parameter file.
> * The settings are really different between v27 and v31 in `wpu.h`. Many of the settinglabels  are in the list twice. https://github.com/arjenhiemstra/ithowifi/blob/master/software/NRG_itho_wifi/main/devices/wpu.h#L28
> 
> The StatusLabels seems to autogen just fine (except from different descriptions, but the order is an exact match)
> 
> I'll upload my python notebook to GitHub so you can have a look: https://github.com/tomkooij/itho_parameters/blob/master/parse_parameters.ipynb
> 
> (This is a "work in progress" notebook, I have not cleaned it up)

I think I manually translated the labels because no english version was available. Some other devices have english labels and some luckily don't get updated as often :)

RobBie1221, 2024-03-13:
Just checking here, I'm now manually reading and trying to interpret the v37 parameter list. Via Access I can see the following:

[image]

Here:

https://github.com/arjenhiemstra/ithowifi/blob/23b57b28fe74c5ad1da1fd70d228a89984759f52/software/NRG_itho_wifi/main/devices/wpu.h#L31

It seems that index 238 refers to position 409 from ithoWPUSettingsLabels.

Position 409 should be:

https://github.com/arjenhiemstra/ithowifi/blob/23b57b28fe74c5ad1da1fd70d228a89984759f52/software/NRG_itho_wifi/main/devices/wpu.h#L433

Right? Or am I seeing this wrong?
Seems this should point to 9?

https://github.com/arjenhiemstra/ithowifi/blob/23b57b28fe74c5ad1da1fd70d228a89984759f52/software/NRG_itho_wifi/main/devices/wpu.h#L43

arjenhiemstra, 2024-03-13:
> Just checking here, I'm now manually reading and trying to interpret the v37 parameter list. Via Access I can see the following: [...]

You should read it as follows:
setting in database for version 37 let take 228
then itho_WPUsetting37[228] = 409

ithoWPUSettingsLabels[409] =  "Log interval (sec)"

which indeed is in the db:

228.0 | 228.0 | P_Data_Interval | dataLogInterval | 5.0 | 0x10 | 300.0 | 5.0 | loginterval tijd | tijd tussen 2 metingen
-- | -- | -- | -- | -- | -- | -- | -- | -- | --

</body></html>

RobBie1221, 2024-03-13:
Ok, thanks. My reasoning was correct, my indexing in ithoWPUSettingsLabel not...

Nevertheless, this overlaps with "9" in the tabel. Somehow it generated a new key in ithoWPUSettingsLabel. 
For example v2:

[image]

There is potential to save memory :)

tomkooij, 2024-03-14:
Okay, My understanding of this as of now:

 - `wpu.h` was generated from the parameter files but with hand-made translations of the names/labels. These cannot be auto generated and will change when autogenerated. This is not really a problem for settinglabels but it is a concern for statuslabels as these appear as keys in MQTT messages. They cannot change without breaking this for users. (This would be unacceptable)
 - I do not have a database with the hand translated stuff 
 - Somewhere in conversion the ordering of tables is messed up for settings. `Parameterlijst_V31` was inserted/used after 2 and before 4 resulting in many duplicate keys (with different translations/slugs) starting from firmware 31.

My proposed solution: Auto-generate `wpu.h` from the current parameter files without breaking everything (so keep hand translated statuslabels). I can tidy up my python code to do this. That's a bit of work now, but would save a lot of work when a new firmware appears. It would especially save (me) from trying to figure this out all over again.
 - I can generate the StatusLabel indices one-to-one from the parameters files. I can include a lookup-table for the hand-made translations as they map one-to-one. The hand made translation can thus be auto inserted with high confidence.
 - Re generate the SettingLabels indices and change all labels to be auto generated. This would remove the 150+ duplicates. The indices from v17 change a little bit (one two indices change) and are totally different from v31.

The solution above would ofcourse introduce the chance of broken settings (mis-aligned labels) but I can at least check it for version 37 myself and this should give a higher confidence in the correctness.

@arjenhiemstra Looking at your php-script above, I don't think the script can do the above mentioned things, right? (I cannot save some work by trying to figure out PHP?).
Would regenerating SettingLabels indices be acceptable?

To get an idea of the difference in settinglabel incides, view the diff: https://www.diffchecker.com/wWbF1Z8F/

tomkooij, 2024-03-14:
O... I've thought a little bit about this, but the solution above is not a real solution: It would work for `wpu.h`, but it it should work for all of `devices/'  (It's easy enough to add that, but perhaps just fixing the PHP code is easier)

Anyway, I'll finish the python script. Propose a new `wpu.h` with the duplicates removed and we can work from there. If you (Arjen) decide not to use it and stick with the PHP code, that's fine as well.

arjenhiemstra, 2024-03-14:
> Okay, My understanding of this as of now:
>
> * `wpu.h` was generated from the parameter files but with hand-made translations of the names/labels. These cannot be auto generated and will change when [...]

If the generator script works correctly, new version additions should only result in labels being added to the end of the array. Only these new labels need translation if no english version is available. This worked up until now with the wpu settings. The datalables I checked in have been created in exact that way, there is a bit of manual work involved but this is minimal.

> * I do not have a database with the hand translated stuff

You do have a "database", this is the current wpu.h ;-)

> * Somewhere in conversion the ordering of tables is messed up for settings. `Parameterlijst_V31` was inserted/used after 2 and before 4 resulting in many duplicate keys (with different translations/slugs) starting from firmware 31.
> 

Proper sorting of the tables before  generating the label should fix this (also for future releases). Because then the order in wich labels are placed in the array with the script are always the same and new additions should end up at the bottom.

> My proposed solution: Auto-generate `wpu.h` from the current parameter files without breaking everything (so keep hand translated statuslabels). I can tidy up my python code to do this. That's a bit [...]

Statuslabels are correct and already auto generated, only the translation is manual. No rework needed here I would say.

> * Re-generate the SettingLabels indices and change all labels to be auto generated. This would remove the 150+ duplicates. The indices from v17 change a little bit (one two indices change) and are totally different from v31. Diff:  https://www.diffchecker.com/wWbF1Z8F/
> 
> The solution above would ofcourse introduce the chance of broken settings (mis-aligned labels) but I can at least check it for version 37 myself and this should give a higher confidence in the correctness.
> 
> @arjenhiemstra Looking at your php-script above, I don't think the script can do the above mentioned things, right? (I cannot save some work by trying to figure out PHP?). Would regenerating SettingLabels indices be acceptable?

I think we should indeed use an updated script to autogenerate the settings labels (at first it is fine when this is in Dutch). Then we need to check of the generated settings labels match up with the settings for that version, it seems we can easily check that with your devices for version 37 and 41, some manual and/or automated test can be generated for the other versions.
If we have I correct settings map, I can redo the translation once more, after that the process should be repeatable and new additions end up being added to the end and manual work is minimal again.

Other option is to manually add it now for version 41 and hope there will be no other wpu firmware versions but I think that's not the best idea ;-)
I dont't have real knowledge of php, @jasperslits was the original author of the script, maybe he has a quick idea how to approach this?

arjenhiemstra, 2024-03-14:
> Okay, I can do most of this. I’ll submit a PR.

and PHP or Python, either is fine. I think Python might be a bit easier to maintain

jasperslits, 2024-03-14:
I agree with Pythong being easier. I created that PHP script for a quick headstart of WPU support, didn't expect it would be considered after 2? years again.

tomkooij, 2024-03-15:
@RobBie1221 #240 seems ready (and tested on fw37 by myself). Can you test?

(I guess you can build it yourself, please do, or use this: https://www.dropbox.com/scl/fi/9hnr8whhgetirn74o51hz/nrgitho-v2.8.0-pr240.bin?rlkey=kq86zy0o9o42zwkolcf0nth85&dl=0)

RobBie1221, 2024-03-15:
I can build myself. I will test it this evening. Thanks for the effort!

jelmerderonde, 2024-09-09:
Hi, I have an Itho WPU 5G here at home, hardware version 83, firmware version 41. I have a board with hardware revision: NON-CVE 1 and firmware 2.9.0-beta1.

The board came with firmware 2.8.0 and I did not see any status values. I updated to 2.9.0-beta1 and briefly saw some values and after configuring mqtt some values made it to Home Assistant as well.

However after a couple of hours everything stopped working. The controller is reporting everything is well and I don't see any I2C issues, however no data.

I would love some pointers to help fix this and maybe improve the firmware as well!

jelmerderonde, 2024-09-26:
As said in the previous comment, I have WPU 5G running firmware 41 at home. I'v now got the message " i2c bus stuck - please power cycle the Itho unit", with the following logs:

```
QUERY_STATUS	25623988	5513	I2C_ERROR_SCL_LOW	I2C_NOK
QUERY_STATUS	25563975	2		I2C_OK
QUERY_STATUS	25503964	2		I2C_OK
QUERY_STATUS	25443949	2		I2C_OK
QUERY_STATUS	25383934	2		I2C_OK
QUERY_STATUS	25323926	2		I2C_OK
QUERY_STATUS	25263919	2		I2C_OK
QUERY_STATUS	25203916	2		I2C_OK
QUERY_STATUS	25143892	2		I2C_OK
QUERY_STATUS	25083887	2		I2C_OK
QUERY_STATUS	25023871	2		I2C_OK
QUERY_STATUS	24963867	2		I2C_OK
QUERY_STATUS	24903852	2		I2C_OK
QUERY_STATUS	24843838	2		I2C_OK
QUERY_STATUS	24783822	2		I2C_OK
```

Louwke80, 2024-12-13:
> Hi, I have an Itho WPU 5G here at home, hardware version 83, firmware version 41. I have a board with hardware revision: NON-CVE 1 and firmware 2.9.0-beta1.
>
> The board came with firmware 2.8.0 and I [...]

Similar issue here as well. Data is being transferd through mqtt as I can see the updates in mqtt Explorer.
On the status page itself I see the values, but they do not update at all. Utc timer for example is stuck on the same value for hours. After a x amount of time some values drop to zero.
After a reboot of the device it shows update values, but get stuck right away. 
I2C status is being reporter as connected and no issues liggend in syslog. 

Currently running firmware 2.9.0 beta 6
Hardware is version non cve 1

Itho device type: Heatpump
Itho hw version: 83
Itho fw version: 41
Manufacturer: 0

arjenhiemstra, 2024-12-13:
Thx for the info! If you go to the debug page and click the 'Send Status Query' button, are the values updated after that?

Louwke80, 2024-12-13:
I tested the Query Status and it shows Failed as result.
After that I tried Query device type, it did give a result.
Query Status Format also returned failed.
Query 2410, did give a result

arjenhiemstra, 2024-12-13:
Ok, that is valuable information!

Query device type, Status Format and Status all use the same i2c function to get their data. As Device Type seems to be working, that indicates the underlying i2c code of the add-on is still working fine.

I now suspect a reboot of the add-on does not fix the issue? And a power cycle of the WPU? 
Perhaps the issue is within the itho firmware...

Louwke80, 2024-12-14:
Yes sadly a reboot of the add on only refreshes the status values once.
A powercycle on the wpu didn't solve it either.

Louwke80, 2024-12-15:
I have downloaded and tried the firmware. On syslog it's shows send_i2c_query ended without result about every 30s.

Seems the code still tries to reuqest every 30s as set as update frequency.

arjenhiemstra, 2024-12-15:
Ok thanks!
Lets dig a little bit deeper in the reply. Apparently there is something in the reply but somewhere a check of this reply fails.

I've changed the logging to hopefully get this part:

```C
  uint16_t opcode = (i2c_command[2] << 8 | i2c_command[3]);
  if (len > 1 && receive_buffer[len - 1] == checksum(receive_buffer, len - 1) && check_i2c_reply(receive_buffer, len, opcode))
  {
    return len;
  }
  else
  {
    E_LOG("send_i2c_query failed - len:%d, rb[l-1]:%d, chk:%d, chkr:%d" , len, receive_buffer[len - 1], checksum(receive_buffer, len - 1), check_i2c_reply(receive_buffer, len, opcode));
    std::string log = i2cbuf2string(receive_buffer, len);
    E_LOG("receive_buffer:%s", log.c_str());
  }

  return 0;
```
firmware file:
[nrgitho-dev-v2.9.0-beta6-wputest2.bin.zip](https://github.com/user-attachments/files/18140553/nrgitho-dev-v2.9.0-beta6-wputest2.bin.zip)

Louwke80, 2024-12-15:
Thanks for the Quick replies.

Ive dikke the update and the following shows up in the syslog now

2024-12-15 15:12:32 E: receive_buffer:80 82 A4 01 01 D5 01 90 18 08 18 08 06 4C 09 00 0B 9C 09 8A 05 FA 09 72 09 D4 09 2C 00 B5 00 00 00
2024-12-15 15:12:32 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:220, chkr:1
2024-12-15 15:12:32 E: receive_buffer:80 82 A4 01 01 D5 01 90 18 08 18 08 06 4C 09 00 0B 9C 09 8A 05 FA 09 72 09 D4 09 2C 00 B5 00 00 00
2024-12-15 15:12:32 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:220, chkr:1
2024-12-15 15:12:02 E: receive_buffer:80 82 A4 01 01 D5 01 90 18 08 18 08 06 4C 09 00 0B 9C 09 8A 05 FA 09 72 09 D4 09 2C 00 B5 00 00 00
2024-12-15 15:12:02 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:220, chkr:1
2024-12-15 15:11:33 E: receive_buffer:80 82 A4 01 01 D5 01 90 18 08 18 08 06 4C 09 00 0B 9C 09 8A 05 F2 09 6E 09 D4 09 2C 00 B5 00 00 00
2024-12-15 15:11:32 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:232, chkr:1
2024-12-15 15:11:03 E: receive_buffer:80 82 A4 01 01 D5 01 90 18 08 18 08 06 4C 09 00 0B 9C 09 8A 05 F2 09 6F 09 D4 09 2C 00 B5 00 00 00
2024-12-15 15:11:02 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:231, chkr:1
2024-12-15 15:10:32 E: receive_buffer:80 82 A4 01 01 D5 01 90 18 08 18 08 06 4C 09 00 0B 9C 09 8A 05 F2 09 72 09 D4 09 23 00 B5 00 00 00
2024-12-15 15:10:32 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:237, chkr:1

arjenhiemstra, 2024-12-15:
Thanks! 

There seems to be a clear issue, I only don't know yet what causes it.
80 82 A4 01 01 **D5**

D5 hier indicated the length of the message from that point, which is 213 in decimal. There is not nearly that amount of data in the reply, which causes the checksum to fail discarding the message.

Question now is why does this happen. The buffer with a size of 512 should have more than enough space.

I'll dig a little deeper and try some code changes.

arjenhiemstra, 2024-12-15:
Ok, I've refactored the i2c_slave_receive function to simplify a few things and let it use lesser memory. 
In this updated version there is no temporary / extra global buffer used anymore as intermediate storage (why on earth did I do that for....) and the i2c timeout has been increased from 10ms to 200ms

Not sure if this helps though, I'm very curious what your logging will show (loggin is the same as the previous version)

Firmware file
[nrgitho-dev-v2.9.0-beta6-wputest3.bin.zip](https://github.com/user-attachments/files/18141156/nrgitho-dev-v2.9.0-beta6-wputest3.bin.zip)

Refactored code:
```C
size_t i2c_slave_receive(uint8_t *i2c_receive_buf)
{
  i2c_config_t conf_slave = {
      .mode = I2C_MODE_SLAVE,
      .sda_io_num = slave_sda_pin,
      .scl_io_num = slave_scl_pin,
      .sda_pullup_en = I2C_SLAVE_SDA_PULLUP,
      .scl_pullup_en = I2C_SLAVE_SCL_PULLUP,
      .slave = {
          .addr_10bit_en = 0,
          .slave_addr = I2C_SLAVE_ADDRESS,
          .maximum_speed = 400000UL,
      },
      .clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL, // optional
  };

  i2c_param_config(I2C_SLAVE_NUM, &conf_slave);

  i2c_driver_install(I2C_SLAVE_NUM, conf_slave.mode, I2C_SLAVE_RX_BUF_LEN, I2C_SLAVE_TX_BUF_LEN, 0);

  i2c_set_timeout(I2C_SLAVE_NUM, 0xFFFFF);

  i2c_receive_buf[0] = I2C_SLAVE_ADDRESS << 1;

  int len_rec = 0;

  len_rec = i2c_slave_read_buffer(I2C_SLAVE_NUM, &i2c_receive_buf[1], I2C_SLAVE_RX_BUF_LEN - 1, 200 / portTICK_PERIOD_MS);
  if (len_rec <= 0)
    return 0;

  i2c_slave_deinit();
  return len_rec + 1; // add 1 for the slave addr inserted at i2c_receive_buf[0]
}
```
Old code:
```C
size_t i2c_slave_receive(uint8_t *i2c_receive_buf)
{
  i2c_config_t conf_slave = {
      .mode = I2C_MODE_SLAVE,
      .sda_io_num = slave_sda_pin,
      .scl_io_num = slave_scl_pin,
      .sda_pullup_en = I2C_SLAVE_SDA_PULLUP,
      .scl_pullup_en = I2C_SLAVE_SCL_PULLUP,
      .slave = {
          .addr_10bit_en = 0,
          .slave_addr = I2C_SLAVE_ADDRESS,
          .maximum_speed = 400000UL,
      },
      .clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL, // optional
  };

  i2c_param_config(I2C_SLAVE_NUM, &conf_slave);

  i2c_driver_install(I2C_SLAVE_NUM, conf_slave.mode, I2C_SLAVE_RX_BUF_LEN, I2C_SLAVE_TX_BUF_LEN, 0);

  i2c_set_timeout(I2C_SLAVE_NUM, 0xFFFFF);

  i2cbuf[0] = I2C_SLAVE_ADDRESS << 1;
  buflen = 1;

  while (1)
  {
    int len1 = i2c_slave_read_buffer(I2C_SLAVE_NUM, i2cbuf + buflen, sizeof(i2cbuf) - buflen, 50 / portTICK_PERIOD_MS);
    if (len1 <= 0)
      break;
    buflen += len1;
  }
  if (buflen > 1)
  {
    for (uint16_t i = 0; i < buflen; i++)
    {
      i2c_receive_buf[i] = i2cbuf[i];
    }
  }
  i2c_slave_deinit();
  return buflen;
}
```

Louwke80, 2024-12-15:
No worries, happy to test.

Here is the log after a reboot

2024-12-15 18:40:03 E: receive_buffer:80 82 A4 01 01 D5 03 20 17 C3 17 D1 07 0F 08 AA 09 BB 09 07 06 DF 08 B8 09 2C 09 09 00 B6 00 00 00
2024-12-15 18:40:03 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:96, chkr:1
2024-12-15 18:39:33 E: receive_buffer:80 82 A4 01 01 D5 03 20 17 C3 17 D1 07 0F 08 A9 09 BB 09 07 06 DD 08 B8 09 2C 09 09 00 B5 00 00 00
2024-12-15 18:39:33 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:100, chkr:1
2024-12-15 18:39:04 E: receive_buffer:80 82 A4 01 01 D5 03 20 17 C3 17 D1 07 0F 08 AF 09 BB 09 07 06 DB 08 B9 09 2C 09 09 00 B5 00 00 00
2024-12-15 18:39:03 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:95, chkr:1
2024-12-15 18:38:33 E: receive_buffer:80 82 A4 01 01 D5 03 20 17 C3 17 D1 07 0F 08 AE 09 BB 09 07 06 DB 08 B9 09 2C 09 09 00 B5 00 00 00
2024-12-15 18:38:33 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:96, chkr:1
2024-12-15 18:38:03 E: receive_buffer:80 82 A4 01 01 D5 03 20 17 C1 17 D1 07 0F 08 AF 09 BB 09 07 06 DB 08 BA 09 2C 09 09 00 B6 00 00 00
2024-12-15 18:38:03 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:95, chkr:1
2024-12-15 18:37:53 I: I2C init: i2c sniffer enabled
2024-12-15 18:37:53 I: I2C init: QueryStatus
2024-12-15 18:37:52 I: I2C init: QueryStatusFormat - items:0
2024-12-15 18:37:52 E: receive_buffer:80 82 A4 00 01 86 92 92 92 92 92 92 92 92 92 92 92 92 92 92 0C 0C 0C 0C 92 10 0C 00 00 00 00 00 0C 0C 0C 0C 0C 10 92 92 00 00 00 00 92 92 10 90 00 00 10 0C 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 10 12 00 00 20 00 00 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C
2024-12-15 18:37:52 E: send_i2c_query failed - len:89, rb[l-1]:12, chk:199, chkr:1
2024-12-15 18:37:51 I: Hostname: nrg-itho-7644
2024-12-15 18:37:50 I: mDNS: started
2024-12-15 18:37:50 I: I2C init: QueryDevicetype - mfr:0x00 type:0x0D fw:0x29 hw:0x53
2024-12-15 18:37:50 I: Webserver: started
2024-12-15 18:37:50 I: MQTT: connected, System config: 1
2024-12-15 18:37:50 I: Setup: CC1101 RF module found, chip version: 0x14
2024-12-15 18:37:48 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2024-12-15 18:37:48 I: WiFi: connection successful
2024-12-15 18:37:40 I: wifi AP mode started
2024-12-15 18:37:39 E: Unable to set wifi disconnect
2024-12-15 18:37:37 I: I2C sniffer capable hardware: yes
2024-12-15 18:37:36 I: HW rev: NON-CVE 1, FW ver.: 2.9.0-beta6-wputest3
2024-12-15 18:37:36 I: Device UUID: 85ae0de0-ecc7-4a68-9a50-da7493d98734
2024-12-15 18:37:36 I: System boot, last reset reason: SW_RESET

Louwke80, 2024-12-16:
1 more today after a reset.
2024-12-16 19:01:27 E: receive_buffer:80 82 A4 01 01 D5 03 E8 16 BD 16 CB 06 0C 08 52 0A F6 08 F3 05 90 08 B8 09 85 08 14 00 B6 00 00 00
2024-12-16 19:01:27 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:200, chkr:1
2024-12-16 19:00:57 E: receive_buffer:80 82 A4 01 01 D5 03 E8 16 BD 16 CB 06 0B 08 52 0A F6 08 F3 05 90 08 B8 09 85 08 11 00 B6 00 00 00
2024-12-16 19:00:57 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:204, chkr:1
2024-12-16 19:00:28 E: receive_buffer:80 82 A4 01 01 D5 03 E8 16 BD 16 CB 06 0B 08 52 0A F7 08 F3 05 88 08 B8 09 85 08 11 00 B6 00 00 00
2024-12-16 19:00:27 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:211, chkr:1
2024-12-16 18:59:57 E: receive_buffer:80 82 A4 01 01 D5 03 E8 16 BD 16 CB 06 0B 08 52 0B 03 08 F3 05 88 08 AF 09 85 08 08 00 B6 00 00 00
2024-12-16 18:59:57 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:216, chkr:1
2024-12-16 18:59:27 E: receive_buffer:80 82 A4 01 01 D5 03 E8 16 BD 16 CB 06 0B 08 52 0B 07 08 F3 05 88 08 AF 09 86 08 08 00 B5 00 00 00
2024-12-16 18:59:27 E: send_i2c_query failed - len:33, rb[l-1]:0, chk:212, chkr:1
2024-12-16 18:59:17 I: I2C init: i2c sniffer enabled
2024-12-16 18:59:16 I: I2C init: QueryStatus
2024-12-16 18:59:16 I: I2C init: QueryStatusFormat - items:134
2024-12-16 18:59:16 I: Hostname: nrg-itho-7644
2024-12-16 18:59:16 I: mDNS: started
2024-12-16 18:59:15 I: Webserver: started
2024-12-16 18:59:14 I: MQTT: connected, System config: 1
2024-12-16 18:59:14 I: I2C init: QueryDevicetype - mfr:0x00 type:0x0D fw:0x29 hw:0x53
2024-12-16 18:59:14 I: Setup: CC1101 RF module found, chip version: 0x14
2024-12-16 18:59:12 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2024-12-16 18:59:11 I: WiFi: connection successful
2024-12-16 18:59:04 I: wifi AP mode started
2024-12-16 18:59:03 E: Unable to set wifi disconnect
2024-12-16 18:59:01 I: I2C sniffer capable hardware: yes
2024-12-16 18:59:00 I: HW rev: NON-CVE 1, FW ver.: 2.9.0-beta6-wputest3
2024-12-16 18:59:00 I: Device UUID: 85ae0de0-ecc7-4a68-9a50-da7493d98734
2024-12-16 18:59:00 I: System boot, last reset reason: SW_RESET
2024-12-16 18:58:55 I: Reboot requested

arjenhiemstra, 2024-12-16:
Thanks for testing! This means that the function "i2c_slave_read_buffer" does not work as expected or that the itho is sending bogus replies... 
I have been digging around a bit and found this somewhat disturbing description for the existence of the repo:
https://github.com/schveiguy/espi2cslave

I'll have a look if I can port this driver to the ESP32, I'll also compile the code with the latest SDK of Espressif and share that bin here, it might that some issues have been solved in the mean time.

arjenhiemstra, 2024-12-18:
Here is a version build with a newer SDK version. Lets see if it makes a difference!

[nrgitho-dev-v2.9.0-beta7-wputest.bin.zip](https://github.com/user-attachments/files/18188600/nrgitho-dev-v2.9.0-beta7-wputest.bin.zip)

jelmerderonde, 2024-12-19:
Chiming in here: I had some time so updated to the v.2.9.0-beta7 @arjenhiemstra uploaded here. Unfortunately this seems to have bricked the device. The red-led intermittently flashes very fast. I see the `nrg-itho-xxxx` network appear and disappear every few seconds, joining the network sometimes work but connection is broken before I can open the webinterface, because the device reboots.

I don't have the necessary cables or know-how to try and restore a regular version of the firmware unfortunately.

arjenhiemstra, 2024-12-19:
What type of add-on do you have? You should always be able to do a fail save reset and upload a new firmware to http://192.168.4.1/update

Here is a video how to perform the reset:

https://youtu.be/3sWclzq73n4?si=2-L3Kbfl32UTnubG

If you skip the last rest click the ad-on will be in fail save mode

jelmerderonde, 2024-12-20:
I have the Itho NON-CVE Wifi controller, HW rev: 1, Batch: 14-224. There appear to be two small buttons on the board. Which one should I press, how many times?

Louwke80, 2024-12-20:
Goodmorning,

I also tried to update tot the beta7 wputest. Sadly here also the same happend. Site not available. 

Tried the reset and it got connected again. Selecte firmware 6 again and pressed update. Sadly after the update it boots flashing red and after a minuten or so it shuts down. No activity anymore on any lights even the rj45 connector lights stop flashing and the wifi is also gone.

Power cycle doesn't work as the wifi comes on and off. Though getting it into failsafe flash mode seems to work. But after pressing update you will see red lights flashing and the rj45 lights blinking and eventually the lights all go out and it doesn't reboot.

arjenhiemstra, 2024-12-20:
Strange, I'm not sure why I'm not seeing this happening... sorry!

Maybe this picture helps to get the module through fail save.

[image]

janbredenbeek, 2024-12-20:
Unfortunately I have bricked my device too. It's HW v1.1 and it only has a Fail save button, no Reset.
Would repairing via a USB cable be an option? (the serial USB port seems to be working, it sends data to a terminal app).

Louwke80, 2024-12-20:
> Strange, I'm not sure why I'm not seeing this happening... sorry!
> 
> Maybe this picture helps to get the module through fail save.
> 
> ![fail save non-cve](https://private-user-images.githubusercontent.com/7045663/397728809-b4de21bf-a65d-4678-9c2d-9117969d6255.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ3MDEyMDgsIm5iZiI6MTczNDcwMDkwOCwicGF0aCI6Ii83MDQ1NjYzLzM5NzcyODgwOS1iNGRlMjFiZi1hNjVkLTQ2NzgtOWMyZC05MTE3OTY5ZDYyNTUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjBUMTMyMTQ4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGJkNTNhZDg0NDA5OWQ2ZTFkOTlmZGFhODk5YTg1MThiMjdiZDU3Zjk0MGIyZDlkODk2MTI2MzUyMWFlZjFhNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.xpjLd0ocB-EcwfOGSEePkkA-V4HiN-tAa0-xltPyIhA)

The picture works perfectly, helpfull for sure!

But after updating the lights stop so I reset the device and it reboot but shuts down after trying to connector to the wifi.
After this the failsafe mode works, but same result as above

arjenhiemstra, 2024-12-20:
:( bummer, this is the very first time this happens in all the time working on this module, very unfortunate and sorry for the inconvenience caused!

for the non-cve version, you could flash the module through the micro-usb interface. You need a working install of the flash tool **esptool**
Here is info how to get this from the Tasmota website:
https://tasmota.github.io/docs/Esptool/

With that installed you can use the upload script and rest of the files from this document:
[firmware_parts.zip](https://github.com/user-attachments/files/18213482/firmware_parts.zip)

Here is everything you need to reflash the module from scratch.

The upload script for MacOS and Linux:
```BASH
#/bin/bash
fwfile="firmware.bin"
porta=$(ls /dev/tty.usbserial-*)
baud="460800"
esptool="/location/of/esptool.py"

$esptool \
--chip esp32 --port $porta --baud $baud --before default_reset --after hard_reset write_flash -z \
--flash_mode dio --flash_freq 80m --flash_size detect \
0xe000 ./boot_app0.bin \
0x1000 ./bootloader.bin \
0x10000 ./$fwfile \
0x8000 ./partitions.bin
```
You need to either remote "**/location/of/**" or let it point to the installation is esptool of you system.

On Windows you can is the command prompt and put all the parts of the script in one row, ie.:
```SHELL
esptool --chip esp32 --port COM1 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 boot_app0.bin 0x1000 bootloader.bin 0x10000 firmware.bin 0x8000 partitions.bin
```

Where you need to change COM1 to the proper COM poort on your system.

This will flash the module back to 2.8.0, this method should always work. If this poses a too big of an effort, don't hesitate to contact me with you order details and I'll make sure you get a working module.

janbredenbeek, 2024-12-20:
It's working again, thanks.

Louwke80, 2024-12-21:
Same here usb tool install showed me alot, but manager to install all en it's back to version 2.8.0 again. Thanks.

Meanwhile I talked to my neighbour we are planning a test with my addon on his system, as he got the same wpu as me and his seems to work. So hopefully we can also try and figuren out of it is my wou causing the issue as well.

Louwke80, 2024-12-21:
Ok update on this one. After the USB update i flashed it to 2.9.0 beta 6 again. Seemed as if the update page was working this time. Set up mqtt again and some how it solved all my issues I had.

Why and how I still have got no clue at all, but I getting utc time update values etc.

arjenhiemstra, 2024-12-21:
Ok, thats really weird!

I've tested some time with the beta7 version and also on my itho cve I occasionally (a few times a day) see this error message: 
`send_i2c_query failed - len:33, rb[l-1]:0, chk:216, chkr:1`

It seems that sometimes the i2c function just reads 32 bytes from the i2c bus (33 in the log because the i2c address is manually added), I still have no real clue on what the reason is unfortunately...

daenny, 2025-01-16:
I also received my Itho module now. 
Apparently I also have a too new WPU:
```
Itho device type: Heatpump
Itho hw version: 83
Itho fw version: 41
```
I tried updating to 2.9.0beta10 now. But I do not get any values in the Itho settings (pressing retrieve does not do anything) and also nothing in the Itho status.

Is there anything I can do to help debugging?

WilcoSnel, 2025-05-15:
I have the below version as well. 

Itho device type: Heatpump
Itho hw version: 83
Itho fw version: 41

Should I try beta 12? And what can I do to help after?