arjenhiemstra/ithowifi #242: [Feature] provide FW update available of addon via MQTT

thyzzv, 2024-03-17:
**Describe the feature**
Would it be possible to provide the 'Firmware update available' of the add on itself via MQTT?
That way we wouldn't have to login to the add on itself to check for newer versions (or follow the GoT toic very precisely) 

**Expected behaviour**
Provide an 'fw_update_available' boolean on the mqtt status topic for the addon.
or a 'available firmware version' field

arjenhiemstra, 2024-03-18:
certainly possible. The json file could be parsed from github lets say once a day. Only downside is that connecting to github requires https client support in the firmware which is quite costly firmware size (adds about 180kb).
Maybe some other solution is possible...

thyzzv, 2024-03-19:
Ah I had not noticed that the firmware check is done locally in the browser. Not on the device
180kb is indeed a lot.

arjenhiemstra, 2024-12-12:
Since SSL support was needed for something else aswel, this feature has now been implemented with 2.9.0-beta5. The HA Discovery now includes a sensor with firmware information and besides that this information is also available on the MQTT API.

closed 2024-12-12 by arjenhiemstra