arjenhiemstra/ithowifi #243: Frost valve opening and worrying sound coming from it after connecting module

WVan, 2024-03-19:
**Describe the bug**
I recently purchased and installed the non-CVE module. But soon after starting the set up process I started hearing a worrying sound from the HRU 350 unit it was connected to. Also the frost valve opened. I think the sound is from the frost valve motor trying to open further after reaching the maximum, but that's just a hunch. I tried a full reset of the module, and recorded the behaviour on video (even before configuring the wifi), see attached below. Initially I had to disconnect the power from the HRU to make it stop (disconnecting the module didn't work), but after the reset of the module also disconnecting the module made the sound stop and the valve close after a few seconds.

Any idea what's causing this? It's quite unusable this way, and it doesn't sound like it's supposed to do this (and perhaps would damage the unit in the long run).

**To Reproduce**
Steps to reproduce the behaviour:
1. Attach module.
2. Wait a few seconds.
3. Frost valve opens and worrying noise comes from the unit.

**Expected behaviour**
Frost valve stays closed and no noise comes from the HRU unit.

**Screenshots**

https://github.com/arjenhiemstra/ithowifi/assets/1194696/8fe57fde-28a2-43ea-93b6-7ebc80352502

**Device information**
- Firmware version: 2.8.0
- Hardware revision: non-CVE rev 1.
- Type/model number Itho (on sticker outside the box): HRU ECO 350 HR
- CC1101 RF module enabled: yes and no (did a reset)

arjenhiemstra, 2024-03-19:
not sure what it is, there was one other user with the same kind of issues:
https://gathering.tweakers.net/forum/list_message/70327192#70327192
I've asked in the forum what was the issue and solution. I believe it was not firmware related.

WVan, 2024-03-20:
That seemed related to temperature sensors. The thing is that it was triggered for me only when the new module was connected. I tried some more things tonight and after a full format (not just the reset) it seems to be fixed now. Might there have been some setting that I could have accidentally changed during my initial setup that could trigger this kind of behaviour? Anyway, I guess it's solved for me at least. Thanks.

closed 2024-03-20 by WVan