arjenhiemstra/ithowifi #244: Adding a RF remote: ID error, please use HEX notation separated by ',' (ie. 'A1,34,7F')

joho500, 2024-03-30:
**Describe the bug**
I've added a RF remote which is recognized with the ID **C5,5,51**. When updating I get  the error:

`ID error, please use HEX notation separated by ',' (ie. 'A1,34,7F')`

With another remote of the same type the error didn't occur. The ID of that device is **51,17,20**.

**To Reproduce**
Steps to reproduce the behaviour:
1. Set IthoWifi in Learn mode
2. Add the RF remote by holding two buttons simultaneously
3. Select row of newly found remote
4. Press button **Update**

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
- Firmware version  2.8.0
- Hardware revision 2
- Type/model number Itho: CVE Silent
- CC1101 RF module enabled: yes

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Browser Firefox
 - Version 124.0.1

arjenhiemstra, 2024-03-30:
Thanks, I already noticed it and the fix is in the line for the next release, committed here:
https://github.com/arjenhiemstra/ithowifi/commit/6f5a113a1b836c3d70584b624efd02469f3d7a05

as workaround you could fill in: C5,05,51

joho500, 2024-03-30:
Thanks, Arjen

closed 2024-06-07 by stale[bot]