arjenhiemstra/ithowifi #250: MQTT settings are reset to defaults when module restarts

silvanverschuur, 2024-05-12:
**Describe the bug**
The MQTT settings are restored to default values when the module restarts. This happens when restarting the device from the web page but also when the cable between the Itho WPU and the module is disconnected. The MQTT settings are saved using the 'Save' button on the MQTT page

**Device information**
- Firmware version: 2.9.0-beta1
- Hardware revision: non-cve
- CC1101 RF module enabled: yes

arjenhiemstra, 2024-05-12:
Thanks for reporting this! I also noticed a reset of the system settings yesterday, trying to find out what causes this.

silvanverschuur, 2024-07-12:
@arjenhiemstra I'm wondering if you have found the issue? If so, is it possible to release a beta to test it?

printplaatreparatie, 2024-08-15:
It took me a long time to find that this was the issue for me. Let me give some background information so others that use tls can find this issue (and fix) easier.

I noticed that the Itho box was no longer communicating with Home Assistant. The mosquitto logs were showing these two errors: 

```
New connection from 192.168.xxx.xxx:xxxxx on port 8883.
OpenSSL Error[0]: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
2024-08-15 23:29:04: Client <unknown> disconnected: Protocol error.
```

Not helpful and even misleading. The fix is easy: After every reboot, change the port number to 1883 in the ngr-itho webinterface. Save, change it back to 8883, save.

silvanverschuur, 2024-10-24:
@arjenhiemstra Do you know when a new (beta) version is released with this fix?

arjenhiemstra, 2024-12-12:
Hopefully this will not happen anymore with the latest 2.9.0-beta5 release. I suspect the config was somehow saved before being loaded properly. This should be prevented from happing since this beta.

silvanverschuur, 2024-12-21:
I still have the same issue with the latest beta (2.9.0-beta6). Also the `WPU counters (4210)` setting isn't preserved when power is lost. So maybe all setting are reverted to the defaults when the module is disconnected from the power?

arjenhiemstra, 2024-12-21:
Already found the issue, will try to compile a new version this evening

silvanverschuur, 2024-12-21:
I'm glad to hear that you found the issue! Looking forward to test the next beta.

arjenhiemstra, 2024-12-22:
https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.9.0-beta7
I believe the config save issues are now really fixed

shostakovits, 2025-01-12:
Unfortunately the issue is still not resolved for me.
I'm running  [2.9.0-beta10](https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.9.0-beta10)
Every time I reboot the module, the MQTT settings are gone.

I Used the "Edit filesystem" tool to find the file where the credentials / settings are stored, but I was unable to find it.

Enabling the log in 'debug' mode didn't bring much either. No extra info is generated.

_I'm using the following hardware:
Itho device type: CVE-Silent
Itho hw version: 46
Itho fw version: 25
Manufacturer: 0_

closed 2025-03-21 by stale[bot]