arjenhiemstra/ithowifi #254: Losing System status (2401) counters after a while

ndewijer, 2024-05-29:
**Describe the bug**

After booting up, All counters work. 2401 and 4210. After a few hours, only the 4210 counters remain.

Working situation at 29-05-2024 14:35

This is from the MQTT server:

```{"Outside temp (°C)":20.66,"Boiler temp down (°C)":59.45,"Boiler temp up (°C)":58.92,"Evaporator temp (°C)":16.58,"Suction gas temp (°C)":28.6,"Compressed gas temp (°C)":52.04,"Liquid temp (°C)":42.5,"Temp to source (°C)":15.89,"Temp from source (°C)":17.84,"CV supply temp (°C)":25.16,"CV return temp (°C)":30.5,"CV pressure (Bar)":2.17,"Compressor current (A)":0,"Current e-element (A)":0,"Pressure switch":0,"Tariff":0,"Condensation protection":0,"Spare input":0,"Correction measurement delta source temperature (K)":-0.57,"Flow sensor (lt_hr)":0,"Phase detection":0,"Cv pump (%)":0,"Well pump (%)":0,"Boiler pump (%)":0,"Free cooling valve (%)":100,"CV_DHW or cooling temp valve (%)":0,"Compressor":0,"Element":0,"Trickle heating":0,"Error":1,"Free cooling":1,"Expansion valve (pls)":0,"Room temp (°C)":20.08,"Requested room temp (°C)":20,"Heat demand thermost. (%)":0,"Status":4,"Sub_status":255,"Blockage":0,"Calculated evaporation temp CV (°C)":16.08,"Calculated CV condensation temp (°C)":45.18,"Calculated evaporation temp DHW (°C)":16.08,"Calculated condensation temp DHW (°C)":45.18,"Error_found":0,"Error_retry":0,"Task active":0,"Manual operation":0,"Compr block (sec)":0,"Elek block (sec)":0,"Cv for _ naddraai (sec)":0,"Source for _ nadraai (sec)":0,"Boiler for _ nadraai (sec)":0,"Delay electr (sec)":0,"Min running time compr (sec)":0,"Drain time (sec)":0,"EV pressure offset (sec)":0,"EV adjust (sec)":0,"Free cooling interval (sec)":180,"Manual control (sec)":0,"Low pressure timer (sec)":0,"Compressor start (sec)":0,"Compr power on delay (sec)":0,"Delay cv start (sec)":0,"Delay cv stop (sec)":0,"Adaptive timer (sec)":0,"Adaptive overheat (K)":5.5,"adaptive fifo index":8,"Fault highest priority":0,"UTC time":1,"Max CV return temp (°C)":10,"Source pump speed free cooling mode":0,"Compressor in Cv mode":0,"Compressor in boiler mode":0,"Element in CV mode":0,"Element in boiler mode":0,"CV mode blocked":0,"Boiler mode blocked":0,"Free cooling mode blocked":0,"Bleeding mode blocked":0,"Electr element blocked":0,"Electr element DHW blocked":0,"Compressor blocked":0,"Off mode active":0,"CV mode active":0,"Boiler mode active":0,"Free cooling mode active":1,"CV pump prime":0,"Well pump prime":0,"Element released":0,"Additional cooling release":0,"CV enabled":0,"Free cooling enabled":1,"Tariff low from thermostat":0,"Venting from thermostat":0,"ECO selected on thermostat":0,"Comfort selected on thermostat":1,"Boiler blocked from thermostat":0,"Boiler boost from thermostat":0,"Heat demand total (%)":0,"E-consumption during stand-by (kWh)":0,"E-consumption during heating (kWh)":0,"E-consumption during DHW (kWh)":0,"E-consumption during cooling (kWh)":0,"Preheat tap water":0,"Element blocked during retry":1,"Pi error sourceflow":0,"Current source valve position (%)":0,"Utc Time offset (min)":0,"Free cooling block time (min)":0,"Cooling temp control valve setpoint (%)":0,"CO valve position (%)":0,"HRU blowout temp (°C)":22.66,"HRU blowout flow (m3_h)":150,"Regeneration active":0,"Latest valid source supply temp. (°C)":12.54,"Source pump flow setpoint (l_h)":1600,"Well pump speed at compr (%)":0,"Source pump speed at free cooling (%)":0,"Well pump speed at airreg (%)":0,"Free cooling mode":2,"Time CO valve start position (sec)":0,"Minimum release time external cooling (sec)":0,"Block time-release external cooling (sec)":0,"External heating release time (sec)":0,"Blocking time trickle low after power-up (sec.)":0,"Blocking time trickle low after CV operation (sec)":0,"Block time pre-heating tap water (sec)":0,"Slow start well pump (sec)":0,"Source return temperature too low (sec.)":0,"Source flow control period (sec)":0,"Stabilisation waiting time-free cooling (sec.)":0,"Minimum time for preheating potable water (sec)":0,"Maximum time for preheating potable water (sec)":0,"Electrical element release (min)":0}```

Syslog since turning on:

```
  18442 I: I2C init: QueryStatus
     18256 I: Hostname: nrg-itho-9e4c
     18078 I: mDNS: started
     17664 I: I2C init: QueryStatusFormat - items:134
     17410 I: Webserver: started
     17144 I: MQTT: connected, System config: 1
     16917 I: I2C init: QueryDevicetype - mfr:0x00 type:0x0D fw:0x25 hw:0x4E
     14795 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
     14628 I: WiFi: connection successful
      8710 I: wifi AP mode started
      6963 E: Unable to set wifi disconnect
      4524 I: I2C sniffer capable hardware: yes
      4322 I: HW rev: NON-CVE 1, FW ver.: 2.8.0
      3804 I: Device UUID: 993c8f2b-1d34-4771-82fe-35acee3d2f54
      3492 I: System boot, last reset reason: POWERON_RESET
```

counters
``` 80 82 C2 10 01 35 1A 11 49 12 B2 01 9A 05 B8 00 00 04 80 01 AF 0D 4C 27 1C 0B 33 0B 42 06 51 08 95 00 00 06 1A 02 C2 01 14 00 11 00 00 00 00 00 00 00 00 00 00 00 00 02 C2 02 40 07 ```

Connection:
Config version: 005
Itho I2C connection status: connected
File system: 32768 bytes used / 131072 bytes total

Then, at 29-05-2024 21:12

MQTT
```
{"CH pump on (h)":4432,"Source pump on (h)":4792,"DHW pump on (h)":410,"Compressor on (h)":1464,"Electrical element on (h)":0,"CH mode (h)":1152,"DHW mode (h)":431,"Free cooling mode (h)":3410,"Total runtime (h)":10018,"CH pump starts":2868,"Source pump starts":2883,"DHW pump starts":1617,"Compressor starts":2197,"Electrical element starts":0,"CH mode starts":1562,"DHW mode starts":706,"Free cooling starts":276,"System starts":17,"DHW element on (h)":0,"DHW element starts":0,"E-consumption stand-by (kWh)":0,"E-consumption heating (kWh)":0,"E-consumption DHW (kWh)":0,"E-consumption cooling (kWh)":0,"Energy out of source (MWh)":706,"Energy returned to source (MWh)":577}
```

Syslogs since last messages:
```
2024-05-29 16:48:34 I: I2C init: QueryStatus
2024-05-29 16:48:34 I: I2C init: QueryStatusFormat - items:0
2024-05-29 16:48:34 I: I2C init: QueryDevicetype - mfr:0x00 type:0x0D fw:0x25 hw:0x4E
2024-05-29 16:48:34 I: Hostname: nrg-itho-9e4c
2024-05-29 16:48:33 I: mDNS: started
2024-05-29 16:48:33 I: Webserver: started
2024-05-29 16:48:33 I: MQTT: connected, System config: 1
2024-05-29 16:48:31 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2024-05-29 16:44:49 I: WiFi: connection successful
2024-05-29 16:44:43 I: wifi AP mode started
2024-05-29 16:44:41 E: Unable to set wifi disconnect
2024-05-29 16:44:38 I: I2C sniffer capable hardware: yes
2024-05-29 16:44:38 I: HW rev: NON-CVE 1, FW ver.: 2.8.0
2024-05-29 16:44:38 I: Device UUID: 993c8f2b-1d34-4771-82fe-35acee3d2f54
2024-05-29 16:44:37 I: System boot, last reset reason: OWDT_RESET
2024-05-29 16:44:33 E: Unable to set wifi disconnect
2024-05-29 16:44:30 I: I2C sniffer capable hardware: yes
2024-05-29 16:44:30 I: HW rev: NON-CVE 1, FW ver.: 2.8.0
2024-05-29 16:44:30 I: Device UUID: 993c8f2b-1d34-4771-82fe-35acee3d2f54
2024-05-29 16:44:30 I: System boot, last reset reason: OWDT_RESET
2024-05-29 16:44:25 E: Unable to set wifi disconnect
2024-05-29 16:44:23 I: I2C sniffer capable hardware: yes
2024-05-29 16:44:22 I: HW rev: NON-CVE 1, FW ver.: 2.8.0
2024-05-29 16:44:22 I: Device UUID: 993c8f2b-1d34-4771-82fe-35acee3d2f54
2024-05-29 16:44:22 I: System boot, last reset reason: SDIO_RESET
2024-05-29 16:47:34 W: Warning: Task MQTT timed out!
```
counters
```
80 82 C2 10 01 35 1A 11 50 12 B8 01 9A 05 B8 00 00 04 80 01 AF 0D 52 27 23 0B 34 0B 43 06 51 08 95 00 00 06 1A 02 C2 01 14 00 11 00 00 00 00 00 00 00 00 00 00 00 00 02 C2 02 41 EA
```

Connection:
Config version: 005
Itho I2C connection status: connected
File system: 32768 bytes used / 131072 bytes total

**To Reproduce**
Steps to reproduce the behaviour:
unknown

**Expected behaviour**
All counts stay active.

**Screenshots**
[image]
Import of the MQTT into homeassistant. You can see the counters working until 16:45. 

**Device information**
- Firmware version 2.8.0
- Hardware revision non-cve 1.
- Type/model number Itho (on sticker outside the box) wpu 5g
hw version: 78 
fw version: 37
- CC1101 RF module enabled no
- In case of wifi issues, wifi access point make and model:

**Debug logging**
Please provide debug logging from the debug page if possible:

See above in description.

arjenhiemstra, 2024-05-29:
What is the output of "Query Status" on the debug page? (FYI; Query Status == 2401 command)

ndewijer, 2024-05-29:
Query status
```Result: 80 82 A4 01 01 D5 05 E9 16 5D 16 79 07 06 08 02 09 F8 09 3A 06 7C 05 44 07 08 07 46 01 21 00 00 00 00 00 00 00 00 FF C7 00 F5 00 64 06 00 64 64 00 00 00 01 01 00 00 07 E5 07 D0 00 04 FF 00 06 D4 09 B9 06 D4 09 B9 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 26 08 00 66 57 AF 45 0A 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 64 00 06 B1 00 96 00 06 7C 06 40 00 06 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 8C```

and Query 2410
```
Result: 80 82 A4 10 01 13 00 00 00 00 00 00 00 00 FF FF FF FF 00 00 00 01 10 00 F0 39
Current: 0
Minimum value: 0
Maximum value: 65535
```

ndewijer, 2024-05-30:
Just to clarify, the 2401 counters don't show up on the "Itho Status" page ether. MQTT matches what's on the status page.

closed 2024-08-06 by stale[bot]