arjenhiemstra/ithowifi #257: HA MQTT: received on topic itho/ithostatus. 'high' is not a valid preset mode

roelvandermade, 2024-06-18:
**Describe the bug**
SInce a certain release of either ithowifi or HA (sorry, not sure which triggered it) I receive MQTT errors about 'low' or 'high' modes not being a valid preset mode

**To Reproduce**
A running HA with itho integration connected through MQTT (Mosquitto integration in HA)

**Expected behaviour**
No similar warning message in the HA logs

**Screenshots**
N/A

**Device information**
- Firmware version 2.9.0-beta1
- Hardware revision 2
- Type/model number Itho MV-unit 468, RV-sensor inside, Perilex stekker
- CC1101 RF module enabled yes

**Debug logging**
```
2024-06-18 17:29:49.039 WARNING (MainThread) [homeassistant.components.mqtt.fan] '{"temp":22.4,"hum":59.2,"ppmw":10108,"Speed status":100,"Internal fault":0,"Frost cycle":0,"Filter dirty":0,"AirQuality (%)":"not available","AirQbased on":0,"CO2level (ppm)":"not available","Indoorhumidity (%)":59,"Outdoorhumidity (%)":"not available","Exhausttemp (°C)":"not available","SupplyTemp (°C)":"not available","IndoorTemp (°C)":"not available","OutdoorTemp (°C)":"not available","SpeedCap":63488,"BypassPos (%)":"not available","FanInfo":"auto","ExhFanSpeed (%)":100,"InFanSpeed (%)":0,"RemainingTime (min)":0,"PostHeat (%)":"not available","PreHeat (%)":"not available","InFlow (l sec)":"not available","ExhFlow (l sec)":"not available","Ventilation setpoint (%)":100,"Fan setpoint (rpm)":2040,"Fan speed (rpm)":2044,"Error":0,"Selection":7,"Startup counter":35,"Total operation (hours)":4008,"Absence (min)":0,"Highest CO2 concentration (ppm)":"not available","Highest RH concentration (%)":59,"RelativeHumidity":59.2,"Temperature":22.4}' received on topic itho/ithostatus. 'high' is not a valid preset mode
```

arjenhiemstra, 2024-07-18:
This part of the debug logging
```
2024-06-18 17:29:49.039 WARNING (MainThread) [homeassistant.components.mqtt.fan] '{"temp":22.4,"hum":59.2,"ppmw":10108,"Speed status":100,"Internal fault":0,"Frost cycle":0,"Filter dirty":0,"AirQuality (%)":"not available","AirQbased on":0,"CO2level (ppm)":"not available","Indoorhumidity (%)":59,"Outdoorhumidity (%)":"not available","Exhausttemp (°C)":"not available","SupplyTemp (°C)":"not available","IndoorTemp (°C)":"not available","OutdoorTemp (°C)":"not available","SpeedCap":63488,"BypassPos (%)":"not available","FanInfo":"auto","ExhFanSpeed (%)":100,"InFanSpeed (%)":0,"RemainingTime (min)":0,"PostHeat (%)":"not available","PreHeat (%)":"not available","InFlow (l sec)":"not available","ExhFlow (l sec)":"not available","Ventilation setpoint (%)":100,"Fan setpoint (rpm)":2040,"Fan speed (rpm)":2044,"Error":0,"Selection":7,"Startup counter":35,"Total operation (hours)":4008,"Absence (min)":0,"Highest CO2 concentration (ppm)":"not available","Highest RH concentration (%)":59,"RelativeHumidity":59.2,"Temperature":22.4}' received on topic itho/ithostatus.
```
 seems to have no relation with
```
'high' is not a valid preset mode
```

The first part seems correct (at least JSON syntax wise), why there is a warning I cannot see.

"'high' is not a valid preset mode" is an error message related to HA templates. There must be some more info after this part in the logging I beieve.

jasperslits, 2024-07-18:
Check your config in HA on fan definition. You most likely have this somewhere in your YAML. 

You need to specify the presets. Your CVE looks a bit different but the culprit is missing preset_modes:

```yaml
- name: "Itho HRU 350 "
   unique_id: "Itho_hru_Fan"
   state_topic: "ithohru/lwt"
   state_value_template: "{% if value == 'online' %}ON{% else %}OFF{% endif %}" 
   command_topic: "ithohru/cmd"
   preset_mode_state_topic: "ithohru/ithostatus"
   preset_mode_command_template: "{ rfremotecmd: '{{ value }}'}"
   preset_mode_value_template: >
        {% set am = value_json['Actual Mode'] | int %}
          {% if am == 1 %}
            low
          {% elif am == 2 %}
            medium 
          {% elif am == 3 %}
            high
          {% elif am == 13 %}
             timer
          {% elif am == 24 %}
            auto
          {% elif am == 25 %}
            autonight
          {% else %}
            {{ am }}
          {% endif %}
   preset_mode_command_topic: "ithohru/cmd"
   preset_modes:
       - "low"
       - "medium"
       - "high"
       - "auto"
       - "autonight"
       - "timer1"
       - "timer2"
       - "timer3"
       - "timer"
```

roelvandermade, 2024-07-19:
> Check your config in HA on fan definition. You most likely have this somewhere in your YAML.
>
> You need to specify the presets. Your CVE looks a bit different but the culprit is missing [...]

Hey good point, will look into this and report back, thanks!

roelvandermade, 2024-07-19:
So this is my current config with 4 sensors defined, am not sure where to include the preset_modes, I could use an example :)

```
- sensor:
   name: "Itho Percentage"
   unique_id: "itho_percentage"
   state_topic: "itho/ithostatus"
   unit_of_measurement: "%"
   value_template: "{{ value_json['Ventilation setpoint (%)'] | round (0) }}"

- sensor:
   name: "Itho setpoint (rpm)"
   unique_id: "itho_fan_setpoint"
   state_topic: "itho/ithostatus"
   unit_of_measurement: "rpm"
   value_template: "{{ value_json['Fan setpoint (rpm)'] | round (0) }}"

- sensor:
   name: "Itho speed (rpm)"
   unique_id: "itho_fan_speed"
   state_topic: "itho/ithostatus"
   unit_of_measurement: "rpm"
   value_template: "{{ value_json['Fan speed (rpm)'] | round (0) }}"

- sensor:
   name: "Itho humidity"
   unique_id: "itho_humidity"
   state_topic: "itho/ithostatus"
   unit_of_measurement: "%"
   value_template: "{{ value_json['hum'] | round (0) }}"
```

arjenhiemstra, 2024-07-19:
sensors do not use preset modes, must be somewhere else in the config

jasperslits, 2024-07-19:
Yes, that’s why I said you should check the yaml definition for your **fan**. 

Or just do a grep -R on “preset” in your ha config directory

roelvandermade, 2024-07-23:
Will dive into this, closing this issue since it looks not to be itho related. thanks!

closed 2024-07-23 by roelvandermade

Pistoletjes, 2024-11-26:
@arjenhiemstra not sure if adding this comment is useful, but I see the same issue and I think it's related to Home Assistant MQTT Discovery. in MQTT I receive in the homeassistant/fan/nrg-itho-### topic:

[image]

and: 

[image]

I think the case difference between the preset modes and the pr_mode_val_tpl may cause the issue? After updating the preset configuration from 'Low' to 'low', the issue resolved itself for me (until the next discovery, I think).