arjenhiemstra/ithowifi #269: Virtual Remote 0 cannot be a CO2 or RV remote

bgelens, 2024-09-27:
**Describe the bug**
I just got the board installed and I am successfully using RFT auto as virtual remote 0. This remote however is missing the medium capability. Disjoining the virtual remote and joining it as a CO2 or RV remote or just changing the RFT auto to the CO2 or RV remote results in the HRU not responding to the commands send.

**To Reproduce**
Steps to reproduce the behaviour:
Green field join virtual remote 0 on power cycle of the HRU as CO2 or RV

**Expected behaviour**
The capabilities to work using one of the two remote types

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
- Firmware version 2.8.0
- Hardware revision NON-CVE 1
- Type/model number Itho HRU ECO 350
- CC1101 RF module enabled yes

arjenhiemstra, 2024-10-17:
> changing the RFT auto to the CO2 or RV remote results in the HRU not responding to the commands send.

This will never work because the commands are different. I you want to change the remote type you must always rejoin the remote and probably also unlearn the remote first.

>Disjoining the virtual remote and joining it as a CO2 or RV remote

Does this issue only occur on vremote 0? On other slots it does work?

bgelens, 2024-10-27:
Unless I'm doing something totally wrong, it does not work on any slot

I've unjoined all remotes and tried to join in a virtual remote configured as co2 remote. Any commands issued on the vremote won't be executed

When I return to rft auto, things work again

arjenhiemstra, 2024-10-28:
Sorry, I was set on the wrong foot. Have been altering virtual and rf remote code too much lately I guess.
The hint is on the virtual remote page itself:
"There are multiple types of remotes that can be emulated; The RFT CVE (536-0020, 536-0024, 536-0124), RFT AUTO(-N) (536-0150), RFT DemandFlow/QualityFlow (536-0146)."

The CO2 and RV remotes are not included in this list. It is expected that it doesn't work or only limited. The RFT CO2 and RFT RV are remotes that need to be joined bi-directional to the itho. There is currently no support for this in the virtual remotes section, in the RF remote section it is in development. 
They also do not have a legacy mode like the new RFT Auto-N.

It might work if you give the virtual remote the same ID as an original one but it might also not work.

Is there a specific reason you want to use the RFT CO2 as virtual remote?

bgelens, 2024-10-29:
hey @arjenhiemstra. Thanks for the detailed answer. And no worries :) I'm loving the product! Fixed a bug in my unit which I would not know about without it (temp sensors where very much off on the high end causing bypass to be open way too much)

The reason I want to use the CO2 or RV is to have all modes available. Currently by using Auto / Auto-N I'm missing the `Medium` control.

I'm using vremote-0 with mqtt to map controls on Homey Pro

baerdt, 2024-11-02:
Not sure if this is the same issue, but I was struggling with something similar.

I have an HRU 300 so I'm only using RF. In case it's related: I have no virtual remote configured, but the default virtual remote is still there and untouched. 

When adding a physical RFT-CO2 remote (that is already paired with the HRU 300) under RF remotes at index 0, the web API is not working for this remote. However, when I move the entry to any other index, it does work. So it appears something's off with index 0 when set to CO2.

closed 2025-01-09 by stale[bot]