arjenhiemstra/ithowifi #270: Itho Spider Connection drops in 2.9.0-beta1

nielsdrost, 2024-10-11:
**Describe the bug**

I coupled my spider thermostat to the add-on, running 2.9.0-beta1. It works fine, except once a day or so it drops connection.

**To Reproduce**
1. Install 2.9.0-beta1
2. Connect Spider to RF devices as Device 0
3. Wait for some hours
4. Reporting of spider values stops. Other values (such as humidity, temperature, fan state, etc) keep working just fine
5. Reset.
6. Fill in MQTT settings again (is this a bug too?)
7. See functionality return.

Output:

`
2024-10-11 07:36:02 I: Mem free: 155204, Mem low: 96744, Mem block: 63476
2024-10-11 01:36:02 I: Mem free: 155208, Mem low: 96744, Mem block: 63476
2024-10-10 19:36:02 I: Mem free: 155204, Mem low: 96744, Mem block: 63476
2024-10-10 13:36:02 I: Mem free: 155208, Mem low: 96744, Mem block: 63476
2024-10-10 07:36:01 I: Mem free: 155216, Mem low: 96744, Mem block: 63476
2024-10-10 01:36:01 I: Mem free: 155204, Mem low: 96744, Mem block: 63476
2024-10-09 19:36:01 I: Mem free: 155208, Mem low: 96744, Mem block: 63476
2024-10-09 16:20:55 W: Warning: CC1101 Task timed out!
2024-10-09 16:20:52 I: send31D9
2024-10-09 16:20:42 I: send31DA
2024-10-09 16:20:42 I: send31D9
2024-10-09 16:19:47 I: send31DA
2024-10-09 16:19:47 I: send31D9
2024-10-09 16:19:24 I: send31DA
2024-10-09 16:19:24 I: send31D9
`

arjenhiemstra, 2024-12-12:
I'll look into it. Lots of changes to the RF part in the 2.9.0 release, still not finished though. I'll have to test further with. my Spider and other bidirectional RF devices.

nielsdrost, 2025-03-16:
Nudge for the Stale bot, no rush, I'll test again with 2.0.0-beta10

nielsdrost, 2025-03-16:
Running beta10 for the last week or so in "Monitor Only" mode seems to work just fine, nice! I can't quite figure out how the other modes are supposed to work. Ideally I would like to change the setpoint.