arjenhiemstra/ithowifi #272: Highest received rh value > 100% #17

benjamin-dcs, 2024-10-30:
**Describe the bug**
Values for 'Highest received rh value' with HRU eco 350 sometimes exceed 100% RH

**Expected behaviour**
Value never exceeds 100%

**Screenshots**
[image]

**Device information**
- Firmware version 2.8.0
- Hardware revision NON-CVE 1
- Type/model number Itho HRU eco 350

arjenhiemstra, 2024-10-30:
See my comment on https://github.com/jasperslits/haithowifi/issues/17#issuecomment-2446199236

arjenhiemstra, 2024-10-30:
You can simply filter out or ignore all values above 100 or display the error message

benjamin-dcs, 2024-10-30:
Added filtering for Home Assistant Integration in https://github.com/jasperslits/haithowifi/pull/19

closed 2024-10-30 by benjamin-dcs