arjenhiemstra/ithowifi #273: Correct/clarify  the value of ID "FanInfo" in "Itho Status" overview.

WO15, 2024-11-06:
In the overview "Itho Status" there is an ID "FanInfo" this gives the human readable translation of ID "Selection".

Overview of observed values in my Itho CVE-S HE Optima 2 box.

Selection    FanInfo
2            low
3            medium
4            high
5            timer3
7            auto
8            unknown

"Unknown" for "selection" = 8 should probably be "autonight"
"timer3" would probably be better named "timer" as it is used for timer1, timer2, timer3. The difference between these 3 timers is visible in de ID "RemainingTime(min)".

The  "auto night" option can be selected with RFT-CO2 (fast blinking Auto LED) or by pressing the "auto" button on RFT-Auto remote 2x quickly. 

tested on HW + SW NRG-Watch PCB
Current firmware version: 2.8.0
Hardware revision: 2

arjenhiemstra, 2024-11-06:
Autonight seems to be the value 0x18, it was added with commit: https://github.com/arjenhiemstra/ithowifi/commit/da2fc909723f08292855dd34dea7d002ae7b7a94 and will be part of the next version.

which doesn't mean that 0x08 isn't also autonight but I haven't seen it before. Is there a way to confirm this?

> "timer3" would probably be better named "timer" as it is used for timer1, timer2, timer3. The difference between these 3 timers is visible in de ID "RemainingTime(min)".

On CVE devices it seems that this is always timer3 indeed. Not sure about other itho devices though. 
It can be changed of course but it's only cosmetical and might break some integrations. For now I'm leaving it as it is.

WO15, 2024-11-07:
[image]

Above you see a screenshot of Itho Status overview, where I have selected Autonight mode on Itho Optima 2 box. 
Label FanInfo is the first line in the screenshot, Label Selection i the bottom/last line. 
On screenshot below you see the same info when Auto mode is selected. I assume you "translate" the number in Selection to a human readable format in FanInfo. In case you read FanInfo directly from the box, and not "translate" it is seems a small bug in the Itho box.

[image]

arjenhiemstra, 2024-12-12:
Thanks for the info, it is not a translation of Selection, it is actually a lookup of the value provided by the itho on FanInfo. This is a numerical value that is looked on a list, if there is a match the string value is given, of not it results in unknown.
Here is the list:
https://github.com/arjenhiemstra/ithowifi/blob/75d1711d8799f3c38a2155e0d2d6623bc44cb92f/software/NRG_itho_wifi/main/devices/error_info_labels.h#L9-L36

autonight is included since a very recent beta in oct. 23. Could you try the latest beta5 and report if this fixes your issue?

WO15, 2024-12-14:
Tested on firmware version: 2.9.0-beta6 
When Itho box is in auto night mode, item "18 FanInfo" is "autonight". So issue is fixed.
For timer function, "timer 3" is shown for all tree timers short time, medium time and long time. In item "21 RemainingTime (min)", you can see the amount of minutes still remaining, this is a little unclear, "Timer" instead of "Timer 3" would be better for Itho Optima 2, but I am not sure if this is also the case of other supported products. So leave this up to you.

arjenhiemstra, 2024-12-14:
the itho firmware (always?) reports back timer3, indeed only the time remaining seems to change. The add-on firmware just passes this through unchanged.

WO15, 2024-12-15:
Then I would suggest to change the label from "timer3" tot "timer" no big deal but more clear.

arjenhiemstra, 2024-12-22:
there are integrations using timer3 for their logic, that makes it a breaking change.

closed 2025-02-27 by stale[bot]