arjenhiemstra/ithowifi #278: webAPI v2 documentation

sanderkob, 2024-11-27:
**Describe the bug**
documentation for webAPIv2 missing (or not easy to find)

**Device information**
- Firmware version 2.9.0-beta4
- Hardware revision cve 55/27
- Type/model number Itho CVE-S ECO
- CC1101 RF module enabled yes

**Desktop (please complete the following information):**
 - OS: windows 10
 - Browser chrome
 - Version 131.0.6778.86

arjenhiemstra, 2024-11-27:
After activation of API v2, the API page of the add-on contains info about the API v2.

Any suggestions to have this info elsewhere?

arjenhiemstra, 2024-11-27:
>As to the documentation, some parts have not been updated.

Yes of course! Didn't update that part yet.

>I will send the changes I had to make, this may help updating the documentation.

Thanks a lot!!

arjenhiemstra, 2024-11-27:
And to add to that, if you have general feedback about the API, see any inconsistencies etc. please let me know!

sanderkob, 2024-11-28:
I made the changes in Node-RED. For the query `api.html?get=ithostatus` the response key/value pairs are nested so e.g. in Node-RED msg.payload.temp becomes msg.payload.data.ithostatus.temp.
Bracket notation works as before, e.g. msg.payload.data.ithostatus["CO2level (ppm)"].
I noticed that the response to a setting query (`api.html?getsetting=`) in API v1 already did return the new format JSON, so in this case I did not have to make changes going to API v2.
As to the documentation, in API v1 the result from the query `api.html?get=currentspeed` is given as _string_ whereas it actually is a number (similar to the _state_ topic in mqtt). The documentation for API v2 does not need changes but it would be helpful to state that the _ithostatus_ key/value pairs are nested inside the members _data_  and _ithostatus_. An example may help.

arjenhiemstra, 2024-12-03:
> in API v1 the result from the query api.html?get=currentspeed is given as string whereas it actually is a number (similar to the state topic in mqtt).

Thats interesting, it should be a string just like v2.

The code producing the string for v1:
https://github.com/arjenhiemstra/ithowifi/blob/f287881e817b62525102817c3809ccf261c6776c/software/NRG_itho_wifi/main/tasks/task_web.cpp#L584-L590

is the same as for v2:
https://github.com/arjenhiemstra/ithowifi/blob/f287881e817b62525102817c3809ccf261c6776c/software/NRG_itho_wifi/main/tasks/task_web.cpp#L838-L844

Maybe the JSON lib handles string input differently then the WebServer lib, or the browser displays it differently?

sanderkob, 2024-12-04:
> Maybe the JSON lib handles string input differently compared to the

Solved. In Node-Red I used the http request node with return set to parsed
json object. That returned a number.
When set to binary buffer it returned 0x31 0x35, being the string
representation of *15.*

closed 2025-02-09 by stale[bot]