arjenhiemstra/ithowifi #279: Something missing in Dutch manual ?

ObiToo55, 2024-12-10:
Fysieke installatie:

After doing point 4 on a type **CVE-S Optima Inside with CO2- and humidity sensor**

There is already an add-on print on the main print mounted. 

[image]
[image]

How to handle now ? 
Remove the addon, Place the IthoWifi print and piggyback the original onto the IthoWifi print ?

arjenhiemstra, 2024-12-10:
It's also mentioned on the product page of the module but in the wiki there is more information on this topic, please have a look here:
https://github.com/arjenhiemstra/ithowifi/wiki/CO2-sensors#itho-with-built-in-co2--sensor-cve-s-optima-inside

ObiToo55, 2024-12-10:
Ah ! a little bit sneaky mentioned on another chapter ;-)
Now hoping that the header will be long enough....
Thanks for your comment

arjenhiemstra, 2024-12-10:
On the add-on product page you find this piece of info:
[image]

arjenhiemstra, 2024-12-10:
Not to be sneaked away in any way of course! If you have suggestions to make this information available in a better way, please let me know!

closed 2025-02-16 by stale[bot]