arjenhiemstra/ithowifi #280: Logging via syslog stops suddenly, only a reboot helps

janbredenbeek, 2024-12-11:
**Describe the bug**
Logging via syslog server suddenly stops for no apparent reason. Logging via GUI (syslog, RF debug log) is not affected. Turning RF debug log off and on doesn't help, only a reboot helps.

**To Reproduce**
Steps to reproduce the behaviour:
1. Enable logging via syslog server
2. Enable RF debug log level 3
3. Wait some time (sometimes a few hours, sometimes a few days)
4. At some point, the logging via syslog stops but the logging continues in the GUI (also RF debug log)

**Expected behaviour**
Should not occur, versions before 2.9.0-beta3 or 4 were fine.

**Screenshots**
N/A

**Device information**
- Firmware version: 2.9.0-beta4
- Hardware revision: Non-CVE 1
- Type/model number Itho HRU-350
- CC1101 RF module enabled: Yes
- In case of wifi issues, wifi access point make and model: AVM FritzBox 7590

**Debug logging**
No obvious error messages, the log suddenly stops without prior message.

**Desktop (please complete the following information):**
 - OS: Windows 10 for GUI, syslog goes to Raspberry Pi syslog
 - Browser: Firefox
 - Version: 133

**Smartphone (please complete the following information):**
 - Device: N/A
 - OS: N/A
 - Browser: N/A
 - Version: N/A

arjenhiemstra, 2024-12-12:
The last code changes related to syslog were introduced with 2.9.0-beta1. Looking at what has changed those were changes intended to prevent multiple tasks calling the syslog function while another task is using it, if that happens it could lead to a crash.
If you experience changes in behaviour between beta3 and versions thereafter it might well be unrelated to syslog but due to other underlying libs or network related bugs. It syslog uses UDP, the rest of the communication is TCP based. It could be something UDP related. For now I do not really have any information to did further unfortunately.

I would suggest to at least update to beta5. In the mean time I'll setup a syslog test env. to do some long running tests

janbredenbeek, 2024-12-16:
Upgraded to beta6 but the syslog problem still exists. I've also noticed that after a reboot the RF debug setting in syslog has reverted to 'Off', even when I saved it as Level3.

arjenhiemstra, 2024-12-18:
This is a build with ESP IDF 5, its a bit of a long shot but if you have the time, could you please test this?
Also the issues with saving the config file should be solved

[nrgitho-dev-v2.9.0-beta7-wputest.bin.zip](https://github.com/user-attachments/files/18188571/nrgitho-dev-v2.9.0-beta7-wputest.bin.zip)

arjenhiemstra, 2024-12-20:
It seems that for 2 other users of this firmware are having issues (#238) ... maybe better to wait before trying this.

arjenhiemstra, 2024-12-20:
See https://github.com/arjenhiemstra/ithowifi/issues/238#issuecomment-2557425183

janbredenbeek, 2024-12-20:
Successfully unbricked, now running 2.8.0 again. Thanks!

arjenhiemstra, 2024-12-20:
great! thx for the feedback!