arjenhiemstra/ithowifi #281: Login required on setup of non-cve version

wieltar, 2024-12-12:
**Describe the bug**
The issue is that I cannot complete the setup of the unit. I have to provide some username and or password whenever I connect to the module via wifi and navigate to the unit's ip address. (full disclosure, I purchased it from i assume a fellow tweaker on marktplaats)

I had some contact with him but he told me he did not change any defaults.
I tried the defaults from [ESP async webserver](https://github.com/nrgwatch/ESPAsyncWebServer) (admin / admin) however these were of no use.

Following this, I tried the reset and failsafe mode buttons, the failsafe button does not seem to do anything that I can see. I also missed in the dutch manual a description of what they do.

I tried reflashing the module via PIO, using the beta4 build from master. This has made no difference, it only made the login text change from http async authentication required to 'login required'.

A few things I have not tried yet;
- Flash the default provided bin files. (I don't know how, I only know how to use the esp home flasher tool. Dowloaded the flasher from espressif but i did not know how to use it)
- use the /update or /edit endpoints i saw in some threads, they resulted in a 404.

How can I reset the board, retrieve the login info and / or reflash the provided bin files?

**Device information**
- Firmware version 2.0.9-beta4 (flashed a build from master using the beta flag)
- Hardware revision non-cve version
- Type/model number Itho HRU 300
- CC1101 RF module enabled: yes

**Debug logging**
-

**Smartphone (please complete the following information):**
 - Device: Pixel 7
 - OS: android 15
 - Browser Firefox

arjenhiemstra, 2024-12-12:
On the main branch there is a manual, that will help you through the first of your questions.
There might be some issues with beta4 that are solved with beta5, I would urge you to go either to 2.8.0 or the 2.9.0-beta5, the binaries can be found in the folder compiled_firmware_files.

The fail save reset procedure is explained in the manual but there is also a youtube clip that might help:
https://youtu.be/3sWclzq73n4

Furthermore, this github repo also has a quite extensive wiki, please have a look there aswel.

I'll close this issue as I expect all of your issues will be answered by the provided documentation.

arjenhiemstra, 2024-12-12:
manual:
https://github.com/arjenhiemstra/ithowifi/blob/master/handleiding.pdf

and as long as there is an official firmware on the device the fail save procedure should always enable you to reset the module or flash a new firmware using the web interface on the module

closed 2024-12-12 by arjenhiemstra

wieltar, 2024-12-12:
Hey Arjen,

Mistook the red led for the status led. On this board it appears as the green RJ45 port's light.
Followed the manual pressing the reset and failsafe buttons accordingly. It has worked. Sorry should have read a bit better.

Tried it before but I was looking at the red led since I assumed the status led to be on the board.