arjenhiemstra/ithowifi #282: Unable to control HRU 300, or read connect i2c on versions above 2.5.7

wieltar, 2024-12-12:
**Describe the bug**
I've tried to connect my module to my HRU 300. This is not really going well.
- Above version 2.5.7 it is not able to connect the I2C. 
   - Tried version 2.8.0
   - Tried version 2.7.x
   - Tried version 2.7.0
   - Tried powercycling multiple times
   - On 2.5.7 it is generally available, or after a few cycles if it is not

It registers the co2 values from the remotes and whatever action is selected. The two way communication does not work. I read that that is 'fixed' in a higher version.

I connected the unit to the port on the hru300 that has the screen or other unit connected to it. So that one is disconnected.
What am I missing?

**Expected behaviour**
Whenever I press the remote, it will control the fan speed and read the itho status on higher versions.

**Screenshots**
[image]

This is the current state outputed on version 2.5.7:
[image]

These are the remotes I have connected.
[image]

These are the I2C settings
[image]

**Device information**
Itho device type: HRU 250-300
Itho fw version: 10
Current firmware version: 2.5.7
Hardware revision: NON-CVE 1
CC1101 RF module enabled yes

Config version: 005

**Debug logging**
Querying actions on the debug page fails

**Syslog**
```
2024-12-12 17:08:22 I: I2C init: i2c sniffer enabled
2024-12-12 17:08:22 I: I2C init: QueryStatus
2024-12-12 17:08:21 I: I2C init: QueryStatusFormat - items:32
2024-12-12 17:08:21 I: I2C init: QueryDevicetype - fw:10 hw:1
2024-12-12 17:05:37 E: I2C init: QueryDevicetype - failed
2024-12-12 17:05:36 I: Hostname: nrg-itho-0640
2024-12-12 17:05:36 I: mDNS: started
2024-12-12 17:05:36 I: Webserver: started
2024-12-12 17:05:35 I: Setup: init of CC1101 RF module successful
ESP-IDF: E (14200) gpio: gpio_install_isr_service(449): GPIO isr service already installed
2024-12-12 17:05:33 I: WiFi: connection successful
2024-12-12 17:05:33 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2024-12-12 17:05:27 E: Unable to set wifi disconnect
2024-12-12 17:05:25 I: I2C sniffer capable hardware: yes
      4033 I: HW rev: NON-CVE 1, FW ver.: 2.5.7
2024-12-12 17:05:24 I: System boot, last reset reason: SW_RESET
2024-12-12 17:05:19 I: Reboot requested
2024-12-12 17:04:39 E: I2C init: QueryDevicetype - failed
2024-12-12 17:04:39 I: Hostname: nrg-itho-0640
     14952 I: mDNS: started
2024-12-12 17:04:39 I: Webserver: started
2024-12-12 17:04:39 I: Setup: init of CC1101 RF module successful
ESP-IDF: E (13878) gpio: gpio_install_isr_service(449): GPIO isr service already installed
2024-12-12 17:04:35 I: WiFi: connection successful
2024-12-12 17:04:35 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2024-12-12 17:04:30 E: Unable to set wifi disconnect
2024-12-12 17:04:27 I: I2C sniffer capable hardware: yes
2024-12-12 17:04:27 I: HW rev: NON-CVE 1, FW ver.: 2.5.7
2024-12-12 17:04:27 I: System boot, last reset reason: OWDT_RESET
2024-12-12 17:03:50 I: I2C init: i2c sniffer enabled
2024-12-12 17:03:49 I: I2C init: QueryStatus
2024-12-12 17:03:49 I: I2C init: QueryStatusFormat - items:32
2024-12-12 17:03:49 I: I2C init: QueryDevicetype - fw:10 hw:1
     15267 E: I2C init: QueryDevicetype - failed
     14977 I: Hostname: nrg-itho-0640
     14795 I: mDNS: started
     14619 I: Webserver: started
     14471 I: Setup: init of CC1101 RF module successful
ESP-IDF: E (14221) gpio: gpio_install_isr_service(449): GPIO isr service already installed
     12598 I: WiFi: connection successful
     12314 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      6627 E: Unable to set wifi disconnect
      4058 I: I2C sniffer capable hardware: yes
      3935 I: HW rev: NON-CVE 1, FW ver.: 2.5.7
      3519 I: System boot, last reset reason: POWERON_RESET
     61073 I: I2C init: i2c sniffer enabled
     60870 I: I2C init: QueryStatus
     60625 I: I2C init: QueryStatusFormat - items:32
     60460 I: I2C init: QueryDevicetype - fw:10 hw:1
     16038 E: I2C init: QueryDevicetype - failed
```
**Log from the Auto-n joined in legacy mode**

```
12-12-2024, 16:57:08: H:1C _I P0:-- P1:-- 94,87,36 49,42,4C --,--,-- 31E0 04:00,00,01,00 (cmd:unknown)
12-12-2024, 16:57:08: H:1C _I P0:-- P1:-- 95,29,EC 49,4B,93 --,--,-- 2E10 03:00,01,00 (cmd:unknown)
12-12-2024, 16:56:56: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,1E (cmd:timer3)
12-12-2024, 16:56:56: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,1E (cmd:timer3)
12-12-2024, 16:56:56: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,1E (cmd:timer3)
12-12-2024, 16:56:51: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,14 (cmd:timer2)
12-12-2024, 16:56:51: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,14 (cmd:timer2)
12-12-2024, 16:56:51: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,14 (cmd:timer2)
12-12-2024, 16:56:48: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,14 (cmd:timer2)
12-12-2024, 16:56:48: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,14 (cmd:timer2)
12-12-2024, 16:56:48: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,14 (cmd:timer2)
12-12-2024, 16:56:46: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,0A (cmd:timer1)
12-12-2024, 16:56:46: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,0A (cmd:timer1)
12-12-2024, 16:56:46: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F3 03:00,00,0A (cmd:timer1)
12-12-2024, 16:56:42: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F1 03:00,02,04 (cmd:low)
12-12-2024, 16:56:42: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F1 03:00,02,04 (cmd:low)
12-12-2024, 16:56:42: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F1 03:00,02,04 (cmd:low)
12-12-2024, 16:56:40: H:18 _I P0:-- P1:-- 96,AC,E2 --,--,-- 96,AC,E2 1298 03:00,01,F2 (cmd:unknown)
12-12-2024, 16:56:39: H:1C _I P0:-- P1:-- 94,AA,B7 49,52,04 --,--,-- 31E0 04:00,80,01,00 (cmd:unknown)
12-12-2024, 16:56:38: H:0C RQ P0:-- P1:-- 95,26,51 95,26,D8 --,--,-- 4401 14:10,00,00,09,56,F1,00,00,00,00,00,00,00,00,00,00,00,00,00,00 (cmd:unknown)
12-12-2024, 16:56:36: H:1C _I P0:-- P1:-- 95,2D,99 49,55,0E --,--,-- 31E0 04:00,00,01,00 (cmd:unknown)
12-12-2024, 16:56:35: H:1C _I P0:-- P1:-- 95,0F,F8 95,30,56 --,--,-- 4401 14:10,00,00,09,56,F6,00,00,09,56,FF,1B,00,00,00,00,00,00,00,63 (cmd:unknown)
12-12-2024, 16:56:35: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F1 03:00,03,04 (cmd:medium)
12-12-2024, 16:56:35: H:0C RQ P0:-- P1:-- 95,30,56 95,0F,F8 --,--,-- 4401 14:10,00,00,09,56,F6,00,00,00,00,00,00,00,00,00,00,00,00,00,63 (cmd:unknown)
12-12-2024, 16:56:34: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F1 03:00,03,04 (cmd:medium)
12-12-2024, 16:56:34: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F1 03:00,03,04 (cmd:medium)
12-12-2024, 16:56:29: H:1C _I P0:-- P1:-- 95,0F,F8 95,26,52 --,--,-- 4401 14:10,00,00,09,56,F2,00,00,09,56,F9,5A,00,00,00,00,00,00,00,63 (cmd:unknown)
12-12-2024, 16:56:27: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F1 03:00,04,04 (cmd:high)
12-12-2024, 16:56:27: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F1 03:00,04,04 (cmd:high)
12-12-2024, 16:56:27: H:14 _I P0:-- P1:-- --,--,-- --,--,-- 96,0E,97 22F1 03:00,04,04 (cmd:high)
```

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Firefox
 - Version 133.0.3 (64-bits

arjenhiemstra, 2024-12-12:
A few things:
1. I see the i2c sniffer is enabled. This will block the i2c bus if any i2c error is detected. Only activate this if you really need to debug some i2c commands and switch if off thereafter immediately.
2. other users are using 2.8.0 in combination with the HRU300 without issues, indicating that the issues you experience might not be directly related to the firmware versions you mention.
3. I see that the display is not connected. I'm not sure if the I2C bus works as expected without display connected (I suppose the itho firmware expects something to be happing on that bus from the display)
4. The cable construction you made could be a source of the issues (ie. due to noise and cross talk on the i2c bus)
5. some users experience issues with unstable i2c communication which might be related to a weak power supply in some revisions. A unearthed (good ie. low noise) usb power supply might help.

wieltar, 2024-12-13:
Do you mean I need to splice a cable with the display and itho wifi unit connected? I only have a single exposed slot that I can see.

Regarding the logging, I did enable that yes, but not before the issues happened.

I'll try running it with a USB power brick attached to see if that would do anything. Is that okay to run with or without the 12v from the Itho box?

And I'll disable all of the settings that aren't supposed to be on. :)

arjenhiemstra, 2024-12-13:
That where this adapter board is created for:
https://www.nrgwatch.nl/product/hru300-adapter/

If you do not want to buy the board thats fine, but at a minimum I would advice a PCI-e 6pin male-female/female splitter cable and a good quality shielded ethernet cable, you can solder the PCI-e cable to the ethernet cable (that basically is what de adapter board does as well)

wieltar, 2024-12-16:
i re ordered the pci cable, i accidentally cut off the other end and soldered it wrong way round. I had a spare one laying around. so I did not think to create a split between the screen and the itho wifi unit. I thought the itho wifi unit was supposed to replace the screen itself.

Once it comes in ill resolder the splitter and connect both to see what happens.

wieltar, 2024-12-24:
So I got the cable, spliced it so that now both the screen and the Itho wifi unit are connected.

I updates to 2.8.0 and it shows all of the correct settings and the Itho status. Is it correct that I now need to join the Itho wifi module with a join command to the screen unit's rf module?

I cannot execute commands with a remote that is only paired to the wifi unit it seems.

closed 2025-03-01 by stale[bot]