arjenhiemstra/ithowifi #284: [Question] Factory defaults?

ymatsiuk, 2024-12-16:
I was messing around with my "Itho Settings" tab and now I want to start "fresh", is there a way to reset to factory defaults?

Just in case:
```
Itho device type: CVE-Silent
Itho hw version: 57
Itho fw version: 27
Manufacturer: 0

Current firmware version: 2.8.0
Hardware revision: 2
```

arjenhiemstra, 2024-12-16:
No, there is no way to reset it to factory defaults. That is the reason you can download a backup of your settings (before you start changing settings)

But you happen to have the same model as I do, here is a post on the tweakers forum with the default settings:
https://gathering.tweakers.net/forum/list_message/80977126#80977126

ymatsiuk, 2024-12-17:
Thank you! :bow:

closed 2024-12-17 by ymatsiuk