arjenhiemstra/ithowifi #285: Filter timers

supershadow, 2024-12-17:
Hallo Arjan,

is het ook mogelijk om de timers voor filters te zien in de itho wifi software want die zie ik momenteel niet ?
het gaat hier om de HRU 300 Eco ik gebruik de laaste beta versie 
[image]

Met vriendelijke groet
Michel

arjenhiemstra, 2024-12-18:
Waarvan is het screenshot?

Alle beschikbare informatie zou op de itho status pagina zichtbaar moeten zijn. Het kan helpen om de 31da en 31d9 i2c functies te activeren. Dan heb je sowieso alle info die beschikbaar is op de hru via de add-on beschkbaar.

supershadow, 2024-12-19:
screenshot is gemaakt van de ID_Servicetool die ik heb aangesloten op de hru 300 via usb 

ik krijg 32 status zien in de status pagina maar de filter staat er niet tussen
[image]

ik heb ook de 31da en 31d9 i2c functies aangezet maar krijg niet meer info te zien

arjenhiemstra, 2024-12-22:
vreemd dat 31da en 31d9 niet beschikbaar zijn, ik weet helaas niet 100% of die op de HRU ook beschikbaar zijn helaas. 
als ik naar dit comment kijk lijkt dat wel zo te zijn:
https://github.com/arjenhiemstra/ithowifi/issues/97#issuecomment-1366650284

zit er een goede kabel tussen je add-on en de hru?

supershadow, 2024-12-25:
ben nog ff verder aan het prutsen geweest en het vreemde is als ik versie nrgitho-noncve-v2.4.3 krijg ik wel een ouput op Query 31DA en 31D9 krijg 
[image]

supershadow, 2024-12-25:
ga ik terug op nrgitho-v2.9.0-beta9.bin dan krijg ik weer fail

arjenhiemstra, 2024-12-25:
Zijn er versies tussen 2.4.3 en 2.90 die ook werken? Dat zou enorm helpen bij het pinpointen van een eventueel issue

supershadow, 2024-12-25:
van af nrgitho-noncve-v2.4.3.bin lijkt het niet meer te werken ik heb verschillende geprobeerd maar geen geluk

supershadow, 2024-12-26:
Wat is eigelijk het verschil tussen hru250_300.h en hrueco.h ?
Ik heb een HRU eco 300 en de software geeft aan hru250_300

arjenhiemstra, 2024-12-26:
> van af nrgitho-noncve-v2.4.3.bin lijkt het niet meer te werken ik heb verschillende geprobeerd maar geen geluk

Ok, thx, ik duik er even verder in. Kom er zsm op terug

arjenhiemstra, 2024-12-26:
> Wat is eigelijk het verschil tussen hru250_300.h en hrueco.h ? Ik heb een HRU eco 300 en de software geeft aan hru250_300

Ander apparaat. Ik geloof dat de HRU eco op de huidige HRU 350 lijkt, de HRU 250/300 is meer een rechthoekig apparaat. De HRU250/300 lijkt door een andere fabrikant dan itho gemaakt te worden, itho voorziet het apparaat van haar logo en (beperkte) firmware.

Kuchiru, 2025-03-11:
misschien een toevoeging aan dit, het lijkt mij best fijn om de filter clean of filter new functies te kunnen gebruiken via ofwel de remote of via de addon.
Dit zou het proces wel wat gestroomlijnder maken :)