arjenhiemstra/ithowifi #287: 2.9.0b10 - MQTT rfremotecmd call no longer work

jasperslits, 2024-12-31:
**Describe the bug**
In 2.8.0 and prior versions this worked fine but in 2.9.0b10 nothing happens when rfremotecmd is used. In logs below ithohru/deviceinfo and ithohru/remotesinfo trimmed for brevity. I have the non-CVE add-on with CC1101. 

There are no errors in syslog and reverting back to 2.8.0 restores original behaviour. 

```
2024-12-31T13:15:08+0100 ithohru/cmd { rfremotecmd: 'autonight'}
2024-12-31T13:15:55+0100 ithohru/lastcmd {"source":"MQTT API","command":"rfremotecmd:autonight, idx:0","timestamp":1735647308}
2024-12-31T13:15:55+0100 ithohru/state 120
2024-12-31T13:16:38+0100 ithohru/ithostatus {"Requested fanspeed (%)":-0.99,"Balance (%)":99.8,"Supply fan (RPM)":758,"Supply fan actual (RPM)":758,"Exhaust fan (RPM)":760,"Exhaust fan actual (RPM)":765,"Supply temp (°C)":16.92,"Exhaust temp (°C)":7.23,"Status":0,"Room temp (°C)":16.92,"Outdoor temp (°C)":7.23,"Valve position":0,"Bypass position":0,"Summercounter":0,"Summerday (K_min)":0,"Frost timer":0,"Boiler timer":177,"Frost block":120,"Current position":0,"VKKswitch":0,"GHEswitch":0,"Airfilter counter":2282,"Global fault code":0,"Actual Mode":24,"Pir fan speed level":-1,"Highest received CO2 value (Ppm)":562,"Highest received RH value (%RH)":239,"Air Quality (%)":100,"Remaining override timer (Sec)":0,"Fallback speed timer (Sec)":0,"Label out of bound error":0}
2024-12-31T13:16:38+0100 ithohru/lastcmd {"source":"MQTT API","command":"rfremotecmd:autonight, idx:0","timestamp":1735647308}
```

Syslog shows no problems

```
2024-12-30 11:57:35 I: I2C init: QueryStatus
2024-12-30 11:57:35 I: I2C init: QueryStatusFormat - items:33
2024-12-30 11:57:35 I: Hostname: hru
2024-12-30 11:57:34 I: mDNS: started
2024-12-30 11:57:34 I: Webserver: started
2024-12-30 11:57:34 I: MQTT: connected, System config: 1
2024-12-30 11:57:34 I: I2C init: QueryDevicetype - mfr:0x00 type:0x2B fw:0x04 hw:0x0A
2024-12-30 11:57:34 I: Setup: CC1101 RF module found, chip version: 0x14
2024-12-30 11:57:31 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2024-12-30 11:57:31 I: WiFi: connection successful
2024-12-30 11:57:26 I: wifi AP mode started
2024-12-30 11:57:25 E: Unable to set wifi disconnect
2024-12-30 11:57:22 I: I2C sniffer capable hardware: yes
2024-12-30 11:57:22 I: HW rev: NON-CVE 1, FW ver.: 2.9.0-beta10
2024-12-30 11:57:22 I: Device UUID: df15c2bb-d2f7-4944-8e41-21eb26592412
2024-12-30 11:57:21 I: System boot, last reset reason: SW_RESET
2024-12-30 11:57:02 I: Firmware update: nrgitho-v2.9.0-beta10.bin
2024-12-14 22:51:43 I: I2C init: QueryStatus
2024-12-14 22:51:43 I: Hostname: hru
2024-12-14 22:51:42 I: mDNS: started
2024-12-14 22:51:42 I: Webserver: started
2024-12-14 22:51:42 I: MQTT: connected, System config: 1
```

arjenhiemstra, 2024-12-31:
Thanks for reporting this but I cannot reproduce it. Tested with beta10 and 2.8.0 result in the same behaviour. I can also not find any relevant code changes in the MQTT callback function between these versions. 

The only thing I notice in your log and what might have changed due to the lib ArduinoJSON being updated in the mean time; JSON parsing might be more strict. If you actually feed this command to the MQTT API:
`{ rfremotecmd: 'autonight'}`

That is not a valid JSON

Please try this instead:
`{"rfremotecmd":"autonight"}`

jasperslits, 2024-12-31:
Does not make a difference: nothing happens. But upon further testing there's nothing happening when using the buttons in the add-on. Not on the virtual remote nor under RF devices. 

Could pairing be 'broken' somehow?

arjenhiemstra, 2024-12-31:
What remote type have you configured?

jasperslits, 2024-12-31:
RFT CO2 as vremote and same under RF devices.

arjenhiemstra, 2024-12-31:
OK, thanks! Thats useful info. I'll test with the CO2 remote. 
Have you configured bi-directional?

jasperslits, 2024-12-31:
The RF devices are in monitoring only. And they have been like that since day 1.

closed 2025-03-08 by stale[bot]