arjenhiemstra/ithowifi #288: 2.9.0b10 Retrieve ITHO settings not working

ibjr, 2025-01-02:
**Describe the bug**
After upgrading to 2.9.0-b10 it is no longer possible to retrieve the ITHO settings. When clicking the Retrieve Settings button, it does show the ITHO settings, but this is the data stored in the browsers cache. When you clear the browser cache, no data is shown; also the option the refresh the cache (as is previous versions) is not shown. After downgrading to 2.8, everything works as expected. (tested upgrading twice). 
(When there is still old cached data, changing or retrieving a single item  has no effect ("moving dots" forever))

**To Reproduce**
Steps to reproduce the behaviour:
0: Upgrade to 2.9.0 beta 10 and clear browser cache and login with browser
1. Go to 'Itho Settings'
2. Click on 'Retrieve settings'
4. Settings are not retrieved

**Expected behaviour**
Full list of ITHO settings

**Device information**
- Firmware version: 2.9.0 - beta 10
- Hardware revision: non-CVE 1
- Type/model number Itho HRU 350 
- CC1101 RF module enabled: yes

**Debug logging**
2025-01-02 16:18:16 I: I2C init: QueryStatus
2025-01-02 16:18:16 I: I2C init: QueryStatusFormat - items:33
2025-01-02 16:18:16 I: I2C init: QueryDevicetype - mfr:0x00 type:0x2B fw:0x04 hw:0x0A
2025-01-02 16:18:16 I: Setup: done
2025-01-02 16:18:15 I: Hostname: nrg-itho-810c
2025-01-02 16:18:15 I: mDNS: started
2025-01-02 16:18:15 I: Webserver: started
2025-01-02 16:18:14 I: RF debug level = 0
2025-01-02 16:18:14 I: rfsetup: module_rf_id 0x9C,0x81,0x0B
2025-01-02 16:18:14 I: Setup: CC1101 RF module found, chip version: 0x14
2025-01-02 16:18:14 I: Setup: Virtual remotes, start ID: 9C,81,0C - No.: 1
2025-01-02 16:18:13 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2025-01-02 16:18:13 I:   MAC:08:B6:1F:9C:81:0C
2025-01-02 16:18:13 I:   IP:192.168.199.72
2025-01-02 16:18:12 I:   Status:WL_CONNECTED
2025-01-02 16:18:12 I:   Mode:STA+AP
2025-01-02 16:18:12 I:   RSSI:-42dBm
2025-01-02 16:18:12 I:  — removed —
2025-01-02 16:18:12 I: WiFi info:
2025-01-02 16:18:11 I: WiFi: connection successful
2025-01-02 16:18:06 I: wifi AP mode started
2025-01-02 16:18:05 I: Setup: starting wifi access point
2025-01-02 16:18:05 I: WiFi mode WIFI_AP_STA
2025-01-02 16:18:05 E: Unable to set wifi disconnect
2025-01-02 16:18:02 I: I2C sniffer capable hardware: yes
2025-01-02 16:18:02 I: HW rev: NON-CVE 1, FW ver.: 2.9.0-beta10
2025-01-02 16:18:02 I: Device UUID: 91acd14b-f5ab-4d9e-a1fa-f07f606d6291
2025-01-02 16:18:01 I: System boot, last reset reason: SW_RESET
2025-01-02 16:17:57 I: Reboot requested
2025-01-02 13:24:39 I: I2C init: QueryStatus
2025-01-02 13:24:39 I: I2C init: QueryStatusFormat - items:33
2025-01-02 13:24:37 I: I2C init: QueryDevicetype - mfr:0x00 type:0x2B fw:0x04 hw:0x0A
2025-01-02 13:24:33 E: I2C init: QueryDevicetype - failed
2025-01-02 13:24:33 I: Setup: done
2025-01-02 13:24:33 I: Hostname: nrg-itho-810c
2025-01-02 13:24:33 I: mDNS: started
2025-01-02 13:24:33 I: Webserver: started

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser: chrome and safari
 - Version latest

**Smartphone (please complete the following information):**
 - Device: iPad Pro
 - OS: iOS 18.2
 - Browser safari
 - Version [e.g. 22]

arjenhiemstra, 2025-01-02:
Thanks for spotting this! I've found where the issue is and will fix it in the next release

0crap, 2025-03-10:
Let's kick the stale bot around a bit here.

arjenhiemstra, 2025-03-10:
sorry, due to a renovation project, health issues and a recent relocation I wasn't able to work on any issues. I'll have a look asap at this issue again!

0crap, 2025-03-11:
No worries, take care!

closed 2025-04-16 by arjenhiemstra