arjenhiemstra/ithowifi #294: Settings not saved when powercycling (2.9.0 Beta 9 and 10)

Arthur-de-Partuur, 2025-02-03:
**Problem**
For 2.9.0 beta (9 and 10) Upon power reset (off-wait30sec-on) some settings are lost, such as:

- WebAPI version (reverts back to 1.0)
- Itho status update frequency (0-65535 seconds): reverts back to 60 sec
- possibly others as well but did not test

**Steps to reproduce the behaviour:**
1. Power cycle demandflow
2. wait 20 sec
3. power on, wait untill fully operational (I wait for 5 minnutes)
4. Check system setting tab

**Expected behaviour**
I expect the  the settings (particulary api and update frequency) to stay at the set point.
Current behavious causes issues with HomeAssistant / Node red

**Screenshots**

**Device information**
- Firmware version [e.g. 2.9.0 beta9 and beta 10]
- Itho device type: DemandFlow
- Itho hw version: 42 (non-cve)
- Itho fw version: 21
- Manufacturer: 0
- Hardware revision (e.g. non-cve 1.1)
- CC1101 RF module enabled: **yes**

**Debug logging**
[logfile0.current.log](https://github.com/user-attachments/files/18642012/logfile0.current.log)

**Desktop (please complete the following information):**
 - OS: win11
 - Chrome and Safari (latest update)

Kuchiru, 2025-04-10:
I just found out the same yesterday, mqtt user/pw was lost.

arjenhiemstra, 2025-04-12:
Fix available in https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.9.0-beta11

commit dfe30197 (2025-04-12): Fix: config not saved when config is done in a certain order, fixes: #294
closed 2025-04-12 by arjenhiemstra