arjenhiemstra/ithowifi #295: Status not updaing

epalmans, 2025-02-13:
After several days or weeks, status for humidy or temperature (or both) suddenly seems to freeze (not updating). Restarting the module each time solves it again temporarily.
I currently have two CVE-S ECO boxes installed. Only one of them is having this issue

**Screenshots**
[image]

**Device information**
- Firmware version 2.8.0
- Hardware revision 2
- Type/model number Itho: CVE-S ECO HE, 03-00402
- CC1101 RF module enabled: no

Crashlog:
note: restarted right before downloading this. Ie., may or may not include necessary debug info (?)

[coredump_bin.txt](https://github.com/user-attachments/files/18780583/coredump_bin.txt)

Youknowhooo, 2025-02-22:
As of Sunday, I have exactly the same issue.

Device information

Firmware version 2.9.0-beta10 (as of today)
Hardware revision 2
Type/model number Itho: CVE-S ECO 

[image]

sjampeter, 2025-02-27:
i have the same issue. i thougt it was a sensor fault.
i bought a new one, but nothing hapens.
even after e reset the sensor is frozen.

[image]

something in the log i'm missing?

joho500, 2025-03-13:
I have the same problem for a couple of months with the values of a RFT-CO2 sensor connected to the Ithowifi. Even purchased a new sensor but this made no difference. The last week it appeared to work but today I see a lot less frequent updates in Home Assistant. This behavior I saw before with then dropping to no updates at all.

Device information:
- Current firmware version: 2.8.0
- Hardware revision: 2 
- CC1101 RF module enabled: yes
- Itho device type: CVE-Silent
- Itho hw version: 25
- Itho fw version: 11

Marc-David, 2025-04-02:
Same issue here, I tested with firmware 2.6.0, 2.8.0 and 2.9.0-beta10.

The data on mqtt stops updating after 12 to 48 hours. MQTT status shows connected, ITHO I2C status shows connected but the values on the Itho status page do not update anymore. The System Logs shows the following lines every 10 seconds or so:

```
2025-04-02 13:11:17 W: Warning: I2C timeout, trying I2C bus reset...
2025-04-02 13:11:10 E: Error: I2C bus could not be cleared!
```

Restarting the device via the button in the webinterface restarts the device but doesn't resolve the issue. The only thing that helps is unplugging/plugging the whole Itho.

Device information:

    Current firmware version: 2.6.0 (but 2.8.0 and 2.9.0-beta10 behave the same way).
    Hardware revision: 2
    Itho device type: CVE-Silent
    Itho fw version: 27

Happy to help debugging further, upload and relevant logs, etc.

arjenhiemstra, 2025-04-02:
> Same issue here, I tested with firmware 2.6.0, 2.8.0 and 2.9.0-beta10.
> 
> The data on mqtt stops updating after 12 to 48 hours. MQTT status shows connected, ITHO I2C status shows connected but the values on the Itho status page do not update anymore. The System Logs shows the following lines every 10 seconds or so:
> 
> ```
> 2025-04-02 13:11:17 W: Warning: I2C timeout, trying I2C bus reset...
> 2025-04-02 13:11:10 E: Error: I2C bus could not be cleared!
> ```
> 
> Restarting the device via the button in the webinterface restarts the device but doesn't resolve the issue. The only thing that helps is unplugging/plugging the whole Itho.
> 
> Device information:
> 
> ```
> Current firmware version: 2.6.0 (but 2.8.0 and 2.9.0-beta10 behave the same way).
> Hardware revision: 2
> Itho device type: CVE-Silent
> Itho fw version: 27
> ```
> 
> Happy to help debugging further, upload any relevant logs, etc.

Could you provide the syslog?

gpvelzen, 2025-09-02:
I got the same problem:
```
Configuration of the Itho unit

Itho device type: CVE-Silent
Itho hw version: 46
Itho fw version: 25
Manufacturer: 0
```
2025-09-02 14:25:11 W: Warning: CC1101 Task timed out!
2025-09-02 14:25:07 E: Error: I2C bus could not be cleared!

Happens after 2 or 3 weeks

polroskam, 2026-10-23:
Found why the values freeze until a restart. I checked it against Version-3.2.0, and develop (313bf60) has the same code.

In [IthoStatus.cpp L252-L283](https://github.com/arjenhiemstra/ithowifi/blob/Version-3.2.0/software/NRG_itho_wifi/main/ithodevice/IthoStatus.cpp#L252-L283), a RelativeHumidity or Temperature value outside 1 to 100 switches that status field's type to `is_string` with the value "not available". The update loop above it ([L186-L232](https://github.com/arjenhiemstra/ithowifi/blob/Version-3.2.0/software/NRG_itho_wifi/main/ithodevice/IthoStatus.cpp#L186-L232)) only handles `is_byte`, `is_int` and `is_float`, so the field is never parsed again. `ithoHum` and `ithoTemp` keep their last value, and so do `hum`, `temp` and `ppmw`. Only a rebuild of the status format resets the type, and that happens at boot.

One bad read is enough to latch it. That matches the reports here: random intervals, fixed by a restart, nothing in the log. Nothing logs the switch, or the raw value behind it.

Possible fix: keep the type and treat an out-of-range value as invalid for that poll only, for example with a flag that the JSON output turns into "not available". The next good read then recovers by itself. Logging the raw value would also show what the bad reads look like.

My setup: CVE-S Optima (CVE1B, Itho fw 27, hw 57), add-on 3.2.0, status via 2401 every 10 s, I2C safe guard active. It latched three times in four weeks. The `statusformat` debug command should clear it without a reboot (#61 reports the same). I haven't tried that yet.

arjenhiemstra, 2026-10-23:
> 12\. Found why the values freeze until a restart. I checked it against Version-3.2.0, and develop ([313bf60](https://github.com/arjenhiemstra/ithowifi/commit/313bf60eca8824fcde67c03f9168aaa29805d7b7)) has the same code.
>      In [IthoStatus.cpp L252-L283](https://github.com/arjenhiemstra/ithowifi/blob/Version-3.2.0/software/NRG_itho_wifi/main/ithodevice/IthoStatus.cpp#L252-L283), a RelativeHumidity or Temperature value outside 1 to 100 switches that status field's type to `is_string` with the value "not available". The update loop above it ([L186-L232](https://github.com/arjenhiemstra/ithowifi/blob/Version-3.2.0/software/NRG_itho_wifi/main/ithodevice/IthoStatus.cpp#L186-L232)) only handles `is_byte`, `is_int` and `is_float`, so the field is never parsed again. `ithoHum` and `ithoTemp` keep their last value, and so do `hum`, `temp` and `ppmw`. Only a rebuild of the status format resets the type, and that happens at boot.
>      One bad read is enough to latch it. That matches the reports here: random intervals, fixed by a restart, nothing in the log. Nothing logs the switch, or the raw value behind it.
>      Possible fix: keep the type and treat an out-of-range value as invalid for that poll only, for example with a flag that the JSON output turns into "not available". The next good read then recovers by itself. Logging the raw value would also show what the bad reads look like.
>      My setup: CVE-S Optima (CVE1B, Itho fw 27, hw 57), add-on 3.2.0, status via 2401 every 10 s, I2C safe guard active. It latched three times in four weeks. The `statusformat` debug command should clear it without a reboot ([#61](https://github.com/arjenhiemstra/ithowifi/issues/61) reports the same). I haven't tried that yet.

Very nice catch!! I'll release a fix asap!

arjenhiemstra, 2026-10-27:
Fixed in develop with the commits from #394 (thanks @polroskam). A status field now recovers on the next valid reading instead of staying "not available" until a reboot. Will be in 3.3.0-beta3.

polroskam, 2026-10-28:
Captured it. With the fix from #394 (in 3.3.0-beta3), a bad read now clears on the next poll, and on my unit I log the add-on's copy of the reply next to the bus copy from the add-on's own sniffer. Five bad replies since yesterday 14:26, out of 4978 status replies. In all five the bus carried a correct reply, and the add-on received it with one 0x00 missing and a 0x00 appended:

```
bus      80 82 A4 01 01 17 0A 03 8D 03 93 00 00 08 00 4B 00 00 7A 68 00 00 01 A2 31 13 75 08 B7 C1
add-on   80 82 A4 01 01 17 0A 03 8D 03 93 00 00 08 00 4B 00 00 7A 68 00 01 A2 31 13 75 08 B7 C1 00
```

The missing zero sits in a different field each time (Error, Startup counter, operating hours, Absence). The reply keeps its length and passes `checksum()`, since dropping one 0x00 and appending one leaves a two's-complement sum unchanged. So the byte is lost in the add-on's receive path (`i2cSlaveReceive` and the IDF slave driver), not on the bus and not by the unit.

The rate is telling: with the safe guard on I saw one bad reply in about 55 hours; with the sniffer on (and so the safe guard off), five in 14 hours. The bus copies show no collision, which points at the sniffer's level-3 interrupt itself, busy-waiting through each transaction while the slave receives, rather than at bus timing. I'll test that next by pausing the sniffer interrupt during `i2cSlaveReceive`, with the sniffer otherwise on, and post the result here.

polroskam, 2026-10-31:
The pause works. I paused the sniffer interrupt while the add-on sends its status query and receives the reply, and left the sniffer on otherwise. Zero bad replies in 31 hours, out of about 11200 status replies. Before, with the same setup, it was six in 5699 over 16 hours.

The polls kept working: a status query every 10 s, `I2C_OK` throughout in the I2C debug log, and Home Assistant got humidity updates at the same rate as before. So bad replies did not turn into dropped ones.

So the byte is lost when the sniffer's interrupt runs during the slave receive. That probably also explains the rare bad reply with only the safe guard on: the guard keeps the sniffer interrupt enabled outside its 7.5 s windows, so a status query can still overlap it. At one in about 55 hours that's too rare to confirm in a test this short.

Want a PR for this? It would pause the interrupt around every query that waits for a reply, so the 2410 settings reads and writes are covered too.

arjenhiemstra, 2026-10-31:
Nice! I you can have a PR generated for this, that would be nice thanks!

GHBLoos, 2026-11-02:
I am not sure it is related, but the humidity is giving a lot of spikes down recently
[image]

[image]

polroskam, 2026-11-03:
Result from #403 is in: zero bad replies in 24 hours with the sniffer on, against 6 in 5699 before. Details in https://github.com/arjenhiemstra/ithowifi/pull/403#issuecomment-6597073237

arjenhiemstra, 2026-11-03:
Fixed in 3.3.0-beta7: the sniffer interrupt is paused while the add-on receives a reply (#403, thanks @polroskam). Result of the 24 hour test in #403: zero bad replies, against 6 in 5699 before. Closing, feel free to reopen if it shows up again.

closed 2026-11-03 by arjenhiemstra