arjenhiemstra/ithowifi #297: Non-CVE module does not work on HRU-3

tparvais, 2025-02-21:
I got my non-cve module last week

When powered vi usb, I succesfully get the connection to the web UI and I can configure the Wifi setup.

Then I connected directly with straight UTP ethernet cable to my ITHO COM port. No led blinking on RJ-45 female port, nothing on the motherboard, no activity.

I obvioulsy tried several utp cable.

I read that I should not power the module via usb when connected to COM port. So I didn't try this.

Should I consider that this non-CVE module is not compatible with my HRU-3 ?
Or is there a bug somewhere ? or a default ?

Sorry to jeopardize this github, I didn't have any feedback from vendor email nor via forum

arjenhiemstra, 2025-03-10:
Thx for this issue. Could you give a bit more info on the exact version of your HRU? Is there anything that indicated a more specific version on your device maybe?

closed 2025-05-16 by stale[bot]

basrieter, 2025-11-13:
I am wondering the same. HRU-3 SBL-000 here:

[image]