arjenhiemstra/ithowifi #300: Hardware identifiers

benjamin-dcs, 2025-02-24:
Could you help me out identifying all the possible hardware/firmware identifiers? I can't find them in your code. 

(The number that's send in the MQTT deviceinfo topic)

benjamin-dcs, 2025-02-27:
Now I'm doubting if it's the numbers I need. I want to distinguish all possible HW versions (autotemp, cve, hru eco, 250, etc). What field do I need from the `deviceinfo` topic and what values can I expect? TIA!

benjamin-dcs, 2025-02-28:
For now I'm assuming the strings from (after the hex numbers):
https://github.com/arjenhiemstra/ithowifi/blob/92fb6de8a9ce1899777ff1439f7f5188c6c476a9/software/NRG_itho_wifi/main/IthoSystem.cpp#L78-L100

Any way to distinguish the 250 and 300? With `itho_hwversion` maybe?

arjenhiemstra, 2025-03-10:
You found the correct piece of code. These are the known hw identifiers. As far as I know there is no way to distinguish a hru250 from a hru300. It could be that the versions  as defined here:
https://github.com/arjenhiemstra/ithowifi/blob/92fb6de8a9ce1899777ff1439f7f5188c6c476a9/software/NRG_itho_wifi/main/devices/hru250_300.h

could mean that ie. version 1-5 == hru250 and the remaining versions == hru300 but I have no information available atm the determine this.

closed 2025-04-18 by benjamin-dcs