arjenhiemstra/ithowifi #301: Zomernachtverlaging

mvdh02, 2025-02-28:
Beste Arjan,
Is het mogelijk om in de software iets te bouwen wat lijkt op de zomernachtverlaging die Orcon. U standaard levert in zijn nieuwe producten?
(Orcon MVS-15RH CO2B Max Zomernacht Comfort)

Als het in de zomermaand binnen meer dan 23 graden is en de buitentemperatuur is ‘s nachts lager gaat hij bewust de Mv box inzetten om koelere lucht via de raamroosters naar binnen te trekken.

Mogelijk met een externe bron als buitentemperatuur in plaats van een extra buitensensor.

Natuurlijk is dit prima aan te sturen met Domoticz of een andere home service. Maar standalone zou fijn zijn.

arjenhiemstra, 2025-03-10:
dat kan uiteraard maar er zijn zoveel van dit soort use cases en zoveel ondersteunde apparaten dat de firmware (naar mijn mening) te groot en te complex om te onderhouden zou worden. Helaas ben je dus aangewezen op automations via een domotica pakket (ie. domoticz) of zelf de functionaliteit in te bouwen (fork op deze repo bijvoorbeeld)

arjenhiemstra, 2025-05-09:
Dat zou kunnen maar ik probeer zoveel mogelijk van dit soort logica uit de firmware te laten (iig voor nu). 
Er zijn zoveel verschillende apparaten en binnen een apparaat groep weer verschillende versies met verschillende eigenschappen dat ik denk dat dit soort functionaliteit inbouwen (niet alleen dit maar er zijn vele verschillende vragen geweest) de firmware te complex zal maken.

closed 2025-07-15 by stale[bot]