arjenhiemstra/ithowifi #303: Feature request: Connect to Wpa3 wifi

WilcoSnel, 2025-03-10:
Cannot connect to Wpa3 only WiFi.

arjenhiemstra, 2026-10-30:
The add-on uses an ESP32, not an ESP8266 like the D1 mini in the links above, and it supports WPA3-Personal. Since 3.3.0-beta3 the WiFi scan shows WPA3 networks as secured, and the WiFi setup page has a "WiFi security" row with the security in use (WPA2-PSK, WPA3-PSK, ...).

Could you try the current beta on your WPA3-only network? If it connects, the row should read WPA3-PSK.

WilcoSnel, 2026-10-31:
I've updated to 3.3.0 beta 6, changed my 2.4ghz network to mixed wpa2/3 and after a reboot it now shows this in the wifi settings. I then created a test 2.4ghz with forced wpa3. When I tried to change it to that network it only shows the error: failed saving wifi settings, unable to write the config file. Tried this several times including reboots, so cannot test it further.

arjenhiemstra, 2026-10-31:
Thanks for testing. The error is not WPA3 related: the add-on could not open its WiFi config file for writing, so the new SSID was never saved. Could you check a few things?

- Debug page: how much of the filesystem is in use (used / total)?
- Does saving other settings work, e.g. a change on the System settings page?
- Does saving the WiFi page work if you leave the SSID and password unchanged?
- The system log lines from right after a failed save (there should be a "failed to create /wifi.json" line, maybe more).

And the screenshot of the mixed WPA2/WPA3 network didn't come through. What does the WiFi security row show there?

WilcoSnel, 2026-11-01:
- File system: 28672 bytes used / 131072 bytes total
- Yes, changing system settings works
- No, saving wifi settings without changes results in the same error
- error in log shows: 2026-11-01 09:15:54 E: SYS: error - failed to write /wifi.json file

Image of mixed wpa2/3 attached below

[image]

arjenhiemstra, 2026-11-02:
Thx! So the filesystem has plenty of room and other settings save fine, only /wifi.json can't be written. The add-on saves a config file by writing /wifi.json.tmp first and then renaming it to /wifi.json, so most likely one of those two files is stuck or damaged.

Could you try this:

Open http://<ip of the add-on>/edit.html. That's the file editor. If you see /wifi.json.tmp or /wifi.json.bad there, delete it and try saving the WiFi settings again.
If that doesn't help, download the config files first (config.json, remotes.json, vremotes.json, hadisc.json, syslog.json, and wifi.json if possible), then use "Format filesystem" on the Reset page. The add-on restarts in access point mode. Set up WiFi there, then upload the files again through edit.html and reboot.
After that the WPA3-only network should be worth another try. Let me know what the file list shows, it helps to find the cause.

arjenhiemstra, 2026-11-03:
Found it, thanks for the details. Since 3.3.0-beta4 saving the WiFi settings always fails with "Unable to write config file", with or without changes. The add-on kept `/wifi.json` open while saving it, so the file could not be replaced. Your filesystem is fine, no need to format it.

The fix is in develop and will be in the next release. After that, could you try the WPA3-only network again? The WiFi security row should then read WPA3-PSK.

WilcoSnel, 2026-11-03:
Perfect! Thank you. When the new release is out I'll check again :)

arjenhiemstra, 2026-11-05:
Fixed in 3.3.0-beta8: saving the WiFi settings works again, and a WPA3-only network connects here. Closing, reopen if yours doesn't connect. The WiFi security row should read WPA3-PSK.

closed 2026-11-05 by arjenhiemstra

WilcoSnel, 2026-11-06:
Yep, works! Thanks :D

[image]