arjenhiemstra/ithowifi #305: Feature request: Cannot connect to hidden WiFi network (disabled Ssid broadcast)

WilcoSnel, 2025-03-10:
**Describe the bug**
When a WiFi network is configured as hidden, the SSID isn't broadcasted. Therefor the WiFi network cannot be selected when scanning for it and cannot be connected to. 

**To Reproduce**
-

**Expected behaviour**
Usually you can just manually enter the hidden ssid details to connect to a hidden network. But that's not working here. Log shows something like no broadcasting ssid found.  

**Screenshots**
-

**Device information**
- Firmware version 2.8.0
- Hardware revision non-cve 1
- Type/model number Itho WPU
- CC1101 RF module enabled [ no ]
- In case of wifi issues, wifi access point make and model:

**Debug logging**
Please provide debug logging from the debug page if possible:

arjenhiemstra, 2025-03-10:
again little information to go by unfortunately

> Log shows something like no broadcasting ssid found.

Please, do not expect me to guess what log message this is...

And further more:

Which wifi channel are you using? 
What is the SSID name you tried? (you could try to change your wifi ssid and test a connection temporary if you do not want to share the real name)
what kind of wifi devices do you have (make/model, its in the question list for a reason)
do you have multiple access points? etc.

I do not experience this issue nor can I reproduce the issue with the information you provided.

WilcoSnel, 2025-03-11:
Apologies. The exact errors are:

480016 E: Setup: wifi not connected - WL_CONNECT_FAILED
    420134 E: Reconnect WiFi failed!
    419980 E: Setup: wifi not connected - WL_NO_SSID_AVAIL
    361970 E: Reconnect WiFi failed!

And then a bunch of WL_CONNECT_FAILED until I retry another network.

I fixed this like I usually do. Changed wifi settings to make ssid visible. It then successfully connected. After that I hid the SSID again and then keeps working. But this is a slow and annoying process. So therefor I created this feature request to get this working with a hidden ssid 'out of the box'.

As far as my network details. Hardware is a Netgear Orbi Pro. My main WiFi is a 5ghz wpa3 (no mixed WPA2 MODE) mesh network. This didn't connect because of either 5ghz and/or wpa3, probably the latter I'm guessing. Created a seperate feature request for that as well. 

As this happens a lot, my secondary WiFi is a hidden 2.4 GHz wpa2 mesh network.

Hopefully this is enough info to go on. If not, please let me know. Again apologies for the missing info.

allardhoeve, 2025-03-16:
To provide some counterpoint anecdata: I connected my module to a hidden network just fine. More info available if necessary.

TD-er, 2025-03-16:
When connecting an ESP to a hidden-SSID network, you need to set a flag to also include for hidden SSID during scan.
Also keep in mind there is a difference between hidden-SSID networks on channel 1... 11 or those running on channels 12 .. 14  (passive vs. active scan)

Some access points, like Mikrotik, do only allow you to connect when you explicitly use SSID, MAC-address and channel to call `WiFI.begin()`, while others also allow for connecting using a broadcast connect request.
And for some you may need to call `delay(1000)` right after calling `WiFi.begin()` to make sure the ESP will reply promptly to acknowledgements of the (hidden SSID) AP.

@arjenhiemstra 
Be aware you will open Pandora's box when starting to support hidden SSID.

closed 2025-05-22 by stale[bot]