arjenhiemstra/ithowifi #307: Can anyone with a HRU 300 WTW validate these read-outs?

mkleijn, 2025-03-24:
Hi, I successfully connected the non-CVE module to my HRU 300, partly because I want to monitor the WTW performance in HA but also because I wanted to get more information about the error I received on the display. I suspected on or more temp sensors where malfunctioning and the readouts seem to prove that:

| Label                                                 | Value    |
|-------------------------------------------------------|----------|
| Error number                                          | 53       |
| Status                                                | 0        |
| Measured outside temperature (°C)                     | -327.68  |
| Measured temperature of mixed outside air (°C)        | -327.68  |
| The flow of the inflated air (M3/h)                   | 0        |
| Inlet temperature (°C)                                | 25.2     |
| Temperature of the extracted air (°C)                 | 25.17    |
| Temperature of the blown out air of the house (°C)    | 21.63    |
| The flow of the blown air (m3/h)                      | 0        |
| Relative fanspeed (%)                                 | 99.5     |
| Absolute speed of the fan (%)                         | 99.5     |
| Hysteresis use in control mode (K)                    | 0        |
| Timer for how long the house is cooled (hour)         | 0        |
| Timer for how long the house is heated (hour)         | 16       |
| The mass flow of the air entering the house (kg/h)    | 0        |
| The mass flow of the air leaving the house (kg/h)     | 0        |
| Percentage that the bypass valve is open (%)          | 100      |
| The desired inlet temperature (°C)                    | 20       |
| Hysteresis of the frost (K)                           | 0.5      |
| Number of steps the frost valve is open (steps)       | 0        |
| Number of hours of too cold air (hour)                | 6        |
| Temporary speed reduction (rpm)                      | 0        |
| Sample timer in frost mode (min)                      | 0        |
| RPM of the motor (rpm)                                | 2095     |
| Measured waste temperature heated NTC (°C)            | -327.68  |
| Measured blend temperature heated NTC (°C)            | 327.67   |
| Desired capacity (m3/h)                               | 0        |
| Busy doing adjustments (-)                            | 0        |
| Current consumption of fan (mA)                       | 11334    |
| Desired current consumption of fan (mA)               | 0        |
| Highest measured RH (%)                               | 0        |
| Highest measured CO2 (ppm)                            | 0        |

Outside temp, mixed air temp and the waste/blend(?) temp sensors give a invalid values of (-)32K but I know some values are "supposed" to be missing for this model. Of course the error code 53 is a giveaway as well, but I can't seem to find a reference list of these codes on the internet.

Anyway, the whole thing is placed in a not so serviceable location so when I go and try to open the thing up to locate the broken or dirty sensors I prefer to know where to look. Does anyone have a fully functional HRU300 that can share their status readouts for me to compare, or anyone has any info for me that can help me pinpoint the faulty sensors?

mkleijn, 2025-03-24:
Great, thanks. I found that replacement kit (price is ridiculous if you look at the electronics involved, but guess that's how it is with OEM replacement kits). Unfortunately I just left my extended warranty on parts slip by a few months. Sounds like a design fault if the original sensors can't survive the environment they are installed in. Anyway, small hope its just a bad connection or something.

EdwinGH, 2025-03-24:
The properties of my unit:
Itho device type: HRU 250-300
Itho hw version: 6
Itho fw version: 7
Manufacturer: 7

It's current values are as follows: (I think the 25.xx values are sensors replaced with resistors):
Label | Value
-- | --
Error number | 2
Status | 5
Measured outside temperature (°C) | 16.48
Measured temperature of mixed outside air (°C) | 16.11
The flow of the inflated air (M3/h) | 404.5
Inlet temperature (°C) | 20.25
Temperature of the extracted air (°C) | 20.43
Temperature of the blown out air of the house (°C) | 25.12
The flow of the blown air (m3/h) | 307.6
Relative fanspeed (%) | 0.5
Absolute speed of the fan (%) | 4
Hysteresis use in control mode (K) | 0
Timer for how long the house is cooled (hour) | 0
Timer for how long the house is heated (hour) | 0
The mass flow of the air entering the house (kg/h) | 491.93
The mass flow of the air leaving the house (kg/h) | 362.66
Percentage that the bypass valve is open (%) | 74.5
The desired inlet temperature (°C) | 22
Hysteresis of the frost (K) | 0
Number of steps the frost valve is open (steps) | 1400
Number of hours of too cold air (hour) | 0
Temporary speed reduction (rpm) | 0
Sample timer in frost mode (min) | 0
RPM of the motor (rpm) | 356
Measured waste temperature heated NTC (°C) | 25.27
Measured blend temperature heated NTC (°C) | 25.25
Desired capacity (m3/h) | 0
Busy doing adjustments (-) | 0
Current consumption of fan (mA) | 535
Desired current consumption of fan (mA) | 0
Highest measured RH (%) | 0
Highest measured CO2 (ppm) | 704

</body>
</html>

GraceGRD, 2025-03-25:
I no longer have the ithowifi module connected to my HRU and read-out my sensors wirelessly.

Recently I got my sensors replaced as well. 
* Outside temperature and temperature blown into the home have the same value.
* Indoor temperature seems correct
* Blow out temperature seems fixed around 25 deg (looks like a fixed resistor).

mkleijn, 2025-03-25:
Thanks @EdwinGH and @GraceGRD for your reply! I have the same HW and FW versions as EdwinGH mentioned. When I compare my values against his combined what GraceGRD mentions, it seems like my sensors for outside temp and mixed outside air are toasted. And the ones on the heated NTC, but these are replaced by a resistor anyway apparently.

Also I have no values for the flow of inflated and blown air. Anyone knows how this is measured? I don't believe these sensors are part of the sensor replacement pack, or are they?

Maybe it is not measuring because the whole thing is in an error state.

mkleijn, 2025-04-02:
Just a quick update, managed to open the unit and clean some of the sensors and connectors in the fan part of the unit.
Seems to have done something as the "Measured blend temperature heated NTC (°C)" is now showing a valid value.

Outside, mixed and waste are still reporting as -327.68 so I guess those sensors just died. Time to order that replacement kit. Apparently it is a design mistake, the open sensors are prone to corrosion

GraceGRD, 2025-04-15:
They have a warranty check on their website: 
https://www.ithodaalderop.nl/nl-NL/professional/garantiecheck

Standard 2y warranty on parts / labor costs.
If you register your product the parts waranty is extended to 5y.

I had to contact a 3rd party certified installer to get the sensors replaced (Itho no longer does it themselves in our area). This 3rd party installer shipped back the sensors to Itho and got it replaced under waranty. Sadly not the laber cost as my unit was 3y old at the time of service.

Hopefully this info helps.

arjenhiemstra, 2025-04-15:
> > Apparently it is a design mistake, the open sensors are prone to corrosion
> 
> So then the manufacturer should pay, no? 

Unfortunately, manufacturer warranties are not legally bound. 

The only warranties statutory guaranteed are between seller and buyer and in the EU the minimal statutory guaranteed duration of the warranty is 2 years, for consumers.

In The Netherlands there is no fixed maximum duration defined (ie. it can extend for way more than 2 years depending on the expected life-cycle of a product).  
Manufacturer warranties are an "extra" service from the manufacturer (this will differ per country based on local legislation).
If you buy a product directly from a manufacturer (ie. you order a product from Itho instead of buying at company abc), the manufacturer is of course also the seller and in that the manufacturer is also bound to the same consumer warranty rules.
All of this does not apply to B2B sales btw.

> Itho asks ridiculous prices for a temp sensor.

Without any doubt!

closed 2025-06-21 by stale[bot]