arjenhiemstra/ithowifi #313: HRU 350: Supply and Exhaust temperature readings seem to be flipped

dirixmjm, 2025-04-22:
**Describe the bug**
It seems the Supply and Exhaust temperature readings are flipped

**To Reproduce**
Compare temperature with outdoor temperature sensor.

[image]

device:
  identifiers: HRU
  manufacturer: Arjen Hiemstra
  model: Wifi add-on for Itho
  hw_version: NON-CVE 1
  sw_version: 2.8.0
  name: HRU

arjenhiemstra, 2025-04-25:
Could you give a copy of the itho status page of the add-on?

dirixmjm, 2025-04-27:
Something strange happened, not sure if this is due to the jaspers HA interface, or in the CVE, but yesterday evening at 21:50 temperatures seem to trade places. Now Exhaust seems to be exhaust. 

[image]

Here is the current status, there they seem the correct way around for now, which matches the above values.

[image]

I guess I'll have to keep tracking this.

arjenhiemstra, 2025-04-27:
I don't know for sure what this could be but I think it's happing in the itho firmware. All these values are read and passed as is, no post processing on any of these values. 
Could it be that the bypass postion changed and that this changes the temp reading?

dirixmjm, 2025-04-27:
That is exactly what happened, the bypass opened at 21:50.
Not sure why the temperature readout should flip, but probably as you mentioned an ITHO firmware issue.

Fredprivate, 2025-05-01:
I don't think this is an issue, the temperature readout do not flip, the actual temperatures flip. Once your bypass opens the heat exchanger is bypassed, so your supply temp will be (approximately) equal to the outside temp, basicly like opening a window. Function of this is to cool down your house during summernights. [the HRU has some algorithm to determine if a day is a summerday]

So expected behaviour is: 
Bypass open:
Exhaust temp =~ indoor temp
Supply temp =~ outdoor temp

Bypass closed (air flows through heat exchanger):
Exhaust temp =~ outdoor temp (this will not completely be truth because the heat exchanger does not have 100% efficiency)
Supply temp =~ indoor temp (same comment as above)

You can see it is not the readout that flips, but the actual temperatures flip.

sjengfred, 2025-05-01:
I don't think this is an issue, the temperature readout do not flip, the actual temperatures flip. Once your bypass opens the heat exchanger is bypassed, so your supply temp will be (approximately) equal to the outside temp, basicly like opening a window. Function of this is to cool down your house during summernights. [the HRU has some algorithm to determine if a day is a summerday]

So expected behaviour is:
Bypass open:
Exhaust temp =~ indoor temp
Supply temp =~ outdoor temp

Bypass closed (air flows through heat exchanger):
Exhaust temp =~ outdoor temp (this will not completely be truth because the heat exchanger does not have 100% efficiency)
Supply temp =~ indoor temp (same comment as above)

You can see it is not the readout that flips, but the actual temperatures flip.

dirixmjm, 2025-05-15:
OK, well this then depends on the location of the temperature sensors. According to your description this means the temperature sensors are after the heat exchanger instead of before at the actual supply / exhaust gates. If that is the case, then yes that makes sense.

closed 2025-05-15 by dirixmjm