arjenhiemstra/ithowifi #320: MQTT illegal discovery topic

ExquisiteMetropolis, 2025-05-17:
Since running Firmware: 2.9.0-beta12 this error is reported in the Home Assistant log for MQTT, the IP matches the Itho Addon:

`Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:393
integration: MQTT (documentation, issues)

Received message on illegal discovery topic 'homeassistant/device/<IP>/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic`

Unfortunately, it doesn't mention which config topic it exactly concerns.

arjenhiemstra, 2025-05-17:
Please include more info, I have no idea what to look for. There is a template for a reason

TD-er, 2025-05-17:
Looks like you have an empty sub-topic.
See the `//` in the topic.

ExquisiteMetropolis, 2025-05-17:
@arjenhiemstra updated my initial post.
@TD-er , seems that when you use < > in a code block, the content in between is removed, the // you noticed, contains the IP of the addon, which I removed and placed between < IP >

arjenhiemstra, 2025-05-17:
I think you have set the hostname of your add-on to an IP address? If that is the case, please change it to something that complies with standards.

Normally this line:

> homeassistant/device/*IP*/config

would be something like:
```homeassistant/device/nrg-itho-a2b3/config```

where ```nrg-itho-a2b3``` is the hostname configured. 

Hostnames are only allowed to have ASCII letters a through z (in a case-insensitive manner), the digits 0 through 9, and the hyphen-minus character ('-').

I expect this to be the reason for the HA error.

ExquisiteMetropolis, 2025-05-18:
Within the Wifi Setup of the Module, the hostname is set to "nrg-itho-eb18". This is also visible in the log I posted.

ExquisiteMetropolis, 2025-05-18:
When using MQTT explorer, I found there are two entries for the Itho unit:

homeassistant/device/IP & homeassistant/device/nrg-itho-eb18
Perhaps it is pushed twice? 

The entry with "IP", where IP is the internal IP of the controller:
```
{
  "avty_t": "itho/lwt",
  "stat_t": "itho/ithostatus",
  "dev": {
    "ids": "IP",
    "mf": "Arjen Hiemstra",
    "mdl": "Wifi add-on for Itho",
    "name": "CVE-Silent",
    "hw": "2",
    "sw": "2.9.0-beta11",
    "cu": "http://IP.local"
  },
  "o": {
    "name": "CVE-Silent",
    "sw": "2.9.0-beta11",
    "url": "https://www.nrgwatch.nl/support-ondersteuning/"
  },
  "cmps": {
    "1921683015_sensor_i_2": {
      "name": "ppmw",
      "p": "sensor",
      "uniq_id": "1921683015_sensor_i_2",
      "val_tpl": "{{ value_json['ppmw'] }}"
    },
    "1921683015_sensor_i_7": {
      "name": "AirQuality",
      "p": "sensor",
      "uniq_id": "1921683015_sensor_i_7",
      "val_tpl": "{{ value_json['AirQuality (%)'] }}",
      "unit_of_meas": "%"
    },
    "1921683015_sensor_i_9": {
      "name": "CO2level",
      "p": "sensor",
      "uniq_id": "1921683015_sensor_i_9",
      "val_tpl": "{{ value_json['CO2level (ppm)'] }}",
      "unit_of_meas": "ppm"
    },
    "1921683015_sensor_i_10": {
      "name": "Indoorhumidity",
      "p": "sensor",
      "uniq_id": "1921683015_sensor_i_10",
      "val_tpl": "{{ value_json['Indoorhumidity (%)'] }}",
      "unit_of_meas": "%"
    },
    "1921683015_sensor_i_28": {
      "name": "Fan speed",
      "p": "sensor",
      "uniq_id": "1921683015_sensor_i_28",
      "val_tpl": "{{ value_json['Fan speed (rpm)'] }}",
      "unit_of_meas": "rpm"
    },
    "1921683015_sensor_i_31": {
      "name": "Startup counter",
      "p": "sensor",
      "uniq_id": "1921683015_sensor_i_31",
      "val_tpl": "{{ value_json['Startup counter'] }}"
    },
    "1921683015_sensor_i_32": {
      "name": "Total operation",
      "p": "sensor",
      "uniq_id": "1921683015_sensor_i_32",
      "val_tpl": "{{ value_json['Total operation (hours)'] }}",
[... 318 lines cut ...]
        "High",
        "Auto",
        "AutoNight",
        "Timer 10min",
        "Timer 20min",
        "Timer 30min"
      ],
      "pl_off": "0",
      "pr_mode_cmd_tpl": "{%- if value == 'Timer 10min' %}{{'timer1'}}{%- elif value == 'Timer 20min' %}{{'timer2'}}{%- elif value == 'Timer 30min' %}{{'timer3'}}{%- else %}{{value.lower()}}{%- endif -%}",
      "pct_cmd_tpl": "{{ (value | int * 2.55) | round | int }}",
      "pr_mode_val_tpl": "{%- set speed = value_json['Ventilation setpoint (%)'] | int %}{%- if speed > 90 %}High{%- elif speed > 35 %}Medium{%- elif speed > 10 %}Low{%- else %}Auto{%- endif -%}",
      "pct_val_tpl": "{{ (value | int / 2.55) | round | int }}"
    },
    "cve-silent_fwupdate": {
      "name": "Firmware Update",
      "p": "update",
      "uniq_id": "cve-silent_fwupdate",
      "dev_cla": "firmware",
      "ent_pic": "http://192.168.30.15/favicon.png",
      "stat_t": "itho/deviceinfo",
      "val_tpl": "{{ {'installed_version': value_json['add-on_fwversion'], 'latest_version': value_json['add-on_latest_fw'], 'title': 'Add-on Firmware', 'release_url': 'https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.8.0' } | to_json }}"
    }
  }
}
```

arjenhiemstra, 2025-05-18:
Looking at this, the two items are separate items (ie. one created with beta11 and one with beta12)
 
In your first part:

`  "dev": {
    "ids": "IP",`

IP is generated here:
https://github.com/arjenhiemstra/ithowifi/blob/d96f7d92dd4cfd664fbf0f49d017a6460654f21b/software/NRG_itho_wifi/main/HADiscovery.cpp#L32

As you see, that is based on the hostname. That cannot be an IP address (at least not by default).

The beta12 discovery code seems fine. I can use it to discovery a device just fine:

[image]

ExquisiteMetropolis, 2025-05-19:
Well spotted on the beta11 entry. I've deleted that one via MQTT Explorer and gone is the error.
No idea how that got in there in the first place as I've never touched the hostname in the Wifi controller. 
Oh well, case closed. 👍