arjenhiemstra/ithowifi #324: Unreliable connection to WPU

rjvbeek, 2025-05-26:
## Describe the bug
My WPU 5G seems to work very unreliably with the nrgwatch module. Often, the status message contains 0 records until I restart and I have to keep restarting until the I2C connection fails entirely throwing the error "i2c bus stuck - please power cycle the Itho unit". It has worked for a few days and then started failing (I don't think  I changed anything).

## To Reproduce
No specific steps needed, I'm just looking for ways to make the system work reliably.

## Device information
- Firmware version: 2.8.0
- Hardware revision: NON-CVE 1
- Type/model number Itho: WPU 45I-5G / 03-00354
- CC1101 RF module enabled: no

## Debug logging
I2C debugging info:
<html>
<body>

origin | start | duration | error | final
<br />
PWM_CMD | 16934 | 5538 | I2C_ERROR_SCL_LOW | I2C_NOK<br />
QUERY_DEVICE_TYPE | 16800 | 8 | I2C_ERROR_CLEARED_OK | I2C_NOK

</body>
</html>

arjenhiemstra, 2025-05-26:
A little warning; the i2c debug function will halt de i2c bus after an error. So best to not leave this function switched on during normal use.

What kind of UTP cable are you using?

arjenhiemstra, 2025-05-26:
Ok, if the communication with the I2C but is still unreliable, please let me know