arjenhiemstra/ithowifi #329: Feature request: remaining time of timer command

retorquere, 2025-06-07:
## Describe the bug

I have RF remotes paired to the addon on a CVE, and the addon configured in home assistant. If a timer1/timer2/timer3 command is sent, would it be possible to expose the time remaining in a way that it can be read in HA?

## To Reproduce

1. Join the remotes on the addon
2. Give a timer command with the remote, the CVE goes to high
3. The time remaining is not available on the HA device 

## Device information
- Firmware version  2.8.0
- Hardware revision CVE 2
- CC1101 RF module enabled yes

retorquere, 2025-06-07:
Apologies, I should not have submitted this without complying with the submission template -- will get one when I'm home!

retorquere, 2025-06-07:
[logfile1.current.txt](https://github.com/user-attachments/files/20642190/logfile1.current.txt)
[logfile0.txt](https://github.com/user-attachments/files/20642191/logfile0.txt)

arjenhiemstra, 2025-06-08:
Thanks!
Looking at your model number, you should be able to activate the i2c command "System settings" -> "Ventilation status (31DA)" to on

This will give you a few extra items on the "Itho status" page. One of them is: "RemainingTime (min)" which is the remaining timer time if a timer is active.

btw. if you update to the latest beta13 version, you can expose these status items to HA through auto discovery.

retorquere, 2025-06-08:
Alright, I've turned that on and installed beta 13, I have restarted the addon and HA to be on the safe side, but I'm not seeing any entities on the device (which is what I thought you described as being one of the benefits of the beta). In the DevTools/States tab, I currently see

```
preset_modes: Low, Medium, High, Timer 10min, Timer 20min, Timer 30min
percentage: 8
percentage_step: 1
preset_mode: null
Ventilation level (%): 2
Fan setpoint (rpm): 577
Fan speed (rpm): 583
Error: 0
Selection: 3
Startup counter: 97
Total operation (hours): 113907
friendly_name: Ventilatie
supported_features: 57
```

but I think that this is what I saw before.

retorquere, 2025-06-08:
I do have the itho wifi addon installed. I'm new to HA and there's a lot to process, but before you do more work on my behalf let me just read read the wiki again -- a lot of stuff is starting to click now.

retorquere, 2025-06-08:
OK so I have the entities card installed but when filtering I don't see anything relating to time remaining. I tried to download a new log, and it did download (see attachment), but the addon UI is stuck on

﻿
[image]

[logfile1.current.txt](https://github.com/user-attachments/files/20646759/logfile1.current.txt)

arjenhiemstra, 2025-06-08:
If you go to HA Discovery you should be able to check the items you would like to have in HA

[image]

In HA the device will be discovered and that should look something like this:

[image]

arjenhiemstra, 2025-06-08:
https://github.com/user-attachments/assets/5d076eab-e313-4b8b-b998-25490c6a76f5

Here is a small clip of how it should work

retorquere, 2025-06-08:
I set it to 10, saved, and waited one minute, but the "HA Discovery" remained at "I2C command 31da activated but not ready". As soon as I turn that off, the HA Discovery page returns. That is with just `PWM2I2C` and `2401` on, rest off.

Should I try with the release build to see whether that makes a difference?

Minor point, but there's an extra parenthesis in

> Fan system status (31D9))

> If you open the developer console of you browser you might see some relevant error messages related to this.

Nothing at all. When I enable everything, including requests, I see a single call to http://nrg-itho-bc4c.local/curlog when I open the tab, but nothing else, even when I click "download log" (which may mean download just takes whatever is already displayed?). But I've now tried on Firefox, Safari and Edge, and it only happens on Firefox (on the beta FW).

arjenhiemstra, 2025-06-09:
Then I'm afraid your model's firmware version doesn't support the 31DA command unfortunately...

retorquere, 2025-06-09:
Ah well, I'm very happy with the functionality the addon does provide, the timer would just have been the icing on the cake.

I think the addon does expose the commands being sent, is that correct? Then I'll just try to synthesize the time remaining. If not, no problem.

arjenhiemstra, 2025-06-09:
With
`api.html?get=lastcmd`

You can get the last command processed.

If you activate api v2 you get an output like this:

```JSON
{
    "status": "success",
    "data": {
        "timestamp": 1749464119,
        "cmdkey": "get",
        "cmdvalue": "lastcmd",
        "lastcmd": {
            "source": "MQTT API",
            "command": "speed:255",
            "timestamp": 1749462852
        }
    }
}
```

within "lastcmd" there is a timestamp, this was the moment the last command was executed.

retorquere, 2025-06-09:
Excellent!

closed 2025-08-15 by stale[bot]