arjenhiemstra/ithowifi #331: Support: Clarify CO2 and Humidity sensor behavior

PrivatePuffin, 2025-06-08:
## Describe the bug
A clear and concise description of what the bug is.

## To Reproduce
Steps to reproduce the behaviour:
NOT RELEVVANT

## Expected behaviour
NOT RELEVVANT

## Screenshots
NOT RELEVVANT

## Device information
NOT RELEVVANT

## Syslog
NOT RELEVVANT

## Desktop (please complete the following information):
NOT RELEVVANT

## Smartphone (please complete the following information):
NOT RELEVVANT

Just for clarity:

According to the wiki, with an external CO2 sensor:
If set to medium, regardless of the addon fan speed setting, it will always speed up based on CO2 as long as the CO2 sensor is directly connected to the itho and not to the addon?

How about the itho external humidity sensors, would that function in the same way?

---

Also:
Dont offer the empty issues option please, if you've a bot that insta closes them and just remove the option.

arjenhiemstra, 2025-06-08:
>If set to medium, regardless of the addon fan speed setting, it will always speed up based on CO2 as long as the CO2 sensor is directly connected to the itho and not to the addon?

Yes. I believe the CO2 speed setting overrules the PWM2I2C speed setting but I'm not sure. 

>How about the itho external humidity sensors, would that function in the same way?

I have my RFT RV connected to the add-on. I'll have to connect it to my itho to be able to test te behaviour but I expect it to be the same or comparable to the RFT CO2

PrivatePuffin, 2025-06-09:
> Yes. I believe the CO2 speed setting also overrules the PWM2I2C speed setting but I'm not sure.

I've gotten a few CO2 sensors and switched the addon to "virtual remote" mode, without PWM2i2c.
Once thats working well, I'll see with home assistant how-much the behavior will change using PWM2I2C mode :)

arjenhiemstra, 2025-06-09:
Just to be clear;
If you use virtual remote, that looks for the itho firmware just as a regular physical remote and functions the same, only this is send bij de 2x4 pin heaer to the itho firmware directly instead of via RF.

The PWM2I2C only works when the itho is in medium or auto mode. 

You can play with both of course, just my two cents might you have missed this piece of info

PrivatePuffin, 2025-06-09:
@arjenhiemstra Yes I'm aware, thats precisely why I have switched to use virtual remote.
I want to ensure the CO2 sensors work correctly without complications from PWM2I2C.

It gives me the most certainty that I can predict how it will react prior to setting up the new sensors.
Once I know how virtual remote reacts, I can see if the behavior is different with PWM2I2C.

I dont want to change all sorts of variables at once.

PrivatePuffin, 2025-06-10:
Though @arjenhiemstra is it technically possible for the addon to read co2 and rv and pass those through as a toggleable behavior?

Instead of the current "ignore" behavior?

So that we can have the itho handle co2 and rv rampup.

arjenhiemstra, 2025-06-11:
What do you mean with : 
"pass those through as a toggleable behavior"
and
"Instead of the current "ignore" behavior?" 

Sorry, I can make some guesses but im not sure what you mean by these quesions

WO15, 2025-06-13:
Privatepuffin, Not sure which HW you use but for Itho CVE-S optima 2, behavior of box is quite complicated. 

- With no CO2 sensor connected pressing quickly two times of Auto button will not work. Box will go or stay in auto mode (7) 
- With one CO2 connector connected, box seems to react as Optima 2, pressing quickly two times of Auto button will start "auto night" mode (8). speed will increase with values 34-39 under "Itho settings". Exact values are based on settings for 1 of multiple floor building and 1, 2 or multi  persons. 
- With 2 CO2 sensor connected box seems to behave like Optima 2. Speed will increase with values 40-45 under "Itho settings". 
Easy trick to see which mode the box is in, is to increase the settings above to e.g. 83% one by one. Note make screenshot/printout before you start changing these settings.

arjenhiemstra, 2025-06-14:
> * it goes to setting 8 "unknown", possible auto-night?

Where do you see "unknown", I guess this is probably the FanInfo label? If so, could you please post the raw output of the debug function 31DA?